evcc-io/evcc #11804: modbus proxy timeout

markusschaefer, 2024-01-21:
### Describe the bug

Multiple timeouts occure when client is connected to modbus proxy. The client is the huawei_solar integration for home assistant (https://github.com/wlcrs/huawei_solar)

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
modbusproxy:
  - port: 502
    uri: xxx.xxx.xxx.xxx:502
```

### Log details

```text
[proxy-502] DEBUG 2024/01/21 21:41:36 modbus proxy for xxx.xxx.xxx.xxx:502 listening at :502
[proxy-502] DEBUG 2024/01/21 21:42:51 failed to write response: write tcp xxx.xxx.xxx.xxx:502->yyy.yyy.yyy.yyy:41417: i/o timeout
[proxy-502] DEBUG 2024/01/21 21:43:33 failed to write response: write tcp xxx.xxx.xxx.xxx:502->yyy.yyy.yyy.yyy:41417: i/o timeout
[proxy-502] DEBUG 2024/01/21 21:44:51 failed to write response: write tcp xxx.xxx.xxx.xxx:502->yyy.yyy.yyy.yyy:43865: i/o timeout
[proxy-502] DEBUG 2024/01/21 21:46:13 failed to write response: write tcp xxx.xxx.xxx.xxx:502->yyy.yyy.yyy.yyy:58729: i/o timeout
[proxy-502] DEBUG 2024/01/21 21:47:29 failed to write response: write tcp xxx.xxx.xxx.xxx:502->yyy.yyy.yyy.yyy:56717: i/o timeout
[proxy-502] DEBUG 2024/01/21 21:48:16 failed to write response: write tcp xxx.xxx.xxx.xxx:502->yyy.yyy.yyy.yyy:56717: i/o timeout
[proxy-502] DEBUG 2024/01/21 21:50:04 failed to write response: write tcp xxx.xxx.xxx.xxx:502->yyy.yyy.yyy.yyy:56717: i/o timeout
[proxy-502] DEBUG 2024/01/21 21:51:24 failed to write response: write tcp xxx.xxx.xxx.xxx:502->yyy.yyy.yyy.yyy:60761: i/o timeout
[proxy-502] DEBUG 2024/01/21 21:52:05 failed to write response: write tcp xxx.xxx.xxx.xxx:502->yyy.yyy.yyy.yyy:60761: i/o timeout
```

### What type of operating system are you running?

Docker container

### Version

0.123.9

premultiply, 2024-01-22:
Huawei TCP is buggy and slow.
You can do a single request every 5 seconds or so.
If you need fast and stable access use RS485 only.

closed 2024-01-22 by premultiply

premultiply, 2024-01-22:
Dongle on the inverter

markusschaefer, 2024-01-23:
> Dongle on the inverter

The timeout for the proxy clients is hardcoded to 30 seconds when creating the modbus server.
Is it possible to increase the timeout by configuration?

https://github.com/andig/mbserver/blob/1d29cbb5820ebc65c9e5150745d1941125d64bca/server.go#L155