evcc-io/evcc #11807: Workaround für fehlendes "ResetOnDisconnect" in der vehicle-Sektion

Uhrenradio, 2024-01-22:
Analog zur Diskussion in "Gastfahrzeuge mit einem Standard-Config #8513" habe ich meine Wallbox auf der Einfahrt generell frei gegeben und ich möchte gerne meine Wallbox generell für Gäste im PV-Betrieb freigeschaltet haben. 
Das habe ich in der Vergangenheit über das Feature "[ResetOnDisconnect: true](https://github.com/evcc-io/evcc/discussions/8513)" und der Einstellung "Mode: PV" in den LoadPoint-Einstellugnen gelöst.

Mit den neuen Integrationen in dem UI sind die entsprechenden Einstellungen in der evcc.yaml nicht mehr erlaubt.

Damit ich jetzt um eine Node-Red-Automatisierung zum automatischen Rücksetzen des Lademodus bei Gastfahrzeugen herumkomme folgende Idee:
Schaffen einer Verknüpfung zum Fahrzeug mit Namen "guest", dessen Einstellungen immer dann geladen werden, wenn kein anderes der hinterlegten Fahrzeuge erkannt wurde. In dieser vehicle-config kann dann alles das was bei unbekanntem FZG (name: guest) möglich ist eingestellt werden. Insbesondere der default-Lademodus.

Oder habe ich eine simplere Lösung übersehen?

VolkerK62, 2024-01-22:
`resetOnDisconnect` wird doch jetzt "automatisch" durchgeführt.
Der Mode am Loadpoint wird eingestellt, wenn ein Fahrzeug abgesteckt wird.

Lako-Unbekannt, 2024-01-22:
Hi!

Soweit ich das verstanden habe musst du nur Mode:PV in der .yaml lassen. Das ist immer noch erlaubt, und solange ein mode angegeben ist wird resetondisconect als wahr angesehen.

andig, 2024-01-22:
So ist es.

closed 2024-01-22 by andig

Uhrenradio, 2024-01-23:
Gestern so wie vorgeschlagen meine Config verändert und siehe da: Alles läuft wie gewünscht! Perfekt...

BTW: Braucht Ihr ggf. Unterstützung bei der Pflege der Dokumentation auf der evcc.io Seite? Ihr bringt die neuen Features in so hoher Taktrate, da hinkt die Doku ein wenig hinterher...

premultiply, 2024-01-23:
Ja, immer gerne.
Jeder kann die Doku bearbeiten und Änderungen vorschlagen.
Ist auch nur ein git-Repository.
Link dazu findet sich auf jeder Dokuseite.