evcc-io/evcc #11810: Kopplung mit BMW Fahrzeug nicht möglich

robert-ke, 2024-01-22:
### Discussed in https://github.com/evcc-io/evcc/discussions/11809

<div type='discussions-op-text'>

<sup>Originally posted by **robert-ke** January 22, 2024</sup>
Hallo zusammen,

ich bin aktuell am anbinden sämtlicher Komponenten, die im Haus verbaut sind und habe einen Trial Token in Verwendung. Seitdem der Token hinterlegt ist, funktioniert auch wie erwartet meine Hardy Barth Wallbox.

Leider bekomme ich das Auto nicht online - folgender Fehler:
[image]

Das Kennwort wurde neu gesetzt, funktioniert in der BMW iOS App problemlos. Zeitgleich ist der ioBroker BMW Adapter grün und funktioniert:
[image]

Hier ein Screenshot von der evcc.yaml Datei bzgl. des Fahrzeugs:
[image]

EVCC ist in der aktuellen Version frisch installiert:
https://github.com/evcc-io/evcc/releases/tag/0.123.9

und der Token läuft noch ein paar Tage:
[image]

Hat jemand eine Idee, warum die BMW Verbindung nicht klappt und mit einem authentication 401 Fehler fliegt?
Alle gängigen Suchergebnisse wie Kennwort ändern / DNS Themen / Neustarts bereits mehrfach gecheckt.

Danke & Gruß,
Robert</div>

andig, 2024-01-22:
Logfile fehlt

closed 2024-01-22 by andig

robert-ke, 2024-01-22:
Hier ein Auszug aus der syslog - direkt nach einem Neustart des Dienstes:
Jan 22 03:06:57 sv-evcc systemd[1]: Stopped evcc.
Jan 22 03:06:57 sv-evcc systemd[1]: evcc.service: Consumed 2.296s CPU time.
Jan 22 03:06:58 sv-evcc systemd[1]: Reloading.
Jan 22 03:06:59 sv-evcc systemd[1]: Reloading.
Jan 22 03:06:59 sv-evcc systemd[1]: Starting Daily man-db regeneration...
Jan 22 03:07:00 sv-evcc systemd[1]: Started evcc.
Jan 22 03:07:00 sv-evcc evcc[13777]: [main  ] INFO 2024/01/22 03:07:00 evcc 0.123.9
Jan 22 03:07:00 sv-evcc evcc[13777]: [main  ] INFO 2024/01/22 03:07:00 using config file: /etc/evcc.yaml
Jan 22 03:07:00 sv-evcc evcc[13777]: [main  ] INFO 2024/01/22 03:07:00 starting ui and api at :7070
Jan 22 03:07:00 sv-evcc evcc[13777]: [db    ] INFO 2024/01/22 03:07:00 using sqlite database: /var/lib/evcc/evcc.db
Jan 22 03:07:00 sv-evcc systemd[1]: man-db.service: Deactivated successfully.
Jan 22 03:07:00 sv-evcc systemd[1]: Finished Daily man-db regeneration.
Jan 22 03:07:03 sv-evcc evcc[13777]: [main  ] ERROR 2024/01/22 03:07:03 creating vehicle XXXXX failed: cannot create vehicle type 'template': cannot create vehicle type 'bmw': Client authentication failed (e.g., login failure, unknown client, no client authentication included or unsupported authentication method): unexpected status: 401 (Unauthorized)
Jan 22 03:07:03 sv-evcc evcc[13777]: [lp-1  ] DEBUG 2024/01/22 03:07:03 charge total import: 7982.200kWh
Jan 22 03:07:03 sv-evcc evcc[13777]: [site  ] WARN 2024/01/22 03:07:03 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
Jan 22 03:07:03 sv-evcc evcc[13777]: [site  ] INFO 2024/01/22 03:07:03 site config:
Jan 22 03:07:03 sv-evcc evcc[13777]: [site  ] INFO 2024/01/22 03:07:03   meters:      grid ✓ pv ✓ battery ✓
Jan 22 03:07:03 sv-evcc evcc[13777]: [site  ] INFO 2024/01/22 03:07:03     grid:      power ✓ energy ✓ currents ✓
Jan 22 03:07:03 sv-evcc evcc[13777]: [site  ] INFO 2024/01/22 03:07:03     pv 1:      power ✓ energy ✓ currents ✓
Jan 22 03:07:03 sv-evcc evcc[13777]: [site  ] INFO 2024/01/22 03:07:03     pv 2:      power ✓ energy ✓ currents ✓
Jan 22 03:07:03 sv-evcc evcc[13777]: [site  ] INFO 2024/01/22 03:07:03     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✗
Jan 22 03:07:03 sv-evcc evcc[13777]: [site  ] INFO 2024/01/22 03:07:03     battery 2: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
Jan 22 03:07:03 sv-evcc evcc[13777]: [site  ] INFO 2024/01/22 03:07:03   vehicles:
Jan 22 03:07:03 sv-evcc evcc[13777]: [site  ] INFO 2024/01/22 03:07:03     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
Jan 22 03:07:03 sv-evcc evcc[13777]: [lp-1  ] INFO 2024/01/22 03:07:03 loadpoint 1:
Jan 22 03:07:03 sv-evcc evcc[13777]: [lp-1  ] INFO 2024/01/22 03:07:03   mode:        off
Jan 22 03:07:03 sv-evcc evcc[13777]: [lp-1  ] INFO 2024/01/22 03:07:03   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
Jan 22 03:07:03 sv-evcc evcc[13777]: [lp-1  ] INFO 2024/01/22 03:07:03   meters:      charge ✓
Jan 22 03:07:03 sv-evcc evcc[13777]: [lp-1  ] INFO 2024/01/22 03:07:03     charge:    power ✓ energy ✓ currents ✓
Jan 22 03:07:03 sv-evcc evcc[13777]: [site  ] DEBUG 2024/01/22 03:07:03 set buffer soc: 65
Jan 22 03:07:03 sv-evcc evcc[13777]: [site  ] DEBUG 2024/01/22 03:07:03 set buffer start soc: 0

andig, 2024-01-22:
Schick gerne mal Zugangsdaten an info@evcc.io

robert-ke, 2024-01-22:
kein Problem, Zugangsdaten zum BMW Portal sind raus

andig, 2024-01-22:
Soweit ich sehe fragst Du einfach zu oft ab:

    { "statusCode": 403, "message": "Out of call volume quota. Quota will be replenished in 00:03:18." }

robert-ke, 2024-01-22:
okay, kann ich im EVCC den Refresh dafür irgendwo anpassen?

Im ioBroker seh ich gerade, steht er auf 2 Minuten:
[image]

Vielleicht ist das in Summe dann wenn 2 Systeme bei der API abholen zu sportlich...

Ich hab den ioBroker mal auf 10 Min gestellt.

robert-ke, 2024-01-22:
Also, ich hab vor gut ner Stunde den ioBroker BMW Adapter auf 10 Min umgestellt. EVCC gerade neu gestartet, dennoch kommt der gleiche Fehler:

[image]

während der ioBroker Adapter schön alle 10 Min. die Daten abholt. Noch eine Idee?