evcc-io/evcc #11812: Modbus Exception: bei aktiver Batteriesteuerung für SMA Sunny Island SI4.4M-11

andromedavpn, 2024-01-22:
### Describe the bug

Ich nutze evcc as Add-On in Home Assistant mit
SMA Sunny Island SI4.4M-11 672
Firmware-Version des Hauptprozessors 3.02.11.R
Firmware-Version des Protokoll-Konverters 1.03.4.R
Firmware-Version der Logikkomponente 3.02.8.R
Firmware-Version 3.02.6.R

Dieser ist folgendermaßen konfiguriert
```
  - name: sunnyisland
    type: template
    template: sma-inverter-modbus #working measurement
    usage: battery
    # Modbus TCP
    modbus: tcpip
    id: 3
    host: 192.168.188.24 # Hostname
    port: 502 # Port
```
Bei Aktivierung der "Batteriesteuerung" erhalte ich die Fehlermeldung
_battery mode: modbus: exception '2' (illegal data address), function '16'_
und die Aktivierung hat keine Wirkung

Bei Nutzung von sma-hybrid template, wie in [discussion 11702](https://github.com/evcc-io/evcc/discussions/11702) vorgeschlagen
  ```
- name: sunnyisland
    type: template
    template: sma-hybrid #working active battery control
    usage: battery
    # Modbus TCP
    modbus: tcpip
    id: 3
    host: 192.168.188.24 # Hostname
    port: 502 # Port
```
funktioniert zwar die Batteriesteuerung, die Lade- und Entladeleistung des Sunny Islands werden jedoch nicht erkannt und nicht angezeigt.

### Steps to reproduce

1. Configure SMA Sunny Island SI4.4M-11 using sma-inverter-modbus template as described in documentation
2. Activate active battery control
3. evcc reports error _battery mode: modbus: exception '2' (illegal data address), function '16'_
...

### Configuration details

```yaml
- name: sunnyisland
    type: template
    template: sma-inverter-modbus #working measurement
    usage: battery
    # Modbus TCP
    modbus: tcpip
    id: 3
    host: 192.168.188.24 # Hostname
    port: 502 # Port
```
```

### Log details

```text
[site  ] ERROR 2024/01/22 12:07:25 battery mode: modbus: exception '2' (illegal data address), function '16'
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.123.9

andig, 2024-01-22:
Was sagt denn

    evcc sunspec

für den WR?

andig, 2024-01-22:
Da wo's geht- also vmtl. Windows

andig, 2024-01-22:
Da fehlt noch ein `--id 3` laut deiner Config

andromedavpn, 2024-01-22:
Mit --id 3 ist das Ergebnis unverändert
```
PS C:\PersonalData\Portable Apps\evcc> .\evcc.exe -l trace --id 3 sunspec 192.168.188.24:502
[main  ] INFO 2024/01/22 13:56:55 evcc 0.123.9
[main  ] TRACE 2024/01/22 13:56:55 modbus: send 00 01 00 00 00 06 03 03 9c 40 00 02
[main  ] TRACE 2024/01/22 13:56:55 modbus: recv 00 01 00 00 00 03 03 83 02
[main  ] TRACE 2024/01/22 13:56:55 modbus: send 00 02 00 00 00 06 03 03 c3 50 00 02
[main  ] TRACE 2024/01/22 13:56:55 modbus: recv 00 02 00 00 00 07 03 03 04 ff ff ff ff
[main  ] TRACE 2024/01/22 13:56:55 modbus: send 00 03 00 00 00 06 03 03 00 00 00 02
[main  ] TRACE 2024/01/22 13:56:55 modbus: recv 00 03 00 00 00 07 03 03 04 ff ff ff ff
[main  ] FATAL 2024/01/22 13:56:55 not a SunSpec device
```

premultiply, 2024-01-22:
SunSpec ist bei SMA unter ID 126 😉

andig, 2024-01-22:
...also dann kann das aber vor Einschalten von Battery Control nie funktioniert haben 😝

andromedavpn, 2024-01-22:
good to know
Habe den output von 
`.\evcc.exe -l trace --id 126 sunspec 192.168.188.24:502 > info.txt`
angehängt
[info.txt](https://github.com/evcc-io/evcc/files/14010531/info.txt)

andromedavpn, 2024-01-22:
> ...bevor wir das jetzt anschauen: probierst Du bitte Deine Config nochmal mit 126?

Done. Siehe oben

andig, 2024-01-22:
Nee, ich meine evcc als Steuerung laufen zu lassen, nicht sunspec.

andig, 2024-01-22:
> Mit Ausnahme von active control funktionier alles mit id=3 (so wie in evcc doc auch angegeben).

@premultiply wir haben id 3 in den Templates?

andromedavpn, 2024-01-22:
> > Mit Ausnahme von active control funktionier alles mit id=3 (so wie in evcc doc auch angegeben).
> 
> @premultiply wir haben id 3 in den Templates?

Ich meinte die doku für 
Sunny Island (Modbus)
unter
https://docs.evcc.io/docs/devices/meters

premultiply, 2024-01-22:
> wir haben id 3 in den Templates?

Klar. Ist ja auch richtig so.
Die Register des SBS gibt es aber beim SI nicht.
Daher die Fehlermeldung.
Hier bedarf es daher noch eines eigenen, neuen SI-Templates.

andromedavpn, 2024-01-22:
> > wir haben id 3 in den Templates?
> 
> Klar. Ist ja auch richtig so. Die Register des SBS gibt es aber beim SI nicht. Daher die Fehlermeldung. Hier bedarf es daher noch eines eigenen, neuen SI-Templates.

Bin verwirrt. Die Doku sagt doch explizit für Sunny Island und nicht SB
```
Sunny Island (Modbus)
[aktive Batteriesteuerung](https://docs.evcc.io/docs/devices/meters#features)
meters:
  - name: my_battery
    type: template
    template: sma-inverter-modbus
    usage: battery
    
    # Modbus TCP
    modbus: tcpip
    id: 3
    host: 192.0.2.2 # Hostname
    port: 502 # Port
```

premultiply, 2024-01-22:
Die ganze Diskussion ist eigentlich völlig überflüssig. :)
Ich weiß was zu tun ist.

Das Template passt aktuell nicht zum SI obwohl es so heisst.
Das ist ein Bug. Ich werde ihn beheben. Danke :)

andromedavpn, 2024-01-22:
> Ich werde ihn beheben. Danke :)
👍

PR #11909 (2024-01-28): SMA Modbus: split templates
Fixes #11812

There may be other Modbus registers to control SI in a smarter way, but that would require testing on a real device.
closed 2024-01-28 by andig