evcc-io/evcc #11814: Kia EV9 not supported

andig, 2024-01-22:
Das API hat sich geändert. Es braucht Issue und Zugangsdaten für info@evcc.io. Depends on https://github.com/Hyundai-Kia-Connect/hyundai_kia_connect_api/commit/0fa2ca443615c76869d434cb2186d1044390f923

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/11595#discussioncomment-8209720_

VolkerK62, 2024-01-24:
richtiger wäre "Kia/Hyundai: ..." ;-)

closed 2024-02-03 by andig

henrikssn, 2024-04-18:
Is this fixed? I have a Kia EV9 and don't get any data, seems like it doesn't correctly parse the response from Bluelink.

Happy to contribute a fix if you agree.

andig, 2024-04-18:
Fix would be great. Issue was closed due to missing input.

wjasperse, 2025-04-28:
I've got a KIA EV9 as well. Problem is that not all info is available, only odo / total km's driven with the car. That said, if I use the Kia Connect app to login to the KIA servers, the app is updated, but also evcc is able to update the information from the Kia servers. 

It seems there is a session available for around 10 minutes; after that evcc stops pulling data and everything gets red again (all but odometer).

Again, update the Kia Connect refresh the connection and evcc is able to update the infor again.