evcc-io/evcc #11832: Tesla offline

jroegner, 2024-01-23:
### Describe the bug

Seit heute ist mein Tesla nicht mehr über die API erreichbar. Bei Anstecken an die Wallbox wird das Auto als Gastfahrzeug erkannt. Der Tesla wird in der Liste als "Offline" angezeigt.
Das ganze ist erst nach einem Restart des Services aufgetreten.

### Steps to reproduce

evcc vehicle

```
[main  ] ERROR 2024/01/23 22:16:37 creating vehicle ev2 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': cannot get vehicles: 412 Precondition Failed
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'tesla': cannot get vehicles: 412 Precondition Failed
Capacity: 79.5kWh
Features: [Offline]
```

### Configuration details

```yaml
chargers:
- type: template
  template: twc3
  host: x.x.x.x
  name: wallbox3

vehicles:
- type: template
  template: tesla
  title: Model 3P
  icon: car
  accessToken: xxx
  refreshToken: xxx
  vin: xxx
  capacity: 79.5

loadpoints:
- title: Carport
  charger: wallbox3
  mode: off
  phases: 3
  mincurrent: 5
  maxcurrent: 13
  phases: 3
  minCurrent: 5
  maxCurrent: 16
  name: ev2
```

### Log details

```text
[main  ] ERROR 2024/01/23 22:24:01 creating vehicle ev2 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': cannot get vehicles: 412 Precondition
```

### What type of operating system are you running?

Linux

### Version

0.123.9

markusschaefer, 2024-01-23:
Funktioniert bei mir ebenfalls nicht mehr

```
[main  ] ERROR 2024/01/23 22:47:52 creating vehicle my_tesla failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': cannot get vehicles: 429 Too Many Requests
```

yipikaye, 2024-01-23:
Same problem here.
I also noticed tesla tokens cannot sign in

infinus, 2024-01-24:
Tesla is updating their API so using TWC3 will not work at the moment.

Tesla aktualisiert seine API, sodass die Verwendung von TWC3 derzeit nicht funktioniert.

closed 2024-01-24 by andig

braeunli, 2024-02-05:
Wie wurde das gelöst?