evcc-io/evcc #11838: Cannot unmarshal number 77.0 into Go struct field Energy.Energy.Level of type int

PatriotIT, 2024-01-24:
### Describe the bug

Hallo zusammen,

seit heute Morgen haben wir nur noch "Gastfahrzeuge" bei uns im EVCC, kein Fahrzeug wird mehr korrekt zugeordnet und das Log ist voller Fehler.

Angefangen hatte es mit einem Citreon eC4, der seinen SoC Zustand nicht mehr korrekt angezeigt hatte.

Nach einem Neustart von EVCC sind nun alle Fahrzeuge betroffen.

Wir haben 6 Wallboxen und somit 12 Ladepunkte. So ein Phänomen ist allerdings neu... 😄 Ich vermute einen Zusammenhang mit Opel? Diese haben ja aktuell (wieder) ein Problem, weil die API erneuert wurde. Gleichzeitig scheint aber auch die Tesla API aktuell ein Problem zu haben. Ggf. kommt EVCC mit irgendwelchen zurück gelieferten Werten nicht klar, sodass ein Fehler / Timeout Wert nicht korrekt abgefangen wird und dadurch in der Software etwas durcheinander gerät?

### Steps to reproduce

-

### Configuration details

```yaml
-
```

### Log details

```text
[lp-1  ] INFO 2024/01/24 10:35:06 loadpoint 1:
[lp-1  ] INFO 2024/01/24 10:35:06   mode:        pv
[lp-1  ] INFO 2024/01/24 10:35:06   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/01/24 10:35:06   meters:      charge ✓
[lp-1  ] INFO 2024/01/24 10:35:06     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2024/01/24 10:35:06 loadpoint 2:
[lp-2  ] INFO 2024/01/24 10:35:06   mode:        pv
[lp-2  ] INFO 2024/01/24 10:35:06   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-2  ] INFO 2024/01/24 10:35:06   meters:      charge ✓
[lp-2  ] INFO 2024/01/24 10:35:06     charge:    power ✓ energy ✓ currents ✓
[lp-3  ] INFO 2024/01/24 10:35:06 loadpoint 3:
[lp-3  ] INFO 2024/01/24 10:35:06   mode:        pv
[lp-3  ] INFO 2024/01/24 10:35:06   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-3  ] INFO 2024/01/24 10:35:06   meters:      charge ✓
[lp-3  ] INFO 2024/01/24 10:35:06     charge:    power ✓ energy ✓ currents ✓
[lp-4  ] INFO 2024/01/24 10:35:06 loadpoint 4:
[lp-4  ] INFO 2024/01/24 10:35:06   mode:        pv
[lp-4  ] INFO 2024/01/24 10:35:06   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-4  ] INFO 2024/01/24 10:35:06   meters:      charge ✓
[lp-4  ] INFO 2024/01/24 10:35:06     charge:    power ✓ energy ✓ currents ✓
[lp-5  ] INFO 2024/01/24 10:35:06 loadpoint 5:
[lp-5  ] INFO 2024/01/24 10:35:06   mode:        pv
[lp-5  ] INFO 2024/01/24 10:35:06   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-5  ] INFO 2024/01/24 10:35:06   meters:      charge ✓
[lp-5  ] INFO 2024/01/24 10:35:06     charge:    power ✓ energy ✓ currents ✓
[lp-6  ] INFO 2024/01/24 10:35:06 loadpoint 6:
[lp-6  ] INFO 2024/01/24 10:35:06   mode:        pv
[lp-6  ] INFO 2024/01/24 10:35:06   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-6  ] INFO 2024/01/24 10:35:06   meters:      charge ✓
[lp-6  ] INFO 2024/01/24 10:35:06     charge:    power ✓ energy ✓ currents ✓
[lp-7  ] INFO 2024/01/24 10:35:06 loadpoint 7:
[lp-7  ] INFO 2024/01/24 10:35:06   mode:        pv
[lp-7  ] INFO 2024/01/24 10:35:06   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-7  ] INFO 2024/01/24 10:35:06   meters:      charge ✓
[lp-7  ] INFO 2024/01/24 10:35:06     charge:    power ✓ energy ✓ currents ✓
[lp-8  ] INFO 2024/01/24 10:35:06 loadpoint 8:
[lp-8  ] INFO 2024/01/24 10:35:06   mode:        pv
[lp-8  ] INFO 2024/01/24 10:35:06   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-8  ] INFO 2024/01/24 10:35:06   meters:      charge ✓
[lp-8  ] INFO 2024/01/24 10:35:06     charge:    power ✓ energy ✓ currents ✓
[lp-9  ] INFO 2024/01/24 10:35:06 loadpoint 9:
[lp-9  ] INFO 2024/01/24 10:35:06   mode:        pv
[lp-9  ] INFO 2024/01/24 10:35:06   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-9  ] INFO 2024/01/24 10:35:06   meters:      charge ✓
[lp-9  ] INFO 2024/01/24 10:35:06     charge:    power ✓ energy ✓ currents ✓
[lp-10 ] INFO 2024/01/24 10:35:06 loadpoint 10:
[lp-10 ] INFO 2024/01/24 10:35:06   mode:        pv
[lp-10 ] INFO 2024/01/24 10:35:06   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-10 ] INFO 2024/01/24 10:35:06   meters:      charge ✓
[lp-10 ] INFO 2024/01/24 10:35:06     charge:    power ✓ energy ✓ currents ✓
[lp-11 ] INFO 2024/01/24 10:35:06 loadpoint 11:
[lp-11 ] INFO 2024/01/24 10:35:06   mode:        pv
[lp-11 ] INFO 2024/01/24 10:35:06   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-11 ] INFO 2024/01/24 10:35:06   meters:      charge ✓
[lp-11 ] INFO 2024/01/24 10:35:06     charge:    power ✓ energy ✓ currents ✓
[lp-12 ] INFO 2024/01/24 10:35:06 loadpoint 12:
[lp-12 ] INFO 2024/01/24 10:35:06   mode:        pv
[lp-12 ] INFO 2024/01/24 10:35:06   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-12 ] INFO 2024/01/24 10:35:06   meters:      charge ✓
[lp-12 ] INFO 2024/01/24 10:35:06     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2024/01/24 10:35:07 car disconnected
[lp-2  ] INFO 2024/01/24 10:35:36 car connected
[main  ] ERROR 2024/01/24 10:35:37 vehicle status: json: cannot unmarshal number 77.0 into Go struct field Energy.Energy.Level of type int
[main  ] WARN 2024/01/24 10:35:39 vehicle status: >1 matches, giving up
[lp-3  ] INFO 2024/01/24 10:36:06 car disconnected
[lp-4  ] INFO 2024/01/24 10:36:36 car disconnected
[lp-5  ] INFO 2024/01/24 10:37:06 car disconnected
[lp-6  ] INFO 2024/01/24 10:37:36 car connected
[lp-6  ] INFO 2024/01/24 10:37:36 start charging ->
[main  ] ERROR 2024/01/24 10:37:38 vehicle status: json: cannot unmarshal number 77.0 into Go struct field Energy.Energy.Level of type int
[main  ] WARN 2024/01/24 10:37:41 vehicle status: >1 matches, giving up
[lp-6  ] WARN 2024/01/24 10:37:41 charger logic error: disabled but charging
[lp-7  ] INFO 2024/01/24 10:38:06 car disconnected
[lp-8  ] INFO 2024/01/24 10:38:36 car connected
[lp-8  ] INFO 2024/01/24 10:38:36 start charging ->
[main  ] ERROR 2024/01/24 10:38:40 vehicle status: json: cannot unmarshal number 77.0 into Go struct field Energy.Energy.Level of type int
[main  ] WARN 2024/01/24 10:38:42 vehicle status: >1 matches, giving up
[lp-8  ] WARN 2024/01/24 10:38:42 charger logic error: disabled but charging
[lp-9  ] INFO 2024/01/24 10:39:06 car disconnected
[lp-10 ] INFO 2024/01/24 10:39:36 car connected
[lp-10 ] INFO 2024/01/24 10:39:36 start charging ->
[main  ] ERROR 2024/01/24 10:39:40 vehicle status: json: cannot unmarshal number 77.0 into Go struct field Energy.Energy.Level of type int
[main  ] WARN 2024/01/24 10:39:42 vehicle status: >1 matches, giving up
[lp-10 ] WARN 2024/01/24 10:39:42 charger logic error: disabled but charging
[lp-11 ] INFO 2024/01/24 10:40:06 car connected
[lp-11 ] INFO 2024/01/24 10:40:06 start charging ->
[main  ] ERROR 2024/01/24 10:40:07 vehicle status: json: cannot unmarshal number 77.0 into Go struct field Energy.Energy.Level of type int
[main  ] WARN 2024/01/24 10:40:09 vehicle status: >1 matches, giving up
[lp-11 ] WARN 2024/01/24 10:40:09 charger logic error: disabled but charging
[lp-12 ] INFO 2024/01/24 10:40:36 car disconnected
[main  ] ERROR 2024/01/24 10:41:46 vehicle status: Get "https://emea.bff.cariad.digital/vehicle/v1/vehicles/WVWZZZAA6PD902566/selectivestatus?jobs=charging,fuelStatus,climatisation,measurements": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
[main  ] ERROR 2024/01/24 10:41:46 vehicle status: json: cannot unmarshal number 77.0 into Go struct field Energy.Energy.Level of type int
[main  ] WARN 2024/01/24 10:41:48 vehicle status: >1 matches, giving up
[lp-2  ] WARN 2024/01/24 10:41:48 charger logic error: current mismatch (got 32A, expected 16A)
```

### What type of operating system are you running?

Linux

### Version

0.123.9

andig, 2024-01-24:
Already fixed in master

closed 2024-01-24 by andig

PatriotIT, 2024-01-24:
Kannst du mir noch kurz den Hinweis geben, wann der Fix etwa released wird? 😄