evcc-io/evcc #11840: Citroen SOC Wert wird nicht dargestellt

Hammond245, 2024-01-24:
### Describe the bug

Guten Abend,

seit gestern Abend wird der SOC Wert nicht mehr in der Leiste angezeigt. Im Logfile steht der Wert drin aber er wird nicht mehr in die Statusanzeige übernommen. Folgende Info steht dazu im Log File vom Docker Container

_2024/01/24 16:34:28,stdout,[lp-1  ] ERROR 2024/01/24 16:34:28 vehicle odometer: json: cannot unmarshal number 53.0 into Go struct field Energy.Energy.Level of type int_

Hat noch jemand dieses Problem mit einem Citroen oder einem anderen PSA Fahrzeug?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
#uri: 0.0.0.0:7070 # uri for ui
interval: 10s # control cycle interval

log: info

meters:
- type: template
  template: kostal-plenticore
  usage: pv 
  modbus: tcpip 
  name: pv1 
  id: 71 
  host: 192.168.0.199 
  port: 1502
- type: template
  template: kostal-plenticore
  usage: battery 
  modbus: tcpip 
  name: battery2 
  id: 71 
  host: 192.168.0.199 
  port: 1502
  capacity: 10 # optional Batterie Kapazität (kWh)
- type: template
  template: kostal-ksem-inverter
  usage: grid 
  modbus: tcpip 
  name: grid3 
  id: 71 
  host: 192.168.0.199 
  port: 1502
#- type: template
#  name: WP
#  template: shelly-1pm
#  usage: aux
#  host: 192.168.0.162
  
chargers:
- type: template
  template: keba
  host: 192.168.0.232
  name: wallbox5
  #timeout: 30

vehicles:
- type: template
  template: citroen
  title: E-C4 
  user: xxx
  password: xxx
  #vin: xxx
  capacity: 46.3
  mode: pv
  minCurrent: 3
  maxCurrent: 16
  name: ev4

loadpoints:
- title: Carport
  charger: wallbox5
  vehicle: ev4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16 # War 14

tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price
    type: fixed
    price: 0.336 # EUR/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0743 # EUR/kWh
  co2:
    type: grünstromindex
    zip: 78532
site:
  title: Mein Zuhause
  meters:
    grid: grid3
    pv:
    - pv1
    battery:
    - battery2
  residualPower: -200 # 200 W Netzeinspeisung verbleiben im PV Modus (negativer Wert lässt Bezug zu)
  maxGridSupplyWhileBatteryCharging: 200 # Netzbezug bei Batterieladung Haus bis 200W möglich
```

### Log details

```text
2024/01/24 16:34:28,stdout,[lp-1  ] ERROR 2024/01/24 16:34:28 vehicle odometer: json: cannot unmarshal number 53.0 into Go struct field Energy.Energy.Level of type int
```

### What type of operating system are you running?

Docker container

### Version

0.123.9

andig, 2024-01-24:
Fixed in nightly

closed 2024-01-24 by andig

Hammond245, 2024-01-24:
Danke für die schnelle Antwort