evcc-io/evcc #11846: Tesla Wall charger (TWC3) support via vehicle-command API for Tesla vehicles

yipikaye, 2024-01-25:
**Is your feature request related to a problem? Please describe.**
The introduction of the Tesla vehicle-command SDK has changed the APIs for Tesla vehicle management. The TWC3 charger cannot be controled via EVCC and must be managed via changing the charging parameters on the Tesla vehicle via the SDK. The new SDK requires additional Public/Private key pairs and authority from the user for EVCC to manage the vehicle. 

**Describe the solution you'd like**
The new vehicle-command SDK is only implemented in a read-only implementation. Read-write is needed to re-indroduce the EVCC functionallity that existed with the old API. The API gateway is now region specific so additional checks are needed to use the correct gateway:  You can query the users region and appropriate fleet-api URL so you should be able to dynamically assign it
https://developer.tesla.com/docs/fleet-api?shell#region

**Describe alternatives you've considered**
Other projects are also tackeling this same challenge. To date there are implementations by;
- https://chargehq.net/
- https://github.com/pkuehnel/TeslaSolarCharger
- https://www.tessie.com/

**Additional context**
Thank you

vtml, 2024-01-25:
It will be good if this is implemented to control charging, as the charging is then controlled directly on the vehicle itself. That means any charger  (including mobile chargers) can work with a Tesla vehicle, very much like Charge HQ.

andig, 2024-01-25:
Couple of things:

1. Make sure that https://github.com/evcc-io/evcc/pull/10802 works for you, i.e. you can retrieve soc.
2. Get a Tesla developer account- follow steps at https://github.com/teslamotors/vehicle-command and note https://github.com/teslamotors/vehicle-command/issues/111
3. Any insight about other self-hosted solutions is welcome. Key issue here is, that controlling the vehicle needs a private key installed on the vehicle. evcc does have a private key that *can* be installed but we obviously not distribute that.
4. When 1+2 are completed we can look into implementation. Depends on https://github.com/teslamotors/vehicle-command/issues/29.

yipikaye, 2024-01-25:
Hi @andig, 

I'll follow up on these points. Fully agree not to distribute private key, that would be silly. 
A thought, when users are generating their keys on https://tesla.evcc.io/ would it be possible for EVCC servers to push the private key to the vehicle and generate a unique public key for each user? 
I don't believe EVCC users would be able to retrieve the private key from their Tesla's.

FuR1u5, 2024-01-26:
Hi @andig 

Wenn ihr jemanden zum testen braucht, könnt ihr euch gerne melden. Mir stehen zwei evcc Instanzen mit jeweils einem TWC und einem Tesla zur Verfügung.

andig, 2024-01-26:
Problem is that in order to send vehicle commands, one must have the private key corresponding to the public key installed on the vehicle. That private key is a **SECRET**. One way of not distributing the secret might be running a vehicle command proxy (see https://github.com/teslamotors/vehicle-command/tree/main/cmd/tesla-http-proxy). Not sure if thats doable- needs more investigation.

yipikaye, 2024-01-26:
This sounds promising, the app would run the proxy or is it something EVCC needs to host?

infinus, 2024-01-27:
@andig confirmed #1 is now working in US and can retrieve SOC with latest nightly. @yipikaye can you try and see if it works for you too?

yipikaye, 2024-01-27:
@infinus yes I plan to, I'm away for a few days and for some reason it seems I can't ssh into home 😔
When is it going into the stable branch, I'll see if I can get it done before then

FuR1u5, 2024-01-27:
> Problem is that in order to send vehicle commands, one must have the private key corresponding to the public key installed on the vehicle. That private key is a **SECRET**. One way of not distributing the secret might be running a vehicle command proxy (see https://github.com/teslamotors/vehicle-command/tree/main/cmd/tesla-http-proxy).
> 
> Not sure if thats doable- needs more investigation:
> 
> > Do not listen on a network interface without adding client authentication. Unauthorized clients may
> > be used to create excessive traffic from your IP address to Tesla's servers, which Tesla may respond
> > to by rate limiting or blocking your connections.
> 
> ...which would bring us to client certificates.

Sounds great, can't wait to test it :)

manuelmrmorgado, 2024-01-30:
@andig May I suggest a creation of Template based on Tessie API? https://developer.tessie.com/docs/about
They provide their users an API that I guess works the same way Tesla API used to work.

Thanks! ;)

infinus, 2024-01-30:
@manuelmrmorgado 

Tessie has private servers and can host the client secret which is needed and should not be shared. EVCC is open sourced and usually run at home so they have can’t protect those like Tessie can. They are looking at setting up a proxy but that it a work in progress. It is much easier for projects like Tessie, teslafi, or Optiwatt to introduce these changes than open source projects like EVCC or Teslamate. It’s trying to find a delicate balance between ease of use, privacy, and security.

yipikaye, 2024-01-30:
This proxy seems promising!
It's great to see the improvements and the active team. This is what open source is all about!

andig, 2024-01-30:
Feel free to open a PR for Tessi integration.

haroldboom, 2024-02-01:
Hi @andig, would this issue be why I am getting `[main  ] ERROR 2024/02/01 19:28:58 creating vehicle ev3 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla-command': oauth2: "login_required" "Login required" "https://auth.tesla.com/error/reference/a15ecfb6-0272-46a6-b3d8-af10870b8af4-1706776138785"` after switching to tesla-command and 124.0?

andig, 2024-02-01:
@haroldboom this Issue is about TWC. Please don't cross-post. Tesla integration is confirmed working. Feel free to share login information with info@evcc.io for further diagnosis.

andig, 2024-02-01:
@manuelmrmorgado re Tessi that would mean that the old Tesla integration would continue to work by swapping the URL. Since we're using https://github.com/bogosj/tesla under the hood, swapping URI and adding API key would need to be done there. I suggest to open an issue with bogosj/tesla.

user1272349087234, 2024-02-01:
I upgraded to 0.124.0 Version now and changed the config to new api. Also added both tokens to config. Car is now recognizing from evcc. But EVCC is not able to controll, start or stop charging.

Likemike24, 2024-02-01:
> @haroldboom this Issue is about TWC. Please don't cross-post. Tesla integration is confirmed working. Feel free to share login information with [info@evcc.io](mailto:info@evcc.io) for further diagnosis.

I´m confused. If I understand it correct, the Tesla integration with reading values is now working again with the new API, but we cannot send any commands to the car atm, which can be done via a proxy integration?

As far as for TWC3: Didn´t the car control the whole charging in the past, since the TWC3 is not controlable? Has anything changed here?

haroldboom, 2024-02-04:
Hi @andig is the proxy in the nightly or still in draft stage?

andig, 2024-02-04:
See PR- it is nowhere yet.

mucki12, 2024-02-06:
Don't know if it helps, but here they are also looking at a solution with a proxy:
https://github.com/llamafilm/tesla-http-proxy-addon?tab=readme-ov-file

Likemike24, 2024-02-07:
I also want to add, that Tesla Solar Charger is working with flawlessly with the new API. I´m using it as a temporary solution now until EVCC catches up.

andig, 2024-02-09:
Does anyone use Tessie and wants to share his credentials with info@evcc.io? That could be a simple drop-in replacement for controlling the vehicle. At least a workaround until we can build our own Tesla Proxy.

andig, 2024-02-10:
I gave Tessie a quick try (thanks Brin). We should be able to add Tessie, but its really its own integration using a different API structure from Tesla. See https://github.com/evcc-io/evcc/issues/12183.

mucki12, 2024-02-10:
@andig 
First of all, thank you for your efforts! In the context of the ideas with Tessie, I am currently thinking about trying this out.
One question: In addition to controlling the Tesla, would the vehicle's other data also be retrieved directly via Tessie or would the existing evcc Tesla implementation would do this job?

andig, 2024-02-10:
Its the same thing

andig, 2024-02-12:
Short update: we're continuing to investigate running our own Tesla Proxy. Apart from testing, we still need to figure how  to handle token refreshes. This depends on resolving any lingering token refresh issues that couldn't be pin-pointed yet.

yipikaye, 2024-02-14:
I see the PR is now closed. Does this mean the next release will have the functionality with some limitations around token refreshes or will you hold off until the token refresh is solved?

haroldboom, 2024-02-15:
> I see the PR is now closed. Does this mean the next release will have the functionality with some limitations around token refreshes or will you hold off until the token refresh is solved?

https://github.com/evcc-io/evcc/pull/12174 Still open

yipikaye, 2024-02-15:
I've got to stop drinking 🍺

andig, 2024-02-15:
Good news: we've just merged a new `tesla-proxy` template which will talk via evcc's Tesla API Proxy to the Tesla Fleet API. This requires:

- Sponsor token for proxy authentication
- Tesla tokens for Fleet API access
- evcc virtual private key installed on vehicle

If you're missing the last step just create new tokens, the virtual key is visible after token creation. Let's try it!

Nightly will be available in 30 minutes.

closed 2024-02-15 by andig

FuR1u5, 2024-02-15:
> Good news: we've just merged a new `tesla-proxy` template which will talk via evcc's Tesla API Proxy to the Tesla Fleet API. This requires:
> 
> * Sponsor token for proxy authentication
> * Tesla tokens for Fleet API access
> * evcc virtual private key installed on vehicle
> 
> If you're missing the last step just create new tokens, the virtual key is visible after token creation. Let's try it!
> 
> `tesla-proxy` and `tesla-command` are allowed to share tokens. evcc will manage lifecycle internally. Tokens **MUST NOT** be shared with other applications since token refresh will invalidate other applications token.
> 
> Nightly will be available in 30 minutes.

Perfect, thanks :)

How can i install the private key on the vehicle? I redo the last step but i didn't get a virtual key...

andig, 2024-02-15:
Seems I need to rebuild the proxy, too. Will shortly.

TheJDJ13, 2024-02-15:
I'm ready to test when it is ready. Model 3/Enphase Solar/TWC3.

I do not have a Powerwall or any other ESS.

haroldboom, 2024-02-15:
> Good news: we've just merged a new `tesla-proxy` template which will talk via evcc's Tesla API Proxy to the Tesla Fleet API. This requires:
>
> * Sponsor token for proxy authentication
> * Tesla tokens [...]

Hey Andy, so after I get the new nightly do I recreate the token via Tesla.evcc.io and I should get a prompt about a private key? I tried this and nothing after the tokens appear

oysteing, 2024-02-15:
I tried with evcc/evcc:nightly.

My config:
```
vehicles:
  - name: tesla
    type: template
    template: tesla-proxy
    accessToken: [token]
    refreshToken: [token]
    capacity: 60
```
I seem to get 500 Server when trying to control the vehicle. I didn't get the part about virtual key, and didn't see anything after creating the tokens.

> [lp-1  ] INFO 2024/02/15 22:45:19 car connected
[lp-1  ] DEBUG 2024/02/15 22:45:19 pv timer elapse
[lp-1  ] DEBUG 2024/02/15 22:45:19 pv timer inactive
[lp-1  ] INFO 2024/02/15 22:45:19 start charging ->
[lp-1  ] DEBUG 2024/02/15 22:45:19 wake-up timer: stop
[lp-1  ] DEBUG 2024/02/15 22:45:20 vehicle status: C
[lp-1  ] DEBUG 2024/02/15 22:45:20 vehicle soc: 41%
[lp-1  ] DEBUG 2024/02/15 22:45:20 vehicle soc limit: 50%
[lp-1  ] DEBUG 2024/02/15 22:45:20 vehicle range: 168km
[lp-1  ] WARN 2024/02/15 22:45:20 charger out of sync: expected disabled, got enabled
[lp-1  ] DEBUG 2024/02/15 22:45:20 forced charging at vehicle soc 41% (< 50% min soc)
[lp-1  ] ERROR 2024/02/15 22:45:20 max charge current 16A: 500 Internal Server Error

andig, 2024-02-15:
> Seems I need to rebuild the proxy, too. Will shortly.

Done- should offer the virtual key now.

> I seem to get 500 Server when trying to control the vehicle

Virtual key installed?! Likely not. Trace log for Tesla would help.

TheJDJ13, 2024-02-15:
It's working!

Thanks andig and everyone.

I want to charge on solar, but also I am stuck pre-PTO so it's extra important for me.

yipikaye, 2024-02-15:
I probably should have asked this a while ago, are there nightly deb files so I don't have to keep doing it manually?

haroldboom, 2024-02-15:
> > Seems I need to rebuild the proxy, too. Will shortly.
> 
> Done- should offer the virtual key now.
> 
> > I seem to get 500 Server when trying to control the vehicle
> 
> Virtual key installed?! Likely not. Trace log for Tesla would help.

Can confirm tesla.evcc.io virtual key is working and EVCC is working with my TWC3. Will test today and post back if something goes wrong

vtml, 2024-02-15:
Does this mean the charging rate is controlled at the Tesla vehicle, instead of the EVSE ? Does this mean that this approach is EVSE agnostic ?

yipikaye, 2024-02-15:
I get a 500 server error but I have generated new tokens... Do i need to revoke the existing and start again?

TheJDJ13, 2024-02-15:
> I get a 500 server error but I have generated new tokens... Do i need to revoke the existing and start again?

I did not need to revoke existing.

Did you continue below and add the virtual key and use the QR to add it in the Tesla app?

yipikaye, 2024-02-15:
No i missed that last step as I cannot read German :) I updated the above comment. 
It is working well!

TheJDJ13, 2024-02-15:
> No i missed that last step as I cannot read German :) I updated the above comment. It is working well!

I had to have my browser translate the page to English so I could read that part.

yipikaye, 2024-02-15:
Is anyone else getting server errors from the proxy?
` ERROR 2024/02/16 10:22:11 charger disable: Post "https://tesla.evcc.io/api/1/vehicles/LRWYHCEK5PC906710/command/charge_stop": read tcp [2406:2d40:406f:d900:9e8c:167c:3ba4:ba50]:60448->[2a09:8280:1::69:9409]:443: read: connection reset by peer`

TheJDJ13, 2024-02-15:
> Is anyone else getting server errors from the proxy? ` ERROR 2024/02/16 10:22:11 charger disable: Post "https://tesla.evcc.io/api/1/vehicles/LRWYHCEK5PC906710/command/charge_stop": read tcp [2406:2d40:406f:d900:9e8c:167c:3ba4:ba50]:60448->[2a09:8280:1::69:9409]:443: read: connection reset by peer`

I am not seeing that or any other errors.

yipikaye, 2024-02-15:
Hmm something has gone wrong. It was working for a short while, now:
`ERROR 2024/02/16 10:38:42 charger enable: 500 Internal Server Error`

yipikaye, 2024-02-15:
It appears the car went to sleep and with the proxy it doesn't wake. I restarted the EVCC service and got the following

`Feb 16 10:43:28 raspberrypi evcc[11093]: [main  ] ERROR 2024/02/16 10:43:28 vehicle status: asleep
Feb 16 10:43:28 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:43:28 max charge current 5A: vehicle not capable of current control
Feb 16 10:43:30 raspberrypi evcc[11093]: [lp-1  ] INFO 2024/02/16 10:43:30 vehicle updated: unknown -> Pikachu
`

yipikaye, 2024-02-15:
```
Yes, if I wake the car via the moble app then the charging control starts working properly again with no intervention on EVCC from me. 

So we have a minor wake the vehicle issue :) 

```Feb 16 10:43:30 raspberrypi evcc[11093]: [lp-1  ] INFO 2024/02/16 10:43:30 vehicle updated: unknown -> Pikachu
Feb 16 10:43:42 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:43:42 vehicle odometer: asleep
Feb 16 10:43:42 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:43:42 vehicle soc: asleep
Feb 16 10:43:45 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:43:45 max charge current 5A: 500 Internal Server Error
Feb 16 10:44:00 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:44:00 max charge current 5A: 500 Internal Server Error
Feb 16 10:44:15 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:44:15 max charge current 5A: 500 Internal Server Error
Feb 16 10:44:30 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:44:30 max charge current 5A: 500 Internal Server Error
Feb 16 10:44:44 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:44:44 max charge current 5A: 500 Internal Server Error
Feb 16 10:44:59 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:44:59 max charge current 5A: 500 Internal Server Error
Feb 16 10:45:15 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:45:15 max charge current 5A: 500 Internal Server Error
Feb 16 10:45:42 raspberrypi evcc[11093]: [lp-1  ] INFO 2024/02/16 10:45:42 start charging ->
Feb 16 10:46:27 raspberrypi evcc[11093]: [lp-1  ] INFO 2024/02/16 10:46:27 stop charging <-
```

haroldboom, 2024-02-16:
> Yes, if I wake the car via the moble app then the charging control starts working properly again with no intervention on EVCC from me.
> 
> So we have a minor wake the vehicle issue :)
> 
> ```
> Feb 16 10:43:42 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:43:42 vehicle odometer: asleep
> Feb 16 10:43:42 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:43:42 vehicle soc: asleep
> Feb 16 10:43:45 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:43:45 max charge current 5A: 500 Internal Server Error
> Feb 16 10:44:00 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:44:00 max charge current 5A: 500 Internal Server Error
> Feb 16 10:44:15 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:44:15 max charge current 5A: 500 Internal Server Error
> Feb 16 10:44:30 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:44:30 max charge current 5A: 500 Internal Server Error
> Feb 16 10:44:44 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:44:44 max charge current 5A: 500 Internal Server Error
> Feb 16 10:44:59 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:44:59 max charge current 5A: 500 Internal Server Error
> Feb 16 10:45:15 raspberrypi evcc[11093]: [lp-1  ] ERROR 2024/02/16 10:45:15 max charge current 5A: 500 Internal Server Error
> Feb 16 10:45:42 raspberrypi evcc[11093]: [lp-1  ] INFO 2024/02/16 10:45:42 start charging ->
> Feb 16 10:46:27 raspberrypi evcc[11093]: [lp-1  ] INFO 2024/02/16 10:46:27 stop charging <-
> ```

Hi all, I also have the same issue, EVCC cant wake the vehicle to charge. If I wake manually with the app it will start. Also getting the
evcc[632]: [lp-1  ] ERROR 2024/02/16 00:20:02 max charge current 5A: 500 Internal Server Error when trying to wake

TheJDJ13, 2024-02-16:
No errors for me so far, but I will test this.

As soon as I run out of sunlight today, I will set my car to higher than the current SOC. I'll leave it alone until tomorrow and see if it wakes up to charge when there is enough solar. I generally get enough sun by 10AM (PST).

haroldboom, 2024-02-16:
> No errors for me so far, but I will test this.
> 
> As soon as I run out of sunlight today, I will set my car to higher than the current SOC. I'll leave it alone until tomorrow and see if it wakes up to charge when there is enough solar. I generally get enough sun by 10AM (PST).

I just let the car go to sleep and switched EVCC to fast or minpv and that forces the error for me

oysteing, 2024-02-16:
> > Seems I need to rebuild the proxy, too. Will shortly.
> 
> Done- should offer the virtual key now.
> 
> > I seem to get 500 Server when trying to control the vehicle
> 
> Virtual key installed?! Likely not. Trace log for Tesla would help.

I can also confirm that vehicle control works after installing the virtual key. Thanks for the great work!

yipikaye, 2024-02-16:
> I just let the car go to sleep and switched EVCC to fast or minpv and that forces the error for me

Yes I can replicate the same too.

TheJDJ13, 2024-02-16:
> > I just let the car go to sleep and switched EVCC to fast or minpv and that forces the error for me
> 
> Yes I can replicate the same too.

OK I can replicate it as well. That is not something I will do normally so I am going to continue my planned test, but I expect I will see that same error.

haroldboom, 2024-02-16:
I think the vehicle is also going to sleep while charging and EVCC cant change the current, see below
`Feb 16 04:05:07 testrig evcc[630]: [lp-1  ] ERROR 2024/02/16 04:05:07 max charge current 27A: Post "https://tesla.evcc.io/api/1/vehicles/LRWYHCFS3NC428511/command/set_charging_amps": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
Feb 16 04:05:12 testrig evcc[630]: [site  ] DEBUG 2024/02/16 04:05:12 ----
Feb 16 04:05:12 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:12 charge power: 391W
Feb 16 04:05:12 testrig evcc[630]: [site  ] DEBUG 2024/02/16 04:05:12 pv power: 8030W
Feb 16 04:05:12 testrig evcc[630]: [site  ] DEBUG 2024/02/16 04:05:12 grid power: -4388W
Feb 16 04:05:12 testrig evcc[630]: [site  ] DEBUG 2024/02/16 04:05:12 site power: -4388W
Feb 16 04:05:12 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:12 charge voltages: [244 1.5 0]V
Feb 16 04:05:12 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:12 detected connected phases: 1p
Feb 16 04:05:12 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:12 charge currents: [1.6 0 0.2]A
Feb 16 04:05:12 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:12 detected active phases: 1p
Feb 16 04:05:12 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:12 charger status: C
Feb 16 04:05:12 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:12 soc estimated: 72.38% (vehicle: 72.00%)
Feb 16 04:05:12 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:12 vehicle soc: 72%
Feb 16 04:05:12 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:12 vehicle soc limit: 100%
Feb 16 04:05:12 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:12 vehicle range: 301km
Feb 16 04:05:12 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:12 pv charge current: 22.7A = 3.6A + 19.1A (-4388W @ 1p)
Feb 16 04:05:22 testrig evcc[630]: [lp-1  ] ERROR 2024/02/16 04:05:22 max charge current 22A: Post "https://tesla.evcc.io/api/1/vehicles/LRWYHCFS3NC428511/command/set_charging_amps": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
Feb 16 04:05:27 testrig evcc[630]: [site  ] DEBUG 2024/02/16 04:05:27 ----
Feb 16 04:05:27 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:27 charge power: 756W
Feb 16 04:05:27 testrig evcc[630]: [site  ] DEBUG 2024/02/16 04:05:27 pv power: 8143W
Feb 16 04:05:27 testrig evcc[630]: [site  ] DEBUG 2024/02/16 04:05:27 grid power: -5033W
Feb 16 04:05:27 testrig evcc[630]: [site  ] DEBUG 2024/02/16 04:05:27 site power: -5033W
Feb 16 04:05:27 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:27 charge voltages: [244 1.5 0]V
Feb 16 04:05:27 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:27 detected connected phases: 1p
Feb 16 04:05:27 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:27 charge currents: [3.1 0 0.2]A
Feb 16 04:05:27 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:27 detected active phases: 1p
Feb 16 04:05:27 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:27 charger status: C
Feb 16 04:05:27 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:27 soc estimated: 72.39% (vehicle: 72.00%)
Feb 16 04:05:27 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:27 vehicle soc: 72%
Feb 16 04:05:27 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:27 vehicle soc limit: 100%
Feb 16 04:05:27 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:27 vehicle range: 301km
Feb 16 04:05:27 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:27 pv charge current: 27A = 5.1A + 21.9A (-5033W @ 1p)
Feb 16 04:05:37 testrig evcc[630]: [lp-1  ] ERROR 2024/02/16 04:05:37 max charge current 26A: Post "https://tesla.evcc.io/api/1/vehicles/LRWYHCFS3NC428511/command/set_charging_amps": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
Feb 16 04:05:42 testrig evcc[630]: [site  ] DEBUG 2024/02/16 04:05:42 ----
Feb 16 04:05:42 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:42 charge power: 1193W
Feb 16 04:05:42 testrig evcc[630]: [site  ] DEBUG 2024/02/16 04:05:42 pv power: 8156W
Feb 16 04:05:42 testrig evcc[630]: [site  ] DEBUG 2024/02/16 04:05:42 grid power: -4579W
Feb 16 04:05:42 testrig evcc[630]: [site  ] DEBUG 2024/02/16 04:05:42 site power: -4579W
Feb 16 04:05:42 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:42 charge voltages: [244 1.5 0]V
Feb 16 04:05:42 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:42 detected connected phases: 1p
Feb 16 04:05:42 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:42 charge currents: [4.9 0 0.2]A
Feb 16 04:05:42 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:42 detected active phases: 1p
Feb 16 04:05:42 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:42 charger status: C
Feb 16 04:05:42 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:42 soc estimated: 72.39% (vehicle: 72.00%)
Feb 16 04:05:42 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:42 vehicle soc: 72%
Feb 16 04:05:42 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:42 vehicle soc limit: 100%
Feb 16 04:05:42 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:42 vehicle range: 301km
Feb 16 04:05:42 testrig evcc[630]: [lp-1  ] DEBUG 2024/02/16 04:05:42 pv charge current: 26.8A = 6.9A + 19.9A (-4579W @ 1p)
Feb 16 04:05:52 testrig evcc[630]: [lp-1  ] ERROR 2024/02/16 04:05:52 max charge current 26A: Post "https://tesla.evcc.io/api/1/vehicles/LRWYHCFS3NC428511/command/set_charging_amps": net/http: request canceled (Client.Timeout exceeded while awaiting headers)
`

Stacykid, 2024-02-16:
I am apparently also on the new API because it stopped working recently and I have been following this issue.

I have now done everything exactly according to the instructions but my vehicle is only specified as a guest vehicle.

Can anyone help me?

yipikaye, 2024-02-16:
Did you install the nightly build, setup the car again with the new template and new tokens?

Stacykid, 2024-02-16:
> Did you install the nightly build, setup the car again with the new template and new tokens?

I have done the following:
-evcc update, but only the one that is available for everyone
-new token for Tesla with all approvals
-added sponsoring (is also recognised)

Could it be because I only did a normal update to 124.2 and the unstable nightly is not integrated?

[11 more replies]