evcc-io/evcc #11854: Tesla not working anymore

miniluma, 2024-01-25:
### Describe the bug

I see my Tesla Model Y as offline.
Trace states "Endpoint is only available on fleetapi. Visit https://developer.tesla.com/docs for more info"
Regenerated Tokens to make sure this is not the problem.
Also created a second config via Webinterface. Same result.

### Steps to reproduce

1. Configure Tesla and Log level to Trace
2. Start EVCC
3. Read Log
...

### Configuration details

```yaml
vehicles:
- name: ev3
  type: template
  template: tesla
  title: Model Y  
  accessToken: ...
  refreshToken: ...
```

### Log details

```text
an 25 12:20:09 pi4 evcc[79184]: [tesla ] TRACE 2024/01/25 12:20:09 POST https://auth.tesla.com/oauth2/v3/token
Jan 25 12:20:10 pi4 evcc[79184]: [tesla ] TRACE 2024/01/25 12:20:10 client_id=ownerapi&grant_type=refresh_token&refresh_token=***
Jan 25 12:20:10 pi4 evcc[79184]: --
Jan 25 12:20:10 pi4 evcc[79184]: {"access_token":"eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCIsImtpZCI6IkpJcFJCQWlsQ0V1cXZRRjlwUHRtZkhKWnNjYyJ9.eyJpc3MiOiJodHRwczovL2F1dGgudGVzbGEuY29tL29hdXRoMi92MyIsImF6cCI6Im93bmVyYXBpIiwic3ViIjoiMWI3ZDM5ZDMtNGQzNi00NjA3LWE0ZTYtNTJhNTM2NmU3YmJmIiwiYXVkIjpbImh0dHBzOi8vb3duZXItYXBpLnRlc2xhbW90b3JzLmNvbS8iLCJodHRwczovL2F1dGgudGVzbGEuY29tL29hdXRoMi92My91c2VyaW5mbyJdLCJzY3AiOltdLCJhbXIiOltdLCJleHAiOjE3MDYyMTA0MTAsImlhdCI6MTcwNjE4MTYxMCwib3VfY29kZSI6IkVVIiwibG9jYWxlIjoiZGUtREUifQ.e08SMpct4_qqfEug4mfwZikzM_NxEihZOYJoP6L7NPTxiZQ50LMdyYxaY6mvp-fgPtyo7dmC9R_tYQ5I5LEyFuWCt8WUmPBALeIeYJOItdlOC0vEpaBYK3aqHMe0Z5gS7fpQ09OCNB5dtW1Ekg2O242eWMPdPTyZdGq_fJps7xfgTP3V1mEw-Nk2yNNPzZ4KqSq020Fz2dUgUcNXl6tjHDbIH3BkerQJWkwOVmPjErXtfWcOLwIQFs9Lbyz0B26_jCkXdGjfhasffh_MgPgHeoxzKnyistM2NqueWRTjJ7ncTDgut1MITXm8rowiGKes--I-5GNwW0nw4OPHoqjn-A","id_token":"eyJhbGciOiJSUzI1NiIsInR5cCI6IkpXVCIsImtpZCI6IkpJcFJCQWlsQ0V1cXZRRjlwUHRtZkhKWnNjYyJ9.eyJpc3MiOiJodHRwczovL2F1dGgudGVzbGEuY29tL29hdXRoMi92MyIsImF1ZCI6Im93bmVyYXBpIiwic3ViIjoiMWI3ZDM5ZDMtNGQzNi00NjA3LWE0ZTYtNTJhNTM2NmU3YmJmIiwiZXhwIjoxNzA2MjEwNDEwLCJpYXQiOjE3MDYxODE2MTAsImFtciI6W10sInVwZGF0ZWRfYXQiOjE2NjExNzMxMDh9.iLxo3mX8_nI613lEogf9S8E1USh5JDx_S0ZfXc32FYPGckE2wvHO9W-jb9EZMof2GlLxrPDTKecLxmFp0ad41cAvh9Goe23rND-k0HPrFjMQqTNKGgkwgrKn65PkD9WQIQOTa2SnkRsh2uVx4maGngvtjDdf0KKUv7LX9G2B9zMyfpuG9rvx4SelMoSPJBahkw3uqNMlMHf8LwpVa90trTxKmM2baPT7jDa9vhE7Q1nNKWWNzfYaS2NkJSAxJ9D2xtQWKBNXGykywppb3lfBSxtzaeRb8KK9sQFWMHYAwxVWXy9-dNTdMftMtJ1367deF7-Z8LRvv4HAAIDMhr0i5Q","expires_in":28800,"token_type":"Bearer"}
Jan 25 12:20:10 pi4 evcc[79184]: [tesla ] TRACE 2024/01/25 12:20:10 GET https://owner-api.teslamotors.com/api/1/vehicles
Jan 25 12:20:11 pi4 evcc[79184]: [tesla ] TRACE 2024/01/25 12:20:11 {"response":null,"error":"Endpoint is only available on fleetapi. Visit https://developer.tesla.com/docs for more info","error_description":""}
Jan 25 12:20:11 pi4 evcc[79184]: [main  ] ERROR 2024/01/25 12:20:11 creating vehicle ev3 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': cannot get vehicles: 412 Precondition Failed
```

### What type of operating system are you running?

Linux

### Version

0.123.9

miniluma, 2024-01-25:
I have the same issue in my FHEM setup, so it could be an API change on Tesla side

miniluma, 2024-01-25:
see here: https://github.com/teslamate-org/teslamate/issues/3629
and here: https://github.com/teslamate-org/teslamate/pull/3630

andig, 2024-01-25:
Solved: https://github.com/evcc-io/evcc/pull/10802

closed 2024-01-25 by andig