evcc-io/evcc #11857: tesla still not working anymore

FuR1u5, 2024-01-25:
### Describe the bug

.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'tesla': cannot get vehicles: 412 Precondition Failed
Capacity: 0.0kWh
Features: [Offline]

### Steps to reproduce

evcc has been updated to the latest version, tokens have been regenerated and inserted. Do I have to keep the permissions for tesla.evcc.io or can I revoke them? Which checkboxes have to be set for this?

### Configuration details

```yaml
vehicles:
- type: template
  template: tesla-command
  title: Shadow
  accessToken: eyJhbGciOiJSUzI1NiIsInR5cCI6Ik....
  refreshToken: eyJhbGciOiJSUzI1NiIsInR5cCI6I....
  vin: LRWY...
  name: ev2
```

### Log details

```text
[main  ] INFO 2024/01/25 14:15:15 evcc 0.123.9 (28f74708)
[main  ] INFO 2024/01/25 14:15:15 using config file: /home/PiAdmin/evcc.yaml
[main  ] INFO 2024/01/25 14:15:15 starting ui and api at :7070
[db    ] INFO 2024/01/25 14:15:15 using sqlite database: /home/PiAdmin/.evcc/evcc.db
[main  ] ERROR 2024/01/25 14:15:17 creating vehicle ev2 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': cannot get vehicles: 412 Precondition Failed
[main  ] FATAL 2024/01/25 14:15:18 failed configuring loadpoints: failed configuring loadpoint: 1 error(s) decoding:

* '' has invalid keys: resetondisconnect
[main  ] FATAL 2024/01/25 14:15:18 will attempt restart in: 5m0s
[server] FATAL 2024/01/25 14:15:18 unlinkat /tmp/evcc: operation not permitted
```

### What type of operating system are you running?

Linux

### Version

0.123.9 (28f74708)

Lako-Unbekannt, 2024-01-25:
Hi, have you seen [this discussion?](https://github.com/evcc-io/evcc/discussions/11855#discussion-6131331) If not it should answer the question.

andig, 2024-01-25:
@FuR1u5 aussagekräftige Issuetitel helfen das Problem schnell zu verstehen ;)

andig, 2024-01-25:
Oder bitte mal probieren:

```
curl --header 'Content-Type: application/json' \
  --header "Authorization: Bearer $TESLA_API_TOKEN" \
  'https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/users/region' 
```

FuR1u5, 2024-01-25:
> @FuR1u5 aussagekräftige Issuetitel helfen das Problem schnell zu verstehen ;) Bitte Deine Config mit [info@evcc.io](mailto:info@evcc.io) teilen.

Hi andig, Mail ist raus :)

FuR1u5, 2024-01-25:
> Oder bitte mal probieren:
> 
> ```
> curl --header 'Content-Type: application/json' \
>   --header "Authorization: Bearer $TESLA_API_TOKEN" \
>   'https://fleet-api.prd.eu.vn.cloud.tesla.com/api/1/users/region' 
> ```

Wo müsste ich das eintragen? Ich habe den Token wie in deiner Anleitung über eure Seite erstellen lassen.

andig, 2024-01-25:
Scopes depend on https://github.com/teslamotors/vehicle-command/issues/63

FuR1u5, 2024-01-25:
Nachdem ich bei der Key Erstellung alles erlaubt habe, danke @andig und noch einmal alles neu eingerichtet habe, funktioniert die Abfrage nun wieder. Die Ladesteuerung klappt leider weiterhin nicht, aber da hoffe ich, dass ihr die Steuerung bald auch wieder einstellt.

closed 2024-01-25 by andig

andig, 2024-01-26:
Ja, alle. Wir werden das im Tokenprozess noch validieren. Leider macht Tesla das aktuell nicht klar: https://github.com/teslamotors/vehicle-command/issues/63

mailkluge, 2024-01-26:
Hi. Ich wollte es auch versuchen. Leider erhalte ich nach Auswahl aller Berechtigungen eine Fehlermeldung "insufficient Scopes". Habt ihr einen Tipp?

andig, 2024-01-26:
Alle scopes auswählen.

mailkluge, 2024-01-26:
@andig Danke - genau das habe ich getan. Konkret die 4 möglichen Scopes / Berechtigungen (Profilinfos, Fahrzeugdaten, Fahrzeugbefehle, Fahrzeugladeverwaltung). Beim redirect auf tesla.evcc.io erhalte ich die Fehlermeldung 
Tokenerzeugung fehlgeschlagen. Insufficient Scopes: Missing token scopes.

Ich bin bei dir - die Fehlermeldung besagt dies, aber ich wähle alles aus was geht. Irgendein Tipp oder Idee?

andig, 2024-01-27:
> Missing token scopes

Bitte Token revoken und neues erzeugen. Siehe https://github.com/evcc-io/evcc/issues/11880