evcc-io/evcc #11858: Bug: evcc lädt bei vollem Akku ohne PV

cg089, 2024-01-25:
### Describe the bug

Vorab: Ich hoffe, die Bugmeldung auf Deutsch ist okay, sonst schreibe ich sie gerne auf englisch.

Ich habe einen Hybdridwechselrichter mit Batteriespeicher. Diesen kann man auch in einen sog. USV-Modus konfigurieren. Der Speicher wird dabei auf 100% (von PV und Netzseite) gefüllt und erst entladen, wenn das Netz ausfällt.

Ich betreibe ihn derzeit im USV-Modus, d.h. der Batteriespeicher ist immer zu 100% voll. PV-Überschuss geht zuerst ins ins Haus + Auto und dann ins Netz.

In dieser Konfiguration hat aber mein Auto auch nachts, trotz "PV" Einstellung in evcc, mit ca. 1,7 kW dauerhaft geladen.

Erwartet:
evcc lädt in der Einstellung "PV" das Auto nur, wenn a) Batteriespeicher über konfiguriertem Limit und b) PV-Überschuss vorhanden ist.

Tatsächlich:
evcc lädt in der Einstellung "PV" das Auto auch dann, wenn der Batteriespeicher voll ist und keinerlei PV-Erzeugung stattfindet.

### Steps to reproduce

1. Batteriespeicher in dern UV-Modus setzen
2. evcc in den Modus PV setzen
3. Auto anschliessen
4. Schauen, was ohne PV passiert.

### Configuration details

```yaml
network:
  schema: http
  host: 182.168.1.2 # .local suffix announces the hostname on MDNS
  port: 7070

sponsortoken: xxx

log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  mqtt: debug
  cache: debug
  db: debug

interval: 35s

mqtt:
  broker: 192.168.1.2:1883
  topic: evcc 
  user: mqtt
  password: mqtt

site:
  title: Haus
  meters:
    grid: grid
    pv: pv
    battery: battery
  residualPower: 100 # additional household usage margin

chargers:
  - type: template
    template: openwb
    host: 192.168.1.3
#    connector: 1
    phases1p3p: true
    name: wallbox
    id: 1

vehicles:
  - name: Auto
    type: template
    template: teslamate
    title: Tesla
    phases: 3
    mode: pv
    minCurrent: 6
    maxCurrent: 16
    id: 1 
    capacity: 82

loadpoints:
  - title: OpenWB
    charger: wallbox
    mode: pv
    phases: 0
    maxcurrent: 16

meters:
  - name: grid
    type: custom
    power:
      source: mqtt
      broker: 192.168.1.2:1883
      topic: evcc/sensor/mtec_leistung_am_nvp_zahler
      user: mqtt
      password: mqtt
      scale: 1
      timeout: 30s
    energy: 
      source: mqtt
      broker: 192.168.1.2:1883
      topic: evcc/sensor/mtec_bezug
      user: mqtt
      password: mqtt
      scale: 1
      timeout: 30s
  - name: pv
    type: custom
    power:  
      source: mqtt
      broker: 192.168.1.2:1883
      topic: evcc/sensor/mtec_pv_erzeugung_gesamt
      user: mqtt
      password: mqtt
      scale: 1
      timeout: 30s
    energy: 
      source: mqtt
      broker: 192.168.1.2:1883
      topic: evcc/sensor/mtec_pv_erzeugung_kwh
      user: mqtt
      password: mqtt
      scale: 1
      timeout: 30s
  - name: battery
    type: custom
    power: 
      source: mqtt
      broker: 192.168.1.2:1883
      topic: evcc/sensor/mtec_lade_entladeleistung_der_batterie
      user: mqtt
      password: mqtt
      timeout: 30s
#      scale: -1
    soc: 
      source: mqtt
      broker: 192.168.1.2:1883
      topic: evcc/sensor/mtec_ladezustand_der_batterie
      user: mqtt
      password: mqtt
      timeout: 30s

tariffs:
  currency: EUR 
  grid:
    type: fixed
    price: 0.294

  feedin:
    type: fixed
    price: 0.08
```

### Log details

```text
None available
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.123.9

andig, 2024-01-25:
Incomplete issues are closed without further notice

cg089, 2024-01-25:
Super, da macht das Sponsoring und die Zeit, um Bugs zu melden, gleich doppelt Spaß! Danke!

andig, 2024-01-25:
Dein Kommentar ist völlig unangemessen. Da ohne Log keine Fehleranalyse möglich ist erübrigt sich die Diskussion. Im Issuetemplate steht klar drin was es braucht **UND** dass wir Issues ohne schließen. Also bitte nicht hinterher beschweren. Du gehst ja auch nicht zur Autowerkstatt mit dem Fahrrad und erwartest dass sie Deine defekte Zylinderkopfdichtung beim Daimler diagnostizieren...

cg089, 2024-01-25:
Ich finde die Art und Weise, kommentarlos einen Bugreport zu schließen für jemand, der sich Zeit nimmt, einen solchen zu melden, völlig unangemessen. Offenbar sind vier Worte dazu "kein Log, daher geschlossen", oder aber ein Hinweis, diese nachzuliefern, weil der Bug sonst geschlossen wird, zu viel gewesen - das ist bedauerlich. Denn auch eine Autowerkstatt wird mir nicht kommentarlos die Tür vor der Nase zuschlagen, ohne ein Wort zu sagen.

Offenbar gehst Du davon aus, dass ich mir etwa 5-10 Minuten Zeit nehme, einen Bugreport zu schreiben im Wissen, dass dieser ohnehin sofort wieder geschlossen wird, weil ich absichtlich Informationen vorenthalte. Diese Annahme entbehrt jeder Logik. Denn dann hätte ich mir die Zeit hierfür auch sparen können. Offensichtlich also ich also den "glasklaren" Hinweis entweder nicht gesehen oder nicht verstanden: Ich hielt nämlich Logs, die bei einem ganz normalen Ladevorgang entstehen, für entbehrlich. Ein kleiner Hinweis Deinerseits hierauf hätte mir, und auch dem Projekt, sicher mehr geholfen.

Letztlich sorgt dieser Umgang dafür, dass die Motivation für Unterstützung des Projekts - sei es finanziell oder durch Mitarbeit in Form von Fehlermeldungen - zumindest bei mir deutlich sinkt. Dies aber nur als zwischenmenschlicher Hinweis.

andig, 2024-01-25:
Ich kann mich nur wiederholen. Das stand bereits da als du den Bug aufgemacht hast. Was du aber tun kannst ist die notwendigen Daten zu ergänzen.

andig, 2024-01-25:
[image]

cg089, 2024-01-25:
Vielen Dank für den Hinweis. Das ist aus Programmierer-Sicht sicherlich richtig (ich bin kein Programmierer), meine Kritik bezog sich auf das Zwischenmenschliche. Aber sei's drum.

Ich tue mir mit der Ergänzung leider auch etwas schwer. Das Ganze läuft wie oben geschrieben als Home Assistant Plugin.

> Show evcc log output by running with evcc --log debug. The evcc service MUST be stopped before running this command.

Wo genau soll ich "evcc --log debug" dort eingeben? Unter Einstellungen -> Add-Ons -> evcc -> Protokoll sehe ich Daten, aber wenn ich den evcc Service wie beschrieben stoppe, ist das Protokollfenster leer. Ich ergänze den obigen Beitrag aber nun um die Debug-Daten, die ich dort sehe. Außerdem um einen Screenshot, in der man sieht, dass die Einstellung auf "PV" steht, derzeit 0,0 kW vom Dach kommen und das Auto mit 1,4 kW lädt.

andig, 2024-01-26:
HA kann leider keine vernünftigen Logfiles zur Verfügung stellen. Auch das ist dokumentiert. Es ist super mühsam, wenn die Hinweise so konsequent ignoriert werden:

[image]

cg089, 2024-01-26:
Das hat weder etwas mit Ignorieren, noch mit Unwillen zu tun, sondern mit Unfähigkeit meinerseits. Aber es ist ja ohnehin "Closed".