evcc-io/evcc #11860: Possible support for VOLTRONIC AXPERT KING II 5KW 48V

elalemanyo, 2024-01-25:
Hi,
I have an installation that uses this [model](https://www.wccsolar.net/product-page/inversor-hibrido-5kw-48v-mppt-axpert-king?lang=en), and I was wondering if it would be possible to use it with EVCC.
Thank you very much for your help and for the project.

andig, 2024-01-25:
Is there documentation? Please ask the vendor to get in contact with info@evcc.io if they want to support the implementation.

elalemanyo, 2024-01-25:
@andig [this](https://c805a9dd-39dc-4862-a11b-067edfdcff9b.usrfiles.com/ugd/c805a9_1993556c4054452fb8a9b4f4cc885019.pdf) seems to be the instruction manual
but it does not put much 🙁
I don't think I can talk to the manufacturer

andig, 2024-01-26:
Nothing we could do without documentation :(

closed 2024-01-26 by andig

elalemanyo, 2024-01-26:
https://c805a9dd-39dc-4862-a11b-067edfdcff9b.usrfiles.com/ugd/c805a9_1993556c4054452fb8a9b4f4cc885019.pdf

This is not enough? What kind of documentation would be required?

premultiply, 2024-01-26:
A documentation about the control interface (like Modbus Register table etc.) was meant.