evcc-io/evcc #11862: Tesla outside of EU will not retrieve vehicle data.

infinus, 2024-01-25:
### Describe the bug

The current Tesla integration in latest nightly will not retrieve vehicle data such as SOC for non EU vehicle. 

I believe this is due the the vehicle being in NA but the API is set to use the .eu endpoint. 

Per Tesla API documentation if you use the base URL then it will return the proper region for the user. I think we can either use the base URL to tretreive the correct region or have a selection within EVCC that allows region selection to use appropriate URL:
[https://developer.tesla.com/docs/fleet-api#regional-requirements](https://developer.tesla.com/docs/fleet-api#regional-requirements)

### Steps to reproduce

1. install nightly
2. create new auth token via Tesla.evcc.io and grant all permissions
3. update evcc.yaml with token and  template to tesla-command
4. launch EVCC and look at logs
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace

# unique installation id
plant: 5e866f7fa74360b3e83390f7677cc63bcaa6a0e5542a0cdccffec90af47b531c

interval: 10s # control cycle interval

meters:
- type: template
  template: enphase
  usage: grid
  host: 192.168.1.225
  token: *removed*
  name: grid1
- type: template
  template: enphase
  usage: pv
  host: 192.168.1.225
  token: *removed*
  name: pv2

chargers:
- type: template
  template: twc3
  host: 192.168.1.108
  name: wallbox4

vehicles:
- name: Infinus
  type: template
  template: tesla-command
  accessToken: *removed*
  refreshToken: *removed*
  vin: *removed*
  capacity: 74 # kWh
loadpoints:
- title: Tesla
  charger: wallbox4
  vehicle: Infinus
  mode: pv
  phases: 1
  mincurrent: 5
  maxcurrent: 48

tariffs:
  currency: USD
  grid:
    # static grid price
    type: fixed
    price: 0.0809175 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.0 # [currency]/kWh

site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv2
```

### Log details

```text
2024/01/25 10:32:33	stdout	[db    ] INFO 2024/01/25 10:32:33 using sqlite database: /root/.evcc/evcc.db				
2024/01/25 10:32:33	stdout	[main  ] INFO 2024/01/25 10:32:33 starting ui and api at :7070				
2024/01/25 10:32:33	stdout	[main  ] INFO 2024/01/25 10:32:33 using config file: /etc/evcc.yaml				
2024/01/25 10:32:33	stdout	[main  ] INFO 2024/01/25 10:32:33 evcc 0.123.9 (1989b1df)				
2024/01/25 10:32:34	stdout	[lp-1  ] DEBUG 2024/01/25 10:32:34 charge power: 0W				
2024/01/25 10:32:34	stdout	[site  ] DEBUG 2024/01/25 10:32:34 ----				
2024/01/25 10:32:34	stdout	[site  ] WARN 2024/01/25 10:32:34 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval				
2024/01/25 10:32:34	stdout	[lp-1  ] INFO 2024/01/25 10:32:34 vehicle updated: unknown -> Infinus (offline)				
2024/01/25 10:32:34	stdout	[lp-1  ] DEBUG 2024/01/25 10:32:34 guard timer inactive				
2024/01/25 10:32:34	stdout	[lp-1  ] DEBUG 2024/01/25 10:32:34 pv timer inactive				
2024/01/25 10:32:34	stdout	[lp-1  ] DEBUG 2024/01/25 10:32:34 phase timer inactive				
2024/01/25 10:32:34	stdout	[lp-1  ] INFO 2024/01/25 10:32:34     charge:    power ✓ energy ✗ currents ✓				
2024/01/25 10:32:34	stdout	[lp-1  ] INFO 2024/01/25 10:32:34   meters:      charge ✓				
2024/01/25 10:32:34	stdout	[lp-1  ] INFO 2024/01/25 10:32:34   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗				
2024/01/25 10:32:34	stdout	[lp-1  ] INFO 2024/01/25 10:32:34   mode:        pv				
2024/01/25 10:32:34	stdout	[lp-1  ] INFO 2024/01/25 10:32:34 loadpoint 1:				
2024/01/25 10:32:34	stdout	[site  ] INFO 2024/01/25 10:32:34     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗				
2024/01/25 10:32:34	stdout	[site  ] INFO 2024/01/25 10:32:34   vehicles:				
2024/01/25 10:32:34	stdout	[site  ] INFO 2024/01/25 10:32:34     pv 1:      power ✓ energy ✓ currents ✗				
2024/01/25 10:32:34	stdout	[site  ] INFO 2024/01/25 10:32:34     grid:      power ✓ energy ✗ currents ✗				
2024/01/25 10:32:34	stdout	[site  ] INFO 2024/01/25 10:32:34   meters:      grid ✓ pv ✓ battery ✗				
2024/01/25 10:32:34	stdout	[site  ] INFO 2024/01/25 10:32:34 site config:				
2024/01/25 10:32:34	stdout	[main  ] ERROR 2024/01/25 10:32:34 creating vehicle Infinus failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla-command': cannot get vehicles: unexpected status: 421 (Misdirected Request)				
2024/01/25 10:32:35	stdout	[site  ] DEBUG 2024/01/25 10:32:35 pv power: 5645W				
2024/01/25 10:32:36	stdout	[lp-1  ] DEBUG 2024/01/25 10:32:36 set session energy limit: 0				
2024/01/25 10:32:36	stdout	[lp-1  ] DEBUG 2024/01/25 10:32:36 set session soc limit: 0				
2024/01/25 10:32:36	stdout	[lp-1  ] INFO 2024/01/25 10:32:36 car disconnected				
2024/01/25 10:32:36	stdout	[lp-1  ] DEBUG 2024/01/25 10:32:36 charger status: A				
2024/01/25 10:32:36	stdout	[lp-1  ] DEBUG 2024/01/25 10:32:36 charge currents: [0 0 0]A				
2024/01/25 10:32:36	stdout	[lp-1  ] DEBUG 2024/01/25 10:32:36 charge voltages: [2.1 2.1 2.1]V				
2024/01/25 10:32:36	stdout	[site  ] DEBUG 2024/01/25 10:32:36 site power: -2183W				
2024/01/25 10:32:36	stdout	[site  ] DEBUG 2024/01/25 10:32:36 grid power: -2183W				
2024/01/25 10:32:44	stdout	[lp-1  ] DEBUG 2024/01/25 10:32:44 charge power: 0W				
2024/01/25 10:32:44	stdout	[site  ] DEBUG 2024/01/25 10:32:44 ----				
2024/01/25 10:32:45	stdout	[site  ] DEBUG 2024/01/25 10:32:45 pv power: 5644W				
2024/01/25 10:32:46	stdout	[lp-1  ] DEBUG 2024/01/25 10:32:46 charger status: A				
2024/01/25 10:32:46	stdout	[lp-1  ] DEBUG 2024/01/25 10:32:46 charge currents: [0 0 0]A				
2024/01/25 10:32:46	stdout	[lp-1  ] DEBUG 2024/01/25 10:32:46 charge voltages: [2.1 2.1 2.1]V				
2024/01/25 10:32:46	stdout	[site  ] DEBUG 2024/01/25 10:32:46 site power: -2197W				
2024/01/25 10:32:46	stdout	[site  ] DEBUG 2024/01/25 10:32:46 grid power: -2197W
```

### What type of operating system are you running?

Docker container

### Version

0.123.9 (1989b1df) nightly

andig, 2024-01-25:
Please send credentials to info@evcc.io

andig, 2024-01-25:
Closed in https://github.com/evcc-io/evcc/commit/220db88a069d9509328c0052f5ae41618ca0f300

closed 2024-01-25 by andig

infinus, 2024-01-25:
I will test once nightly updates thank you!