evcc-io/evcc #11875: Tesla Powerwall Battery Control not working anymore

FlNDUS, 2024-01-26:
### Describe the bug

The Feature to decide weather to use the Battery or not while charging in "now" or "planned mode" is not working anymore.
I allowed evcc access via tesla.evcc.io but nevertheless I get an error while starting evcc.

`[main  ] FATAL 2024/01/26 16:06:11 cannot create meter 'my_battery': cannot create meter type 'template': cannot create meter type 'powerwall': Get "https://owner-api.teslamotors.com/api/1/energy_sites/<...>/site_info": oauth2: cannot fetch token: 400 Bad Request
Response:
[main  ] FATAL 2024/01/26 16:06:11 will attempt restart in: 5m0s`

In addition to that, it would be nice to be able to define the default SoC evcc sets at the powerwall via Webinterface or yaml.
Background to that, I have a Tesla Powerwall 2 with Gateway 1 (without Backuprecovery), because of that I am not able to set the SoC Value for the Powerwall within the Tesla app, but the lock mechanism worked in the past, but I cant change it.

### Steps to reproduce

1. Own an Tesla Powerwall 2 (with Gateway 1)
2. Configure Refreshtoken + SiteID for tesla-powerwall
3. Run evcc 

### Configuration details

```yaml
- name: my_battery
    type: template
    template: tesla-powerwall
    usage: battery
    host: <..> # IP-Adresse oder Hostname
    password: <..> 
    refreshToken: <..>
    siteId: <..>
```

### Log details

```text
[main  ] FATAL 2024/01/26 16:06:11 cannot create meter 'my_battery': cannot create meter type 'template': cannot create meter type 'powerwall': Get "https://owner-api.teslamotors.com/api/1/energy_sites/<...>/site_info": oauth2: cannot fetch token: 400 Bad Request
Response:
[main  ] FATAL 2024/01/26 16:06:11 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Docker container

### Version

0.123.9 (28f74708)

andig, 2024-01-26:
@GrimmiMeloni that's bad and currently not fixable. See https://github.com/evcc-io/evcc/issues/11846

andig, 2024-01-26:
Workaround: disable battery control

@GrimmiMeloni for time being we should remove advertised functionality from the template?

GrimmiMeloni, 2024-01-26:
> Workaround: disable battery control
> 
> @GrimmiMeloni for time being we should remove advertised functionality from the template?

Yes.

I just did a capture of my Tesla App traffic. The silver lining here is, that the actual API for the battery control has not changed. This seems to be purely about authentication related changes. However, it is not as complicated as with the vehicles and their certs.

GrimmiMeloni, 2024-01-26:
> > Workaround: disable battery control
> > @GrimmiMeloni for time being we should remove advertised functionality from the template?
> 
> Yes.

@andig hold that. I just spun up a copy from master and I can still successfully boot up using my refresh token, and it also fetches the energy site information.

@FlNDUS can you provide a `trace` log for `powerwall` from your instance? If privacy is a concern, you can also send via Slack.

andig, 2024-01-26:
Does the twc not need certs for sending commands now?

GrimmiMeloni, 2024-01-27:
Apparently not. Based on my observations earlier it still works with a plain access token as before. 

My best guess at this point is, that my refresh token from „way back when“ somehow differs (claims?, scopes?) from more recent ones. @FlNDUS and mywelf will have to compare a bit what we see individually to better understand what has changed. 

As a side note, there was a firmware update rolled to the Tesla Energy Gateways this week, so far my understanding is though that this only affects the local API.

GrimmiMeloni, 2024-01-27:
Just as another follow up. I just created an entire new set of tokens via https://github.com/adriankumpf/tesla_auth. Even with that new regenerated refresh_token I can successfully start and see the energy_site data fetched on start.

So it is does not seem to be a _generic_ problem (yet). But could be of course that Tesla is migrating users or something behind the scenes.

andig, 2024-01-27:
So- anything to do here?

GrimmiMeloni, 2024-01-27:
> So- anything to do here?

Not sure. Let's at least wait for @FlNDUS to followup on the log request, so that we can have a closer look.

andig, 2024-01-27:
Worth noting: using tesla.evcc.io we're currently not requesting the energy products scope so the new tokens at this time cannot be used in any command scenario.

FlNDUS, 2024-01-27:
Good to know my token is created using tesla.evcc.io. Should I use/create a different token then?

I have limited access right now(just my phone), will follow-up as soon as I can. @GrimmiMeloni will send you the trace via Slack like before.

GrimmiMeloni, 2024-01-28:
You might want to check your refresh token for differences, here's the payload section of mine which I created yesterday.

```json
{
  "iss": "https://auth.tesla.com/oauth2/v3",
  "aud": "https://auth.tesla.com/oauth2/v3/token",
  "iat": 1706349992,
  "scp": [
    "openid",
    "offline_access"
  ],
  "ou_code": "EU",
  "data": {
    "v": "1",
    "aud": "https://owner-api.teslamotors.com/",
    "sub": "<MY-UUID>",
    "scp": [
      "openid",
      "email",
      "offline_access"
    ],
    "azp": "ownerapi",
    "amr": [
      "pwd"
    ],
    "auth_time": 1706349992
  }
}
```

andig, 2024-01-28:
@GrimmiMeloni  for clarity: your token is NOT created on tesla.evcc.io, right?

GrimmiMeloni, 2024-01-28:
> @GrimmiMeloni for clarity: your token is NOT created on tesla.evcc.io, right?

Correct. The one shown above was created via https://github.com/adriankumpf/tesla_auth.

FlNDUS, 2024-01-29:
So I updated to the current nightly and created a new token via https://github.com/adriankumpf/tesla_auth now it works again.
Do you still need the logs then @GrimmiMeloni ?

closed 2024-01-29 by andig

GrimmiMeloni, 2024-01-29:
No need. Glad to hear it works.

andig, 2024-01-29:
I'm wondering why that is: the refresh token should not refresh unless using the evcc ClientID plus would have a different audience (i.e. not owners API)- seems it still does, reason unclear. 

Also noted that powerwall uses `refreshtoken` while most other devices use

    tokens:
      refresh: xyz

for the same purpose. Small consistency issue.

GrimmiMeloni, 2024-01-30:
> seems it still does, reason unclear.

Theory: @FlNDUS is also driving a Tesla and hence he was motivated to get a new refresh Token for the sake of getting control of his car back, after the recent changes with the Fleet API. Hence he exchanged the refresh token in his config, and also replaced the one for the powerwall.

> Small consistency issue.

True - on the flipside, as putting ephemeral access tokens into the config does not make sense, the "tokens" structure would have only contained one subelement. 🤷‍♂️  Do other devices have more elements in there?

andig, 2024-01-30:
They do, but really only refresh is needed. Maybe we should update the other way round ;)