evcc-io/evcc #11880: Tesla.evcc.io doesn't create a valid token

pschyrembel258, 2024-01-27:
### Describe the bug

Since 25.1.2024 my Tesla is offline. I searched here for issues and found that the api changed. I changed my configuration in the evcc.yaml and tried to follow the steps (Add Tesla using official vehicle command library):
1. generate a new pair of API tokens here https://tesla.evcc.io/.    **<-- Here I struggle to get a token**
2. switch to the latest nightly build of evcc.    **<-- I'm using the home assistant add-on and have no idea how to get the latest nightly build**
3. change your template from tesla to tesla-command.  **<-- This step is done**

I tried to get new refresh and access tokens on tesla.evcc.io. I could login with my Tesla account, I can select wich permissions I will give to evcc.io and than I get:

Tokenerzeugung fehlgeschlagen
insufficient scopes: missing token scopes

I've tried it with Safari, Firefox and Chrome, but it won't work

### Steps to reproduce

1. login with my Tesla credentials on tesla.evcc.io to get tokens
2. select all permissions
3. Error: insufficient scopes: missing token scopes
4. Delete evcc.io (PROD) in my tesla account
5.  Try it again with a different browser
...

### Configuration details

```yaml
vehicles:
  - name: Tesla
    type: template
    template: tesla-command
    title: Tesla xxxx # Wird in der Benutzeroberfläche angezeigt (Optional)
    accessToken: xxxx
    refreshToken: xxxxx
    vin: xxxx # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (Optional)
    capacity: 62 # Akkukapazität in kWh (Optional)
```

### Log details

```text
Well as the tokens are not right in my config I get:

ERROR 2024/01/27 09:44:17 creating vehicle Tesla failed: cannot create vehicle type 'template': template not found: tesla-command
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.123.9

Al3xand3r1987, 2024-01-27:
I have the same problem. I think the new api has not been changed yet. A few days ago I read that Tesla did something to the api

Al3xand3r1987, 2024-01-27:
I have entered it via the GUI in Evcc. This error message appears when checking

Status: fehlgeschlagen
[prüfen](http://192.168.1.250:8123/api/hassio_ingress/Et1Fmkkwuk5_BrSY1lrSqCRSix2_AI8qjIwF3q0BoxU/#)
cannot create vehicle type 'template': cannot create vehicle type 'tesla': cannot get vehicles: 412 Precondition Failed

andig, 2024-01-27:
> Here I struggle to get a token

Fixed and deployed.

closed 2024-01-27 by andig

pschyrembel258, 2024-01-27:
Thank you for fixing it @andig . Can you help me with the second part? How do I get the nightly build in home assistant add-on?

Thanks

andig, 2024-01-27:
@pschyrembel258 Please use search function- this has been discussed multiple times.