evcc-io/evcc #11887: Add aux meter templates

andig, 2024-01-27:
Es wäre übrigens schick, für den Thor ein Template anzulegen. PR welcome.

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/11885#discussioncomment-8265440_

naltatis, 2024-01-28:
Mir ist der Sinn noch nicht ganz klar. Aktuell kann ich doch in AUX ganz normale Meter verwenden. Was gewinnen wir durch den neuen Typen? Müssten wir den dann nicht bei fast allen besthenden Metern auch nachtragen?

andig, 2024-01-28:
Aux sind intelligente Geräte die ihre Leistung selbst regeln. Diese ist daher als "verfügbar" anzusehen. Für die Geräte brauchts Templates wie heir.

> Aktuell kann ich doch in AUX ganz normale Meter verwenden.

Das kannst Du auch weiter machen, ändert an der Verwendung aber nichts.

naltatis, 2024-01-28:
> Das kannst Du auch weiter machen, ändert an der Verwendung aber nichts.

Ok, ich glaub das war der Part der mir fehlte.

Aktuell wird `usage` "nur" in `evcc configure` und im Config UI verwendet um die Liste der "passenden" Templates für den jeweiligen Anwendungsfall anzubieten. In der evcc.yaml kann man aber aktuell frei kombinieren. Heißt man kann ein Meter, dass nur `usage=battery` hat auch als Grid-Meter verwenden. Das prüfen wir im Core nicht. Richtig? Daher kann ich auch für `aux` weiterhin ein beliebiges Meter verwenden.

In der Web UI (und configure) ist es aktuell so, dass man nur die Templates angeboten bekommt, die für den Verwendungszweck auch passen. Bei Aux wäre das mit diesem PR dann erstmal nur das eine Gerät. Heißt wir müssten dann Stück für Stück die bestehenden Meter "nachlabeln" die für diesen Anwendungsfall heute schon benutzt werden weil bspw. Multipurpose (bspw Shelly 1pm an einem intelligenten Heizstab). Finde ich gut und auch deutlich besser als alle Meter die es gibt anzuzeigen.

Ich hab im Epic Config UI https://github.com/evcc-io/evcc/issues/6029 mal Aux mit aufgenommen. Das fehlte da noch, bekommt mit diesem PR dann aber ne saubere Struktur. Außerdem hab ich dort noch mal mit aufgenommen, dass wir auch die Verwendung von Meter außerhalb ihrer vorgesehenen Usage erlauben sollten. Ob das dann über yaml-Snippet oder UI-geführt (weitere Geräte ...) passiert müssen wir dann mal sehen.

andig, 2024-01-28:
Das gilt wahrscheinlich für alle Bastelzähler (=Steckdosen) wie theoretisch auch für jeden normalen Zähler. Aktuell siehts so aus:

[image]

Warum ausgerechnet der Acrel da rein gerutscht ist bleibt unklar. Auf jeden Fall sind wir jetzt schon nicht konsequent.

closed 2024-01-28 by andig

naltatis, 2024-01-28:
Ok, aber dann haben wir zumindest ein sauberes Modell und eine Single Source of Truth, die wir vtml noch etwas nachpflegen müssen.

PR #11912 (2024-01-28): Add AC Thor
Replace https://github.com/evcc-io/evcc/pull/11894, Fix #11887 

TODO

- [x] add `aux` devices to website (brands.json)
- [ ] add `aux` capability to docs @naltatis https://github.com/evcc-io/docs/issues/520