evcc-io/evcc #11897: Splitted Charge Plan not paused correctly

BeckerWdf, 2024-01-27:
### Describe the bug

Last night I had a splitted charge plan to due the dynamic prices from tibber: "From 3-4  und 13-14 -> Target SoC 80% to be reached at 17:45"

At 4 o'clock charging was not paused but continued until the Target SoC of 80% was reached.

### Steps to reproduce

1. Have a dynamic electricity tariff
2. Create a charge plan on a day where the "cheapest hours" are not next to each other (e.g at 3 o'clock and adt 6 o'clock)
3. Choose a target SoC and target time so that a splitted plan will be created.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

mqtt:
  broker: homepi:1883
  topic: evcc

# unique installation id
plant: ####

interval: 10s # control cycle interval, WARNING: 30s leads to wallbox status "F"

sponsortoken: xxxxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
  - name: netz
    type: template
    template: solaredge-inverter
    usage: grid
    modbus: tcpip
    id: 1
    host: solaredge-dg
    port: 1502 
  - name: solaredge
    type: template
    template: solaredge-inverter
    usage: pv
    modbus: tcpip
    id: 1
    host: solaredge-dg
    port: 1502 

modbusproxy:
  - port: 1502
    uri: solaredge-dg:1502
    rtu: false

chargers:
- type: template
  template: heidelberg 
  id: 1  
  device: /dev/ttyUSB_WB  
  baudrate: 19200  
  comset: 8E1  
  modbus: rs485serial  
  name: wallbox1

loadpoints:
[... 200 lines cut ...]
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/27 05:01:36 charge voltages: [234 231 236]V
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/27 05:01:36 detected connected phases: 3p
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/27 05:01:36 charge currents: [0 0 0]A
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/27 05:01:36 charge total import: 2760.896kWh
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/27 05:01:36 charger status: B
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/27 05:01:36 final charge energy: 22kWh
smarthome-evcc-1  | [lp-1  ] INFO 2024/01/27 05:01:36 stop charging <-
smarthome-evcc-1  | [telegram] DEBUG 2024/01/27 05:01:36 sending to ***
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/27 05:01:36 charge total import: 2760.896kWh
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/27 05:01:38 vehicle status: B
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/27 05:01:38 soc gradient updated: soc: 80.0%, socDiff: 34.0%, energyDiff: 22010Wh, energyPerSocStep: 647.4Wh, virtualCapacity: 64735Wh
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/27 05:01:38 vehicle soc: 80%
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/27 05:01:38 vehicle soc limit: 100%
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/27 05:01:38 vehicle range: 256km
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/27 05:01:38 set db:1 plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/27 05:01:38 limitSoc reached: 80.0% > 80%
```

### What type of operating system are you running?

Docker container

### Version

0.123.8

andig, 2024-01-28:
Der Planner hat eine Reihe von Sonderfällen die zu häufiges Abschalten verhindern sollen:

```go
		// planner was active (any slot, not necessarily previous slot) and charge goal has not yet been met
		switch {
		case lp.clock.Now().After(planTime) && !planTime.IsZero():
			// if the plan did not (entirely) work, we may still be charging beyond plan end- in that case, continue charging
			// TODO check when schedule is implemented
			lp.log.DEBUG.Println("plan: continuing after target time")
			return true
		case lp.clock.Now().Before(lp.planSlotEnd) && !lp.planSlotEnd.IsZero():
			// don't stop an already running slot if goal was not met
			lp.log.DEBUG.Println("plan: continuing until end of slot")
			return true
		case requiredDuration < smallGapDuration:
			lp.log.DEBUG.Printf("plan: continuing for remaining %v", requiredDuration.Round(time.Second))
			return true
		case lp.clock.Until(planStart) < smallGapDuration:
			lp.log.DEBUG.Printf("plan: will re-start shortly, continuing for remaining %v", lp.clock.Until(planStart).Round(time.Second))
			return true
		}
```

Die `smallGapDuration` ist aktuell 1h- deshalb wird zwischen 4:00 und 4:50 nicht mehr abgeschaltet. Änderung eingeführt in https://github.com/evcc-io/evcc/pull/7419. Da müsst ihr Euch einig werden ;)

/cc @Hofyyy @schenlap

andig, 2024-01-28:
Bessere Log messages in https://github.com/evcc-io/evcc/commit/f73e064e9d707beb001b4986d52d1663301ac44c

BeckerWdf, 2024-01-28:
> Die `smallGapDuration` ist aktuell 1h- deshalb wird zwischen 4:00 und 4:50 nicht mehr abgeschaltet. Änderung eingeführt in #7419. Da müsst ihr Euch einig werden ;)

Aber es sollte doch erst um 13 Uhr weitergehen. Das ist ja deutlich mehr als 1h. ;-)

andig, 2024-01-28:
Ist für mich nicht erkennbar:

> plan: charge 1h9m40s starting at 2024-01-27 04:50:20 +0100 CET until 2024-01-27 17:45:00 +0100 CET (power: 11040W, avg cost: 0.233)

Da steht klar 4:50?

BeckerWdf, 2024-01-28:
> Ist für mich nicht erkennbar:
> 
> > plan: charge 1h9m40s starting at 2024-01-27 04:50:20 +0100 CET until 2024-01-27 17:45:00 +0100 CET (power: 11040W, avg cost: 0.233)
> 
> Da steht klar 4:50?

Also dieser Plan wurde gesetzt:

```
smarthome-evcc-1  | [site  ] DEBUG 2024/01/26 20:03:47 set db:1 plan soc: 80 @ 2024-01-27 17:45:00 +0100 CET
smarthome-evcc-1  | [site  ] DEBUG 2024/01/26 20:03:56 ----
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/26 20:03:56 charge power: 0W
smarthome-evcc-1  | [site  ] DEBUG 2024/01/26 20:03:56 pv power: 0W
smarthome-evcc-1  | [site  ] DEBUG 2024/01/26 20:03:56 grid power: 2345W
smarthome-evcc-1  | [site  ] DEBUG 2024/01/26 20:03:56 site power: 2345W
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/26 20:03:56 charge voltages: [235 233 238]V
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/26 20:03:56 detected connected phases: 3p
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/26 20:03:56 charge currents: [0 0 0]A
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/26 20:03:56 charge total import: 2738.879kWh
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/26 20:03:56 charger status: B
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/26 20:03:56 plan: charge 1h59m5s starting at 2024-01-27 03:00:55 +0100 CET until 2024-01-27 17:45:00 +0100 CET (power: 11040W, avg cost: 0.234)
smarthome-evcc-1  | [lp-1  ] DEBUG 2024/01/26 20:03:56 pv charge current: 0A = 0A + -3.4A (2345W @ 3p)
smarthome-evcc-1  | [site  ] DEBUG 2024/01/26 20:04:06 ----
```

Die Preise waren so:
[image]

Im Plan-UI wurden mit zwei Slots einer von 3-4 und einer von 13-14 Uhr angezeigt. Leider sind die Slots im Log nicht ersichtlich.

Wie kommt er dann während des Ladens drauf, dass es besser ist noch die Stunde von 4-5 Uhr mitzunehmen statt die günstigere Stunde von 13-14 Uhr?

schenlap, 2024-01-28:
`03:00:56 plan: charge 1h59m5s starting at 2024-01-27 03:00:57`

Da war der Plan etwas kürzer als 2 Stunden, also werden 2 Slots geplant. Um 3 begonnen ist OK.

`04:00:06 plan: charge 1h9m40s`

Da gibt es jetzt den Sonderfall dass
a) die zuvor geschätzte Ladezeit überschritten wird. Nach 1h laden bleiben noch immer 1h9min Rest. Es muss daher ein dritter Slot verwendet werden.
b) zuvor schon geladen wurde und daher der nächstgünstige Slot (welcher eigentlich erst später aktiviert werden müsste) eingeschaltet bleibt. Das wird auch in der Log-Meldung angezeigt (`will re-start shortly, continuing for remaining 50m13s`).

Der Planer verhält sich so wie geplant. Was hier passiert ist eine Optimierung dass nicht zu oft ein-/ausgeschaltet wird. Die Ladedauer lässt sich leider nicht gut vorhersagen, das kann immer etwas zunehmen oder abnehmen und das ständige ein/aus muss man unterbinden. Er lädt aber trotzdem nur in den günstigsten Stunden. Es kann aber passieren, wie auch in diesem Beispiel, dass er die günstigste Stunde nicht zu 100% nimmt sondern weniger.
Wenn man die ganzen Sonderfälle rausnehmen würde dann wäre er vielleicht um 14:00 noch nicht ganz fertig gewesen (sieht man in Log nicht) und hätte auch nach 14:00 noch laden müssen, da wäre der Preis aber einiges höher gewesen. Ich würde da am Verhalten nichts ändern.

Mit Trace hat er zumindest früher mal die einzelnen Slots angezeigt. Vielleicht geht das immer noch.

schenlap, 2024-01-28:
> Aber es sollte doch erst um 13 Uhr weitergehen. Das ist ja deutlich mehr als 1h. ;-)

Das war nur der ursprüngliche Plan. Da der SoC nicht schnell genug abgenommen hat war ein Teil einer dritten Stunde notwendig, und das war von 4:00 - 5:00. Es hätte wohl gereicht um 4:50 einzuschalten. Das macht er aber nicht da er für die 50 Minuten (in einer günstigen Stunde, wenn auch nicht die günstigste) nicht abschaltet.

andig, 2024-01-28:
Wichtig zu wissen ist, dass bei jedem Durchlauf ein neuer Slot erzeugt wird. Ich denke hier ist alles beantwortet?

BeckerWdf, 2024-01-29:
> Was hier passiert ist eine Optimierung dass nicht zu oft ein-/ausgeschaltet wird. Die Ladedauer lässt sich leider nicht gut vorhersagen, das kann immer etwas zunehmen oder abnehmen und das ständige ein/aus muss man unterbinden. 

Was meinst du mit "ständigem ein/aus schalten"? Wann würde das passieren und warum ist das schlecht?

>Er lädt aber trotzdem nur in den günstigsten Stunden. Es kann aber passieren, wie auch in diesem Beispiel, dass er die günstigste Stunde nicht zu 100% nimmt sondern weniger.

Welche Stunden werden als "günstigste Stunden" interpretiert. In meinem Fall hat er also 1h lang in einer Stunde geladen, die teurer war als eine günstigere, die noch gekommen wäre.

Und wenn da so Verhalten wirklich so gewünscht ist, dann sollte man das dem User auch im UI transparent machen. Ich bin davon ausgegangen, dass er in den beiden Stunden lädt, die mir angezeigt werden.

andig, 2024-01-29:
> In meinem Fall hat er also 1h lang in einer Stunde geladen, die teurer war als eine günstigere, die noch gekommen wäre.

Das würde er nicht tun und gibt Dein Screenshot auch nicht her!

BeckerWdf, 2024-01-29:
> > In meinem Fall hat er also 1h lang in einer Stunde geladen, die teurer war als eine günstigere, die noch gekommen wäre.
> 
> Das würde er nicht tun und gibt Dein Screenshot auch nicht her!

Leider hab ich das Plan-UI nicht als Screenshot festgehalten....

andig, 2024-01-29:
Oben ist dein Tibber Screenshot. Der nächste Slot ist der nächstbilligste. Hier ist alles Lieben Kuß.

BeckerWdf, 2024-01-29:
> Oben ist dein Tibber Screenshot. Der nächste Slot ist der nächstbilligste. Hier ist alles ok.

Der billiste ist um 13 Uhr. Warum wird der nicht gewählt?

andig, 2024-01-29:
Wird er doch 🙄. Aber die eine Stunde reicht nicht, also brauchts noch einen Slot.

BeckerWdf, 2024-01-29:
> Wird er doch 🙄. Aber die eine Stunde reicht nicht, also brauchts noch einen Slot. Long Story short: der Planner funktioniert perfekt.

Das dachte er. Aber es hat halt doch gereicht. Er hat aum 13 gar nicht erst gestartet weil er vorher schon voll geladen hatte.

andig, 2024-01-29:
Ist Dir ein leeres Auto lieber? Es ist halt eine Schätzung wie schnell geladen wird. Die Batterien sind nicht-linear...