evcc-io/evcc #11902: Ladevorgangsauswertung für Firmenwagenabrechnung

cerwin86, 2024-01-28:
Hallo Zusammen,

die bisherigen Reports über die Ladevorgänge spiegeln die echten Kosten des Ladevorganges wieder. Echt super soweit.
Nun fällt mir eine weitere Variante ein. Für die Besitzer von E Firmenwagen wäre es hilfreich, wenn für die Ladevorgänge zusätzlich die theoretischen Kosten ohne PV aufgelistet würden. Also was hätte der Ladevorgang übers Netz gekostet. Besonders bei der Nutzung von Tibber etc. sehr wichtig, da nicht manuell auswertbar.

Das ganze kommt daher, dass Firmenwagenfahrer die theoretischen Kosten mit dem Arbeitgeber abrechnen können.

Siehe: https://www.vattenfall.de/infowelt-energie/e-mobility/firmenwagen-zuhause-laden
Oder
https://www.eismann-partner.de/e-dienstwagen-auslagenersatz-fu%cc%88r-privates-aufladen-des-dienstwagens/

Umsetzung dürfte recht simpel sein, den aktuellen Tibberpreis zusätzlich mitschreiben, bei Export oder Anzeige einen Schalter der Echtpreis oder Abrechnungspreis auswählen lässt?
Stelle ich mir das zu einfach vor? 😂

andig, 2024-01-28:
Das lässt sich ja super einfach in Excel machen. Aktuellen Preis mitschreiben haben wir auf der Liste. Das Ausrechnen würde ich dem Anwender überlassen. 

/cc @naltatis

cerwin86, 2024-01-28:
Hey, das heißt ihr würdet die zum Ladevorgang genutzten Preise protokollieren und dies könnte man dann in der CSV zusammenführen?

andig, 2024-01-28:
> Hey, das heißt ihr würdet die zum Ladevorgang genutzten Preise protokollieren und dies könnte man dann in der CSV zusammenführen?

Ich bin nicht ganz sicher wie das gehen soll da es nicht nur einen Preis für einen Ladevorgang gibt?