evcc-io/evcc #11931: Planner beachtet loadpoint `phases: 1` nicht

StefanSchoof, 2024-01-29:
### Describe the bug

Ausgangslage:
- eine Wallbox ist nur in der Lage mit 1p zu laden
- Am loadpoint ist `phases: 1` hinterlegt
- Das Vehicle ist in der Lage 3p zu laden

Erstellen eine Plan zum laden
[image]
Dieser Plan nimmt 3p (11kw ) an, obwohl die Loadpoint config dies nicht zu lässt. Log Ausgabe `plan: charge 54m21s between 2024-01-30 06:05:39 +0100 CET until 2024-01-30 07:00:00 +0100 CET (power: 11040W, avg cost: 0.000)`

Die config Datei für die Reproduktion basiert zu großen Teil auf https://github.com/evcc-io/evcc/blob/master/tests/basics.evcc.yaml und benötigt keine echte Hardware

Bekanntes Workaround: Am vehicle  `phases: 1` einfügen Log dann `plan: charge 2h43m2s between 2024-01-30 04:16:58 +0100 CET until 2024-01-30 07:00:00 +0100 CET (power: 3680W, avg cost: 0.000)`

### Steps to reproduce

1. `go run main.go --config tests/1pplan.evcc.yaml`
2.  Aktiviere Ladeplan
3. Ladeplan plant 11kw

### Configuration details

```yaml
interval: 0.1s

log: debug

site:
  title: Hello World
  meters:
    grid: grid

meters:
  - name: grid
    type: custom
    power:
      source: js
      script: |
        1000
  - name: charger_meter
    type: custom
    power:
      source: js
      script: |
        500

loadpoints:
  - title: Carport
    vehicle: my_car
    charger: charger
    meter: charger_meter
    mode: pv
    phases: 1

vehicles:
  - name: my_car
    type: template
    template: offline
    title: my car
    capacity: 50

chargers:
  - name: charger
    type: custom
    enable:
      source: js
      script:
    enabled:
      source: js
      script: |
        true
    status:
      source: js
      script: |
        "C"
    maxcurrent:
      source: js
      script: |
        16
```

### Log details

```text
[lp-1  ] DEBUG 2024/01/29 09:42:15 charge power: 500W
[site  ] DEBUG 2024/01/29 09:42:15 grid power: 1000W
[site  ] DEBUG 2024/01/29 09:42:15 pv power: 0W
[site  ] DEBUG 2024/01/29 09:42:15 site power: 1000W
[lp-1  ] DEBUG 2024/01/29 09:42:15 charger status: C
[lp-1  ] DEBUG 2024/01/29 09:42:15 plan: charge 54m21s between 2024-01-30 06:05:39 +0100 CET until 2024-01-30 07:00:00 +0100 CET (power: 11040W, avg cost: 0.000)
[lp-1  ] DEBUG 2024/01/29 09:42:15 pv charge current: 4.55A = 6A + -1.45A (1000W @ 3p)
[lp-1  ] DEBUG 2024/01/29 09:42:15 guard enable in 4m54s
```

### What type of operating system are you running?

Linux

### Version

master (5b1eb755d4b456f939f46b039524b9ff283e161a)

andig, 2024-01-29:
Was sagt /api/state? Vllt sind die `effectivePhases` falsch?

StefanSchoof, 2024-01-29:
`effectivePhases` ist nicht im `/api/state`.

Nein kein 1p3p, sondern einfach nur 1p angeschlossen:

log:
```
[main  ] INFO 2024/01/29 10:15:14 evcc 0.0.0
[main  ] INFO 2024/01/29 10:15:14 using config file: tests/basics.evcc.yaml
[main  ] INFO 2024/01/29 10:15:14 starting ui and api at :7070
[db    ] INFO 2024/01/29 10:15:14 using sqlite database: /home/stefan/.evcc/evcc.db
[lp-1  ] WARN 2024/01/29 10:15:14 deprecated: phases setting is ignored, please remove
[site  ] INFO 2024/01/29 10:15:14 site config:
[site  ] INFO 2024/01/29 10:15:14   meters:      grid ✓ pv ✗ battery ✗
[site  ] INFO 2024/01/29 10:15:14     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/01/29 10:15:14   vehicles:
[site  ] INFO 2024/01/29 10:15:14     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/01/29 10:15:14 loadpoint 1:
[lp-1  ] INFO 2024/01/29 10:15:14   mode:        pv
[lp-1  ] INFO 2024/01/29 10:15:14   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/01/29 10:15:14   meters:      charge ✓
[lp-1  ] INFO 2024/01/29 10:15:14     charge:    power ✓ energy ✗ currents ✗
[lp-1  ] DEBUG 2024/01/29 10:15:14 phase timer inactive
[lp-1  ] DEBUG 2024/01/29 10:15:14 pv timer inactive
[lp-1  ] DEBUG 2024/01/29 10:15:14 guard timer inactive
[lp-1  ] INFO 2024/01/29 10:15:14 vehicle updated: unknown -> my car
[lp-1  ] DEBUG 2024/01/29 10:15:14 max charge current: 6A
[lp-1  ] DEBUG 2024/01/29 10:15:14 guard enable in 5m0s
[site  ] WARN 2024/01/29 10:15:14 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/01/29 10:15:14 ----
[lp-1  ] DEBUG 2024/01/29 10:15:14 charge power: 500W
[site  ] DEBUG 2024/01/29 10:15:14 grid power: 1000W
[site  ] DEBUG 2024/01/29 10:15:14 pv power: 0W
[site  ] DEBUG 2024/01/29 10:15:14 site power: 1000W
[lp-1  ] DEBUG 2024/01/29 10:15:14 charger status: C
[lp-1  ] INFO 2024/01/29 10:15:14 car connected
[lp-1  ] DEBUG 2024/01/29 10:15:14 pv timer elapse
[lp-1  ] DEBUG 2024/01/29 10:15:14 pv timer inactive
[lp-1  ] DEBUG 2024/01/29 10:15:14 charger: guard elapse
[lp-1  ] DEBUG 2024/01/29 10:15:14 guard timer inactive
[lp-1  ] INFO 2024/01/29 10:15:14 start charging ->
[lp-1  ] DEBUG 2024/01/29 10:15:14 wake-up timer: stop
[lp-1  ] DEBUG 2024/01/29 10:15:14 plan: charge 54m21s between 2024-01-30 06:05:39 +0100 CET until 2024-01-30 07:00:00 +0100 CET (power: 11040W, avg cost: 0.000)
[lp-1  ] DEBUG 2024/01/29 10:15:14 pv charge current: 4.55A = 6A + -1.45A (1000W @ 3p)
[lp-1  ] DEBUG 2024/01/29 10:15:14 site power 1000W >= 0W disable threshold
[lp-1  ] DEBUG 2024/01/29 10:15:14 pv disable in 0s
[lp-1  ] DEBUG 2024/01/29 10:15:14 pv disable timer elapsed
[lp-1  ] DEBUG 2024/01/29 10:15:14 charger disable
[lp-1  ] DEBUG 2024/01/29 10:15:14 wake-up timer: stop
[site  ] DEBUG 2024/01/29 10:15:15 ----
[lp-1  ] DEBUG 2024/01/29 10:15:15 charge power: 500W
[site  ] DEBUG 2024/01/29 10:15:15 grid power: 1000W
[site  ] DEBUG 2024/01/29 10:15:15 pv power: 0W
[site  ] DEBUG 2024/01/29 10:15:15 site power: 1000W
[lp-1  ] DEBUG 2024/01/29 10:15:15 charger status: C
[lp-1  ] DEBUG 2024/01/29 10:15:15 plan: charge 54m21s between 2024-01-30 06:05:39 +0100 CET until 2024-01-30 07:00:00 +0100 CET (power: 11040W, avg cost: 0.000)
[lp-1  ] DEBUG 2024/01/29 10:15:15 pv charge current: 4.55A = 6A + -1.45A (1000W @ 3p)
[lp-1  ] DEBUG 2024/01/29 10:15:15 pv enable timer reset
[lp-1  ] DEBUG 2024/01/29 10:15:15 pv timer inactive
[lp-1  ] DEBUG 2024/01/29 10:15:15 guard enable in 5m0s
[site  ] DEBUG 2024/01/29 10:15:15 ----
[lp-1  ] DEBUG 2024/01/29 10:15:15 charge power: 500W
[site  ] DEBUG 2024/01/29 10:15:15 grid power: 1000W
[site  ] DEBUG 2024/01/29 10:15:15 pv power: 0W
[site  ] DEBUG 2024/01/29 10:15:15 site power: 1000W
[lp-1  ] DEBUG 2024/01/29 10:15:15 charger status: C
[lp-1  ] DEBUG 2024/01/29 10:15:15 plan: charge 54m21s between 2024-01-30 06:05:39 +0100 CET until 2024-01-30 07:00:00 +0100 CET (power: 11040W, avg cost: 0.000)
[lp-1  ] DEBUG 2024/01/29 10:15:15 pv charge current: 4.55A = 6A + -1.45A (1000W @ 3p)
[lp-1  ] DEBUG 2024/01/29 10:15:15 guard enable in 5m0s
[site  ] DEBUG 2024/01/29 10:15:15 ----
[lp-1  ] DEBUG 2024/01/29 10:15:15 charge power: 500W
[... 107 lines cut ...]
                "avgCo2": 0,
                "avgPrice": 0,
                "chargedKWh": 0.015667288048749985,
                "solarPercentage": 0
            },
            "total": {
                "avgCo2": 0,
                "avgPrice": 0,
                "chargedKWh": 0.015667288048749985,
                "solarPercentage": 0
            }
        },
        "vehicles": {
            "my_car": {
                "title": "my car",
                "icon": "car",
                "capacity": 50,
                "features": [
                    "Offline"
                ]
            }
        }
    }
}
```

andig, 2024-01-29:
Komisch ist, dass der PV Modus auch 3p denkt.

StefanSchoof, 2024-01-29:
```json
                "phasesActive": 3,
                "phasesConfigured": 1,
```

ist auch komisch. Hätte gedacht, dass active >= configured gilt.

andig, 2024-01-29:
Es braucht ein volles Logfile um rauszufinden woher 

    "phasesActive": 3,

eigentlich kommt. Das ist die Wurzel des Übels.

andig, 2024-01-29:
@StefanSchoof hast Du es geschafft das irgendwie per GUI oder curl in die Datenbank zu schreiben?

StefanSchoof, 2024-01-30:
Ich habe per GUI einen Plan erstellt.

StefanSchoof, 2024-01-30:
Ich habe jetzt nochmal probiert das nur über die shell zu reproduzieren:

```shell
rm ~/.evcc/evcc.db
go run main.go --config tests/1pplan.evcc.yaml > log.txt
curl -X POST http://localhost:7070/api/loadpoints/1/plan/energy/10/2024-01-31T06:00:00.000Z
```

führt zu 

```
[main  ] INFO 2024/01/30 08:24:11 evcc 0.0.0
[main  ] INFO 2024/01/30 08:24:11 using config file: tests/1pplan.evcc.yaml
[main  ] INFO 2024/01/30 08:24:11 starting ui and api at :7070
[db    ] INFO 2024/01/30 08:24:11 using sqlite database: /home/stefan/.evcc/evcc.db
[db    ] TRACE 2024/01/30 08:24:11 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/01/30 08:24:11 CREATE TABLE `settings` (`key` text,`value` text,PRIMARY KEY (`key`)) 0 <nil>
[db    ] TRACE 2024/01/30 08:24:11 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2024/01/30 08:24:11 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/01/30 08:24:11 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/01/30 08:24:11 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/01/30 08:24:11 CREATE TABLE `configs` (`id` integer PRIMARY KEY AUTOINCREMENT,`class` integer,`type` text) 0 <nil>
[db    ] TRACE 2024/01/30 08:24:11 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/01/30 08:24:11 CREATE TABLE `config_details` (`config_id` integer,`key` text,`value` text,CONSTRAINT `fk_configs_details` FOREIGN KEY (`config_id`) REFERENCES `configs`(`id`) ON DELETE CASCADE ON UPDATE CASCADE) 0 <nil>
[db    ] TRACE 2024/01/30 08:24:11 CREATE INDEX `idx_unique` ON `config_details`(`config_id`,`key`) 0 <nil>
[db    ] TRACE 2024/01/30 08:24:11 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/01/30 08:24:11 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[db    ] TRACE 2024/01/30 08:24:11 SELECT * FROM `configs` WHERE `configs`.`class` = 2 ORDER BY id 0 <nil>
[db    ] TRACE 2024/01/30 08:24:11 SELECT * FROM `configs` WHERE `configs`.`class` = 1 ORDER BY id 0 <nil>
[db    ] TRACE 2024/01/30 08:24:11 SELECT * FROM `configs` WHERE `configs`.`class` = 3 ORDER BY id 0 <nil>
[lp-1  ] WARN 2024/01/30 08:24:11 deprecated: phases setting is ignored, please remove
[db    ] TRACE 2024/01/30 08:24:11 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="sessions" -1 <nil>
[db    ] TRACE 2024/01/30 08:24:11 CREATE TABLE `sessions` (`id` integer PRIMARY KEY AUTOINCREMENT,`created` datetime,`finished` datetime,`loadpoint` text,`identifier` text,`vehicle` text,`odometer` real,`meter_start_kwh` real,`meter_end_kwh` real,`charged_kwh` real,`charge_duration` integer,`solar_percentage` real,`price` real,`price_per_kwh` real,`co2_per_kwh` real) 0 <nil>
[db    ] TRACE 2024/01/30 08:24:11 SELECT * FROM `sessions` WHERE `Loadpoint` = "Carport" AND `finished` = "0001-01-01 00:00:00+00:00" 0 <nil>
[site  ] INFO 2024/01/30 08:24:11 site config:
[site  ] INFO 2024/01/30 08:24:11   meters:      grid ✓ pv ✗ battery ✗
[site  ] INFO 2024/01/30 08:24:11     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/01/30 08:24:11   vehicles:
[site  ] INFO 2024/01/30 08:24:11     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/01/30 08:24:11 loadpoint 1:
[lp-1  ] INFO 2024/01/30 08:24:11   mode:        pv
[lp-1  ] INFO 2024/01/30 08:24:11   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/01/30 08:24:11   meters:      charge ✓
[lp-1  ] INFO 2024/01/30 08:24:11     charge:    power ✓ energy ✗ currents ✗
[cache ] TRACE 2024/01/30 08:24:11 siteTitle: Hello World
[cache ] TRACE 2024/01/30 08:24:11 gridConfigured: true
[cache ] TRACE 2024/01/30 08:24:11 pvConfigured: false
[cache ] TRACE 2024/01/30 08:24:11 batteryConfigured: false
[cache ] TRACE 2024/01/30 08:24:11 bufferSoc: 0
[cache ] TRACE 2024/01/30 08:24:11 bufferStartSoc: 0
[cache ] TRACE 2024/01/30 08:24:11 prioritySoc: 0
[cache ] TRACE 2024/01/30 08:24:11 batteryMode: unknown
[cache ] TRACE 2024/01/30 08:24:11 batteryDischargeControl: false
[cache ] TRACE 2024/01/30 08:24:11 residualPower: 0
[cache ] TRACE 2024/01/30 08:24:11 currency: EUR
[cache ] TRACE 2024/01/30 08:24:11 smartCostActive: false
[cache ] TRACE 2024/01/30 08:24:11 smartCostLimit: 0
[cache ] TRACE 2024/01/30 08:24:11 smartCostType: <nil>
[lp-1  ] DEBUG 2024/01/30 08:24:11 phase timer inactive
[cache ] TRACE 2024/01/30 08:24:11 vehicles: map[my_car:{my car car 50 0 0 [Offline] []}]
[lp-1  ] DEBUG 2024/01/30 08:24:11 pv timer inactive
[lp-1  ] DEBUG 2024/01/30 08:24:11 guard timer inactive
[cache ] TRACE 2024/01/30 08:24:11 lp-1/phasesConfigured: 1
[cache ] TRACE 2024/01/30 08:24:11 lp-1/title: Carport
[cache ] TRACE 2024/01/30 08:24:11 lp-1/mode: pv
[cache ] TRACE 2024/01/30 08:24:11 lp-1/priority: 0
[lp-1  ] INFO 2024/01/30 08:24:11 vehicle updated: unknown -> my car
[... cut ...] alue`=`excluded`.`value` 3 <nil>
[db    ] TRACE 2024/01/30 08:24:16 UPDATE `sessions` SET `created`="2024-01-30 08:24:11.59",`finished`="2024-01-30 08:24:16.414",`loadpoint`="Carport",`identifier`="",`vehicle`="my car",`odometer`=NULL,`meter_start_kwh`=NULL,`meter_end_kwh`=NULL,`charged_kwh`=0.00024214640194444444,`charge_duration`="2s",`solar_percentage`=0,`price`=NULL,`price_per_kwh`=NULL,`co2_per_kwh`=NULL WHERE `id` = 1 1 <nil>

```

Da scheint `phasesActive schon beim start auf 3 zu stehen.

StefanSchoof, 2024-01-30:
Gerade mit der 0.123.9 probiert und dort ist es richtig.

andig, 2024-01-30:
@naltatis das dürfte an https://github.com/evcc-io/evcc/pull/11571 liegen?

closed 2024-01-31 by andig