evcc-io/evcc #11934: Gerät anlegen per Ui Konfiguration nicht möglich (Kostal)

StevieC121176, 2024-01-29:
### Describe the bug

Ich wollte per UI Konfiguration meinen Kostal Plenticore anlegen, beim Druck auf den prüfen Button kommt ein Fehler. Komisch das er dort Port 502 anzeigt obwohl ich 1502 eingegeben habe.

[image]
[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
.
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

0.129.9 (0cc666ca)

andig, 2024-01-29:
Könntest du bitte mal in der Browserkonsole (=Entwicklertools) schauen, was für ein Request da geschickt wird?

StevieC121176, 2024-01-30:
> Könntest du bitte mal in der Browserkonsole (=Entwicklertools) schauen, was für ein Request da geschickt wird?

Sorry, ich steh grad auf dem Schlauch, was muss ich dafür tun?

naltatis, 2024-01-30:
Das Template hat Modbus Port 1502 als Default hinterlegt. Dieser Default wird auch korrekt ins UI übergeben und im Formular angezeigt. Im eigentlichen Validierungs/Speicheraufruf ist der Wert momentan nur dann drin, wenn der Nutzer ihn in der UI editiert hat. Also bspw. auf 1503 geändert hat. Sonst wird der Wert nicht mitgeschickt und erwartet, dass beim Ausführen/Speichern der Default des Templates (1502) verwendet wird. Stattdessen wird aber der allgemeine Modbus-Default 503 genommen.

@andig willst du das beim Entgegennehmen der Daten umstellen? Alternativ kann ich auch in der UI sicherstellen, dass Template-Defaults immer mit zurückgegeben werden, auch wenn der Nutzer sie nicht ändert.

andig, 2024-01-30:
Im Moment müssen leider such die Defaults übergeben werden. @premultiply kennt das Thema. Ich finds auch blöd, ist aber so 🙄

naltatis, 2024-01-30:
Gut, dann passe ich die Aufrufe an.

closed 2024-05-08 by andig