evcc-io/evcc #11935: Peugeot e-208 offline

Herbi54, 2024-01-29:
### Describe the bug

Seit etwa 1 Woche wird unser Auto als "(offline)" angezeigt. Regelmässige Updates und Neustarts von evcc nützen nichts. Das Auto ist über die MyPeugeot App erreichbar und zeigt den Ladeszustand korrekt an. Aber in evcc wird der Ladezustand weder angezeigt, noch kann das Ladelimit (z.B. 80%) nicht mehr eingestellt werden.  

### Steps to reproduce

1. Start evcc 
[image]

2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070
log: info
levels:
  cache: debug

# unique installation id
plant:   ***

interval: 30s # control cycle interval

sponsortoken: ***

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: sonnenbatterie 
  usage: grid
  host: 192.168.178.65 
  port: 8080  
  name: Netz
- type: template
  template: sonnenbatterie
  usage: pv
#  modbus: tcpip
#  id: 1
  host: 192.168.178.65  
  port: 8080 
  name: PV-Anlage
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.178.65  
  port: 8080  
  name: Speicher

chargers:
- type: template
  template: easee 
  user: *@*
  password: *** 
  charger: *** 
  name: Wallbox

loadpoints:
- title: Easee Wallbox
  charger: Wallbox
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 16
#  resetOnDisconnect: true

vehicles:
- type: template
  template: peugeot 
  title: Peugeot e-208  
  user: *@*  
  password: *** 
  vin: *** 
  capacity: 50  
  phases: 1  
  icon: car  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 16  
  priority: 1  
  name: E-Auto
  
site:
  title: Zuhause
  meters:
    grid: Netz
    pv:
    - PV-Anlage
    battery:
    - Speicher
#  bufferSoc: 80 # Hausbatterie wird oberhalb Soc 80% als Puffer genutzt
#  prioritySoc: 50 # Hausbatterie bekommt bis zum Soc 50% Priorität beim Laden
  residualPower: 50 # Positiver Wert: Verbleibende Netzeinspeiseleistung,
  # Negativer Wert: Verbleibende Netzbezugsleistung
```

### Log details

```text
pi@raspberrypi:~ $ sudo systemctl stop evcc
pi@raspberrypi:~ $ sudo evcc --log debug
[main  ] INFO 2024/01/29 10:42:43 evcc 0.123.9 (0cc666ca)
[main  ] INFO 2024/01/29 10:42:43 using config file: /etc/evcc.yaml
[main  ] INFO 2024/01/29 10:42:43 starting ui and api at :7070
[db    ] INFO 2024/01/29 10:42:44 using sqlite database: /root/.evcc/evcc.db
[main  ] ERROR 2024/01/29 10:42:45 creating vehicle E-Auto failed: cannot create vehicle type 'template': cannot create vehicle type 'peugeot': login failed: oauth2: "server_error" "Internal Server Error"
[lp-1  ] WARN 2024/01/29 10:42:45 deprecated: minCurrent setting is ignored, please remove
[lp-1  ] WARN 2024/01/29 10:42:45 deprecated: maxcurrent setting is ignored, please remove
[lp-1  ] WARN 2024/01/29 10:42:45 deprecated: phases setting is ignored, please remove
[lp-1  ] DEBUG 2024/01/29 10:42:46 charge total import: 7786.444kWh
[site  ] INFO 2024/01/29 10:42:46 site config:
[site  ] INFO 2024/01/29 10:42:46   meters:      grid â pv â battery â
[site  ] INFO 2024/01/29 10:42:46     grid:      power â energy â currents â
[site  ] INFO 2024/01/29 10:42:46     pv 1:      power â energy â currents â
[site  ] INFO 2024/01/29 10:42:46     battery 1: power â energy â currents â soc â capacity â
[site  ] INFO 2024/01/29 10:42:46   vehicles:
[site  ] INFO 2024/01/29 10:42:46     vehicle 1: range â finish â status â climate â wakeup â
[lp-1  ] INFO 2024/01/29 10:42:46 loadpoint 1:
[lp-1  ] INFO 2024/01/29 10:42:46   mode:        pv
[lp-1  ] INFO 2024/01/29 10:42:46   charger:     power â energy â currents â phases â wakeup â
[lp-1  ] INFO 2024/01/29 10:42:46   meters:      charge â
[lp-1  ] INFO 2024/01/29 10:42:46     charge:    power â energy â currents â
[lp-1  ] DEBUG 2024/01/29 10:42:46 phase timer inactive
[lp-1  ] DEBUG 2024/01/29 10:42:46 pv timer inactive
[lp-1  ] DEBUG 2024/01/29 10:42:46 guard timer inactive
[site  ] DEBUG 2024/01/29 10:42:46 ----
[lp-1  ] DEBUG 2024/01/29 10:42:46 charge power: 0W
[site  ] DEBUG 2024/01/29 10:42:46 pv power: 1695W
[site  ] DEBUG 2024/01/29 10:42:46 battery soc: 50%
[site  ] DEBUG 2024/01/29 10:42:46 battery power: -2W
[site  ] DEBUG 2024/01/29 10:42:46 grid power: 679W
[site  ] DEBUG 2024/01/29 10:42:46 site power: 727W
[lp-1  ] DEBUG 2024/01/29 10:42:46 charge currents: [0.209 0.01 0.015]A
[lp-1  ] DEBUG 2024/01/29 10:42:46 charge total import: 7786.444kWh
[lp-1  ] DEBUG 2024/01/29 10:42:46 charger status: B
[lp-1  ] INFO 2024/01/29 10:42:46 car connected
[lp-1  ] DEBUG 2024/01/29 10:42:46 vehicle api refresh
[lp-1  ] DEBUG 2024/01/29 10:42:46 pv timer elapse
[lp-1  ] DEBUG 2024/01/29 10:42:46 pv timer inactive
[lp-1  ] DEBUG 2024/01/29 10:42:46 charger: guard elapse
[lp-1  ] DEBUG 2024/01/29 10:42:46 guard timer inactive
[lp-1  ] DEBUG 2024/01/29 10:42:46 charge total import: 7786.444kWh
[lp-1  ] DEBUG 2024/01/29 10:42:46 switched phases: 1p
[site  ] DEBUG 2024/01/29 10:43:16 ----
[lp-1  ] DEBUG 2024/01/29 10:43:16 charge power: 0W
[site  ] DEBUG 2024/01/29 10:43:16 pv power: 1706W
[site  ] DEBUG 2024/01/29 10:43:16 battery soc: 50%
[site  ] DEBUG 2024/01/29 10:43:16 battery power: -2W
[site  ] DEBUG 2024/01/29 10:43:16 grid power: 645W
[site  ] DEBUG 2024/01/29 10:43:16 site power: 693W
[lp-1  ] DEBUG 2024/01/29 10:43:16 charge currents: [0.209 0.008 0.009]A
[lp-1  ] DEBUG 2024/01/29 10:43:16 charge total import: 7786.444kWh
[lp-1  ] DEBUG 2024/01/29 10:43:16 charger status: B
[lp-1  ] DEBUG 2024/01/29 10:43:16 pv charge current: 0A = 0A + -3.01A (693W @ 1p)
[lp-1  ] DEBUG 2024/01/29 10:43:16 pv enable timer reset
[lp-1  ] DEBUG 2024/01/29 10:43:16 pv timer inactive
[site  ] DEBUG 2024/01/29 10:43:46 ----
[lp-1  ] DEBUG 2024/01/29 10:43:46 charge power: 0W
[site  ] DEBUG 2024/01/29 10:43:46 pv power: 1709W
[site  ] DEBUG 2024/01/29 10:43:46 battery soc: 50%
[site  ] DEBUG 2024/01/29 10:43:46 battery power: -2W
[site  ] DEBUG 2024/01/29 10:43:46 grid power: 664W
[site  ] DEBUG 2024/01/29 10:43:46 site power: 712W
[lp-1  ] DEBUG 2024/01/29 10:43:46 charge currents: [0.209 0.008 0.009]A
[lp-1  ] DEBUG 2024/01/29 10:43:46 charge total import: 7786.444kWh
[lp-1  ] DEBUG 2024/01/29 10:43:46 charger status: B
[lp-1  ] DEBUG 2024/01/29 10:43:46 pv charge current: 0A = 0A + -3.1A (712W @ 1p)
[site  ] DEBUG 2024/01/29 10:44:16 ----
[lp-1  ] DEBUG 2024/01/29 10:44:16 charge power: 0W
[site  ] DEBUG 2024/01/29 10:44:16 pv power: 1712W
[site  ] DEBUG 2024/01/29 10:44:16 battery soc: 50%
[site  ] DEBUG 2024/01/29 10:44:16 battery power: -3W
[site  ] DEBUG 2024/01/29 10:44:16 grid power: 596W
[site  ] DEBUG 2024/01/29 10:44:16 site power: 643W
[lp-1  ] DEBUG 2024/01/29 10:44:16 charge currents: [0.209 0.008 0.009]A
[lp-1  ] DEBUG 2024/01/29 10:44:16 charge total import: 7786.444kWh
[lp-1  ] DEBUG 2024/01/29 10:44:16 charger status: B
[lp-1  ] DEBUG 2024/01/29 10:44:16 vehicle api refresh
[lp-1  ] DEBUG 2024/01/29 10:44:16 pv charge current: 0A = 0A + -2.8A (643W @ 1p)
[site  ] DEBUG 2024/01/29 10:44:46 ----
[lp-1  ] DEBUG 2024/01/29 10:44:46 charge power: 0W
[site  ] DEBUG 2024/01/29 10:44:46 pv power: 1715W
[site  ] DEBUG 2024/01/29 10:44:46 battery soc: 50%
[site  ] DEBUG 2024/01/29 10:44:46 battery power: -2W
[site  ] DEBUG 2024/01/29 10:44:46 grid power: 575W
[site  ] DEBUG 2024/01/29 10:44:46 site power: 623W
[lp-1  ] DEBUG 2024/01/29 10:44:46 charge currents: [0.209 0.008 0.009]A
[lp-1  ] DEBUG 2024/01/29 10:44:46 charge total import: 7786.444kWh
[lp-1  ] DEBUG 2024/01/29 10:44:46 charger status: B
[lp-1  ] DEBUG 2024/01/29 10:44:46 pv charge current: 0A = 0A + -2.71A (623W @ 1p)
[site  ] DEBUG 2024/01/29 10:45:16 ----
[lp-1  ] DEBUG 2024/01/29 10:45:16 charge power: 0W
[site  ] DEBUG 2024/01/29 10:45:16 pv power: 1721W
[site  ] DEBUG 2024/01/29 10:45:16 battery soc: 50%
[site  ] DEBUG 2024/01/29 10:45:16 battery power: -3W
[site  ] DEBUG 2024/01/29 10:45:16 grid power: 564W
[site  ] DEBUG 2024/01/29 10:45:16 site power: 611W
[lp-1  ] DEBUG 2024/01/29 10:45:16 charge currents: [0.01 0.008 0.009]A
[lp-1  ] DEBUG 2024/01/29 10:45:16 charge total import: 7786.444kWh
[lp-1  ] DEBUG 2024/01/29 10:45:16 charger status: B
[lp-1  ] DEBUG 2024/01/29 10:45:16 vehicle api refresh
[lp-1  ] DEBUG 2024/01/29 10:45:16 pv charge current: 0A = 0A + -2.66A (611W @ 1p)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.123.9 (0cc666ca)

SommesK, 2024-01-29:
Den Fehler haben mehrere (ich leider auch), siehe #11589

closed 2024-01-29 by andig

Herbi54, 2024-01-29:
Sorry, aber wie kann man etwas schließen als erledigt, wenn es immer noch nicht funktioniert???

Henrik0x7F, 2024-01-29:
@Herbi54 Ist ein Duplikat. Klick den Link von SommesK