evcc-io/evcc #11943: Easee home - Ladevorgang aktiv obwohl Fahrzeug abgesteckt

Schuttkuebl, 2024-01-29:
### Describe the bug

Habe heute das erste Mal mit evcc und der Easee Wallbox geladen. Beim Abstecken des Fahrzeugs reagiert evcc nicht und zeigt nach wie vor Ladevorgang aktiv an, teilweise wird sogar die Pausierung bei PV-Überschuss aktiviert, obwohl keine PV-Leistung mehr da ist und natürlich auch kein Auto steckt. Außerdem steht nach Ablaufen des Timers ein Fehler 400 an ("Fassade: charger disable: invalid status: 400").

Seltsam ist auch, dass bei über 5kW Erzeugungsleistung der Pausierungstimer startet, obwohl sogar mehr Leistung da wäre um mit 3,6kW zu laden.

[image]
[image]
[image]

 

### Steps to reproduce

1. Siehe Beschreibung
2. ...
3. ...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# Own sponsor key
sponsortoken: ***

# unique installation id
plant: ****

interval: 10s # control cycle interval

meters:
  - name: mygridmeter
    type: template
    template: fronius-solarapi-v1
    usage: grid
    host: 192.168.***.*** # IP-Adresse oder Hostname
    # port: 502 # Port (optional) 
  - name: pv1
    type: template
    template: fronius-solarapi-v1
    usage: pv
    host: 192.168.***.*** # IP-Adresse oder Hostname
  - name: BYDBattery
    type: template
    template: fronius-solarapi-v1
    usage: battery
    host: 192.168.***.*** # IP-Adresse oder Hostname
  - name: pv2
    type: template
    template: fronius-solarapi-v1
    usage: pv
    host: 192.168.***.*** # IP-Adresse oder Hostname

chargers:
- type: template
  template: elliconnect 
  ski: ****** 
  ip: 192.168.***.***
  name: wallbox1
- name: Easee
  type: template
  template: easee
  user: ******
  password: *******
  charger: *****

loadpoints:
- title: Garage
  charger: wallbox1
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 15
  priority: 2
- title: Fassade
  charger: Easee
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 16
  priority: 1

site:
  - title: PLS1 # display name for UI
    meters:
      grid: mygridmeter # grid meter reference
      pv: # (pvs = veraltet)
        - pv1 # first pv meter reference
        - pv2 # second pv meter reference
      battery: # (batteries = veraltet)
        - BYDBattery # battery meter reference
```

### Log details

```text
[main  ] INFO 2024/01/29 18:38:41 evcc 0.123.4
[main  ] INFO 2024/01/29 18:38:41 using config file: /etc/evcc.yaml
[main  ] INFO 2024/01/29 18:38:41 starting ui and api at :7070
[db    ] INFO 2024/01/29 18:38:42 using sqlite database: /root/.evcc/evcc.db
[eebus ] INFO 2024/01/29 18:38:42 Local SKI:  30deb9910865c472473b9864dab8bd0bde621dbc
[eebus ] DEBUG 2024/01/29 18:38:42 mdns: announce
[eebus ] DEBUG 2024/01/29 18:38:42 mdns: using avahi
[eebus ] DEBUG 2024/01/29 18:38:42 starting websocket server on :4712
[eebus ] DEBUG 2024/01/29 18:38:42 mdns: start search
[eebus ] DEBUG 2024/01/29 18:38:42 ski: 53dee593d9ff787af1efed7849106c947ebddc94 name: Elli-Wallbox-***** brand: Elli model: Wallbox typ: Wallbox identifier: Elli-Wallbox-***** register: false host: wallbox-*****.local port: 4712 addresses: [192.168.***.***]
[eebus ] DEBUG 2024/01/29 18:38:42 delaying connection to 53dee593d9ff787af1efed7849106c947ebddc94 by 1.783s to minimize double connection probability
[eebus ] DEBUG 2024/01/29 18:38:43 incoming connection request from 53dee593d9ff787af1efed7849106c947ebddc94
[eebus ] DEBUG 2024/01/29 18:38:43 Send:  read 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/01/29 18:38:43 Recv: d:_i:47859_Elli-Wallbox-*****: to NodeManagement read 42263 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/01/29 18:38:43 Send: d:_i:47859_Elli-Wallbox-***** reply 2 42263 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/01/29 18:38:43 Send: d:_i:47859_Elli-Wallbox-***** result 3 42263 ResultData 0
[eebus ] DEBUG 2024/01/29 18:38:43 Recv: d:_i:47859_Elli-Wallbox-*****: to NodeManagement reply 42265 1 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/01/29 18:38:43 Send:  call 4 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/01/29 18:38:43 Send: d:_i:47859_Elli-Wallbox-***** read 5 DeviceClassificationManufacturerData
[eebus ] DEBUG 2024/01/29 18:38:43 Send: d:_i:47859_Elli-Wallbox-***** read 6 NodeManagementUseCaseData
[eebus ] DEBUG 2024/01/29 18:38:43 Send: d:_i:47859_Elli-Wallbox-***** read 7 DeviceDiagnosisStateData
[eebus ] DEBUG 2024/01/29 18:38:43 Recv: d:_i:47859_Elli-Wallbox-*****: to NodeManagement call 42267 NodeManagementSubscriptionRequestCall
[eebus ] DEBUG 2024/01/29 18:38:43 Send: d:_i:47859_Elli-Wallbox-***** result 8 42267 ResultData 0
[eebus ] DEBUG 2024/01/29 18:38:43 Recv: d:_i:47859_Elli-Wallbox-*****: to NodeManagement result 42271 4 ResultData 0
[eebus ] DEBUG 2024/01/29 18:38:43 Recv: d:_i:47859_Elli-Wallbox-*****: to NodeManagement reply 42273 6 NodeManagementUseCaseData
[eebus ] DEBUG 2024/01/29 18:38:43 Recv: d:_i:47859_Elli-Wallbox-*****: to DeviceClassification reply 42276 5 DeviceClassificationManufacturerData
[eebus ] DEBUG 2024/01/29 18:38:43 Recv: d:_i:47859_Elli-Wallbox-*****: to DeviceDiagnosis reply 42278 7 DeviceDiagnosisStateData
[eebus ] DEBUG 2024/01/29 18:38:44 Recv: d:_i:47859_Elli-Wallbox-*****: to NodeManagement read 42279 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/01/29 18:38:44 Send: d:_i:47859_Elli-Wallbox-***** reply 9 42279 NodeManagementDetailedDiscoveryData
[eebus ] DEBUG 2024/01/29 18:38:44 Send: d:_i:47859_Elli-Wallbox-***** result 10 42279 ResultData 0
[eebus ] DEBUG 2024/01/29 18:38:44 Recv: d:_i:47859_Elli-Wallbox-*****: to NodeManagement read 42280 NodeManagementDestinationListData
[eebus ] DEBUG 2024/01/29 18:38:44 Send: d:_i:47859_Elli-Wallbox-***** reply 11 42280 NodeManagementDestinationListData
[eebus ] DEBUG 2024/01/29 18:38:44 Send: d:_i:47859_Elli-Wallbox-***** result 12 42280 ResultData 0
[eebus ] DEBUG 2024/01/29 18:38:44 Recv: d:_i:47859_Elli-Wallbox-*****: to NodeManagement read 42282 NodeManagementUseCaseData
[eebus ] DEBUG 2024/01/29 18:38:44 Send: d:_i:47859_Elli-Wallbox-***** reply 13 42282 NodeManagementUseCaseData
[eebus ] DEBUG 2024/01/29 18:38:44 Send: d:_i:47859_Elli-Wallbox-***** result 14 42282 ResultData 0
[lp-2  ] DEBUG 2024/01/29 18:38:44 charge total import: 428.490kWh
[site  ] WARN 2024/01/29 18:38:44 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2024/01/29 18:38:44 site config:
[site  ] INFO 2024/01/29 18:38:44   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/01/29 18:38:44     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/01/29 18:38:44     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/01/29 18:38:44     pv 2:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/01/29 18:38:44     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[lp-1  ] INFO 2024/01/29 18:38:44 loadpoint 1:
[lp-1  ] INFO 2024/01/29 18:38:44   mode:        pv
[lp-1  ] INFO 2024/01/29 18:38:44   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/01/29 18:38:44   meters:      charge ✗
[lp-2  ] INFO 2024/01/29 18:38:44 loadpoint 2:
[lp-2  ] INFO 2024/01/29 18:38:44   mode:        pv
[lp-2  ] INFO 2024/01/29 18:38:44   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-2  ] INFO 2024/01/29 18:38:44   meters:      charge ✓
[lp-2  ] INFO 2024/01/29 18:38:44     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/01/29 18:38:44 phase timer inactive
[lp-1  ] DEBUG 2024/01/29 18:38:44 pv timer inactive
[lp-1  ] DEBUG 2024/01/29 18:38:44 guard timer inactive
[lp-2  ] DEBUG 2024/01/29 18:38:44 phase timer inactive
[lp-2  ] DEBUG 2024/01/29 18:38:44 pv timer inactive
[lp-2  ] DEBUG 2024/01/29 18:38:44 guard timer inactive
[site  ] WARN 2024/01/29 18:38:44 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/01/29 18:38:44 ----
[lp-1  ] DEBUG 2024/01/29 18:38:44 charge power: 0W
[lp-2  ] DEBUG 2024/01/29 18:38:44 charge power: 0W
[site  ] DEBUG 2024/01/29 18:38:44 pv power: 9W
[main  ] INFO 2024/01/29 18:38:44 new version available: 0.123.9
[site  ] DEBUG 2024/01/29 18:38:44 battery soc: 67%
[site  ] DEBUG 2024/01/29 18:38:44 battery power: 6121W
[site  ] DEBUG 2024/01/29 18:38:44 grid power: -3W
[site  ] DEBUG 2024/01/29 18:38:44 site power: 6118W
[lp-1  ] DEBUG 2024/01/29 18:38:44 charger status: A
[lp-1  ] INFO 2024/01/29 18:38:44 car disconnected
[lp-1  ] DEBUG 2024/01/29 18:38:44 set session soc limit: 0
[lp-1  ] DEBUG 2024/01/29 18:38:44 set session energy limit: 0
```

### What type of operating system are you running?

Linux

### Version

0.123.4

andig, 2024-01-29:
Es braucht ein Logfile, das auch den Fehler zeigt. Geschlossen bis eines vorliegt.

closed 2024-01-29 by andig

Schuttkuebl, 2024-01-29:
[evcc.log](https://github.com/evcc-io/evcc/files/14089991/evcc.log)

~ 08:43:08 - Auto angesteckt
~ 10:00:00 - PV timer trotz genug Leistung
~ 16:45 - 17:00 Uhr Auto abgesteckt
~18:00:00 Uhr - Erster Error 400

andig, 2024-01-29:
Ich würde sagen Konfigurationsfehler:

> site power 1W >= 0W disable threshold

0W Threshold ist bei Batterienutzung eher unglücklich.