evcc-io/evcc #11952: Idee: Loadpoint/Vehicle priority per API änderbar machen

VolkerK62, 2024-01-30:
Hallo,

habe bei mir evcc begeistert und soweit erfolgreich in Betrieb nehmen können, mein Setup ist Fronius WR + Fronius Ohmpilot für Warmwasser + zur Zeit noch Wallbox Fronius Wattpilot. 
Ein Knackpunkt war die Frage, wie ich den Ohmpiloten so einbinde, dass evcc die Leistung die vom Wechselrichter dem Ohmpiloten zugewiesen wird "abzwackt" und ins Auto lädt (die Wallbox priorisiert). Hier im Forum gibt es dazu Lösungen, den Ohmpiloten als Batterie einzubinden. Und das funktioniert auch, inklusive Temperatur in [%] als SoC.
In der GUI ist dies allerdings nicht sehr schön. Zum einen ist es nun mal keine Batterie (Symbole etc., SoC nur als Prozent, nicht Temp), zum anderen taucht der Wert jetzt auch auf der "Erzeugungsseite" auf , obwohl daraus elektrisch nichts mehr entnommen werden kann. Darüber hinaus gibt es bei dem Setup auch Probleme, wenn tatsächlich noch eine Batterie im System vorhanden ist (bei mir nicht der Fall).
Da so ein Setup mit Heizstab nicht ungewöhnlich zu sein scheint, wäre es vielleicht sinnvoll, dies entsprechend in die Applikation zu integrieren?

liebe Grüße
Mario

_Originally posted by @marioth1 in https://github.com/evcc-io/evcc/discussions/11939_

VolkerK62, 2024-01-30:
In machen Fällen kann es hilfreich/notwendig/sinnvoll sein, wenn die Loadpoint/Vehicle priority im laufenden Betrieb geändert werden könnte.
Nicht immer und zu jeder Gelegenheit passt die fest zugewiesene priority aus der Config.

Ergänzend wäre zu überlegen, ob man auch AUX-Geräten eine priority zuweisen können sollte. Dann spart man sich die Konfiguration als Loadpoint.

naltatis, 2024-01-30:
Die Priorität Einstellung wird über die Config UI ohnehin kommen. Brauchen die Möglichkeit der Änderung auch im Hauptscreen bzw. im alltäglichen Betrieb? Aus dem Text im Issue wird mir der Anwendungsfall dafür noch nicht ganz klar.

VolkerK62, 2024-01-31:
>Brauchen die Möglichkeit der Änderung auch im Hauptscreen bzw. im alltäglichen Betrieb? 

Im Hauptscreen und alltäglich eher nicht. Im Allgemeinen sollten die vergebenen Prioritäten passen. 
Hier ist die Idee, die Möglichkeit zu haben, dass ohne Neustart ändern zu können, wenn es aus gegebenem Anlass zeitlich begrenzt nötig/sinnvoll wäre.

andig, 2024-03-14:
Kommt mit #12903/#12905

closed 2024-03-14 by andig