evcc-io/evcc #11953: switchsockets: manuelles Schalten aktiviert guardduration

VolkerK62, 2024-01-30:
Ich habe einen Sonoff POW320D mit Tasmota 13.3 geflasht und in die evcc.yaml eingebunden. 
Alles wird, soweit ich das beurteilen kann, im GUI korrekt angezeigt. Auch kann man das Relais (PV Produktion aktuell = 0W) über "Min+PV" oder "Schnell" einschalten. Aber klicke ich dann auf den "Aus" Button, passiert nichts. Erst wenn ich abermals auf "Min+PV" oder "Schnell" klicke und danach wieder auf "Aus" schaltet das Relais ab. Testweise habe ich noch einen Shelly Plus 1 PM (Firmware unverändert) eingebunden. Dieser zeigt ein identisches Verhalten. 

Hatte jemand das gleiche Problem und konnte es lösen?

_Originally posted by @Gelegenheitsbastler in https://github.com/evcc-io/evcc/discussions/11947_

VolkerK62, 2024-01-30:
Mit 0.123.0 funktioniert es ohne Verzögerung.
Seit 0.123.1 wird beim manuellen Schalten guard aktiviert (ohne Anzeige im UI).
Könnte https://github.com/evcc-io/evcc/pull/11170  der "Übeltäter" sein?

```
Jan 30 09:06:52 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:52 set charge mode: now
Jan 30 09:06:52 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:52 charge power: 0W
Jan 30 09:06:52 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:52 charge total import: 0.000kWh
Jan 30 09:06:53 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:53 charger status: B
Jan 30 09:06:53 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:53 charge start energy: 0kWh
Jan 30 09:06:53 raspberrypi evcc[10439]: [lp-6  ] INFO 2024/01/30 09:06:53 car connected
Jan 30 09:06:53 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:53 pv timer elapse
Jan 30 09:06:53 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:53 pv timer inactive
Jan 30 09:06:53 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:53 charger: guard elapse
Jan 30 09:06:53 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:53 guard timer inactive
Jan 30 09:06:53 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:53 charge total import: 0.000kWh
Jan 30 09:06:53 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:53 vehicle soc: 0%
Jan 30 09:06:53 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:53 set Trocknen plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
Jan 30 09:06:53 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:53 max charge current: 16A
Jan 30 09:06:53 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:53 charger enable
Jan 30 09:06:53 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:53 wake-up timer: start
Jan 30 09:06:55 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:55 charge power: 0W
Jan 30 09:06:59 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:59 set charge mode: off
Jan 30 09:06:59 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:59 pv timer reset
Jan 30 09:06:59 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:06:59 pv timer inactive
Jan 30 09:07:00 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:07:00 charge power: 0W
Jan 30 09:07:00 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:07:00 charge total import: 0.000kWh
Jan 30 09:07:00 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:07:00 charger status: A
Jan 30 09:07:00 raspberrypi evcc[10439]: [lp-6  ] INFO 2024/01/30 09:07:00 car disconnected
Jan 30 09:07:00 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:07:00 set session soc limit: 0
Jan 30 09:07:00 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:07:00 set session energy limit: 0
Jan 30 09:07:00 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:07:00 vehicle soc: 0%
Jan 30 09:07:00 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:07:00 set Trocknen plan soc: 0 @ 0001-01-01 00:53:28 +0053 LMT
Jan 30 09:07:00 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:07:00 guard enable in 53s
Jan 30 09:07:00 raspberrypi evcc[10439]: [lp-6  ] DEBUG 2024/01/30 09:07:00 charge power: 0W
```

VolkerK62, 2024-01-30:
Ich habe es auch mit meiner Wallbox getestet. Wenn ich von "aus" auf "minpv" schalte, wird auch der guard aktiviert. Wird auch im UI angezeigt. Aber ich kann problemlos ohne Verzögerung innerhalb der Zeit wieder "aus" schalten.

andig, 2024-01-30:
Auch manuelles schalten ist hin-und-herschalten und würde immer wieder Ladebeginn- und ende auslösen. Warum sollte das ausgerechnet nicht unter Guard Timer fallen?

@GrimmiMeloni wir können unabhängig davon aber mal überlegen, ob es den überhaupt noch braucht und/oder wir ihn vllt. durch einen Rate Limiter mit Burstfunktion ersetzen wollen, so dass Rate Limits auch temporär überschritten werden könnten?

GrimmiMeloni, 2024-01-30:
@andig könnten wir überlegen. Mir ist aber der Anwendungsfall vom OP nicht klar. Geht es hier um einen produktiven Workflow bei dem so schnelle Schaltvorgänge erforderlich sind? Oder ist das eher ein "ich teste/bastel/probiere gerade rum , und dehalb schalte ich manuell permanent hin und her" ?

andig, 2024-01-30:
Dito:

> Warum sollte das ausgerechnet nicht unter Guard Timer fallen?

Ich würde hier sonst schließen.

VolkerK62, 2024-01-31:
> Geht es hier um einen produktiven Workflow bei dem so schnelle Schaltvorgänge erforderlich sind? Oder ist das eher ein "ich teste/bastel/probiere gerade rum , und dehalb schalte ich manuell permanent hin und her" ?

Das (testen) war der Auslöser. 
Da ist mir dann dieses neue Verhalten aufgefallen. Wenn das ok und gewollt ist, kann man hier zu machen.

VolkerK62, 2024-01-31:
>Ich habe es auch mit meiner Wallbox getestet. Wenn ich von "aus" auf "minpv" schalte, wird auch der guard aktiviert. Wird auch im UI angezeigt.

Den Umstand find ich allerdings etwas verwirrend, dass nach dem Einschalten die "Schaltschutz" Anzeige im UI kommt.
Ich vermute, dass hat mit der Trennung von Guard und PV Timer zu tun  https://github.com/evcc-io/evcc/pull/11926

GrimmiMeloni, 2024-01-31:
> Den Umstand find ich allerdings etwas verwirrend, dass nach dem Einschalten die "Schaltschutz" Anzeige im UI kommt.
> Ich vermute, dass hat mit der Trennung von Guard und PV Timer zu tun #11926

Vorsicht: Missverständnis. PVTimer und GuardTimer waren schon immer getrennt. #11926 adressiert lediglich das in #11123 beobachtete Verhalten, daß in bestimmten Konstellationen der GuardTimer nicht mehr gegenüber der UI publiziert wurde, obwohl er noch aktiv war. 
Vielleicht gibt es hier noch einen unbeabsichtigten Seiteneffekt. Ich schaue mir das nochmal an.

VolkerK62, 2024-01-31:
Ich habe mal mit der Demo gespielt.  Wenn man per Hand einen der PV-Modi (PV oder Min+PV) auswählt wird "Schaltschutz" angezeigt. Wählt man "Schnell", dann nicht.

GrimmiMeloni, 2024-02-01:
Danke fürs reproduzieren, @VolkerK62.
Ich kann das Verhalten auch nachvollziehen. Und ja, es ist ein Effekt von #11926.

Was vorher wie nachher gleich geblieben ist: Der GuardTimer wird beim Enable bzw. Disable eines Chargers gesetzt, und läuft dann runter.
Was sich verändert hat, ist das früher der Guard Timer nur auf den Bus gepublished wurde, wenn auch ein weiterer Schaltvorgang erforderlich ist. Sonst ist der GuardTimer quasi "still" im Hintergrund gelaufen und die UI hat den überhaupt nicht mitbekommen.

Das ist jetzt verändert, weil wir den GuardTimer so fern er läuft, jetzt immer publishen. Und dies führt zu dem Effekt, daß er immer nach einem Schaltvorgang in der UI angezeigt wird.

Das ist jetzt erstmal nur kosmetisch, aber irritierend, und sollten wir nochmal ändern. Ich überlege mir mal wie wir sowohl dieses Verhalten verhindern können und gleichzeitig auch Dein ursprüngliches Problem aus #11123 - das der Guard Timer gar nicht mehr verschwindet - lösen können.

GrimmiMeloni, 2024-02-01:
Puuuhh. Das ist echt haarig. 

Ich habe jetzt mehrere Varianten probiert.

Den GuardTimer künstlich mit Laufzeit 0 zu publishen bringt zwar das gewünschte Ergebnis das die UI ihn nicht mehr anzeigt, obwohl er noch Serverseitig läuft , aber dann publishen wir auch bewusst etwas falsches. Das geht zwar nur an die UI, aber so richtig toll ist es nicht, da dann als Seiteneffekt im Server Log auch immer `guardTimer inactive` steht, obwohl der GuardTimer noch runter läuft.
 
Andererseits gibt es aber auch keine andere Variante die ich sehe, da die UI nicht weiß ob ein Schaltvorgang nicht mehr erforderlich ist, und somit diese auch nicht weiß, ob sie den Schaltschutz anzeigen sollte oder nicht.

Wir bräuchten eine Möglichkeit einerseits den guardTimer sauber zu publishen, gleichzeitig müssten wir aber auch der UI signalisieren können ob der Schaltvorgang überhaupt geblockt wurde.

@naltatis kann ich über irgendeine Nachricht auf dem Ui Channel der UI signalisieren das sie den GuardTimer nicht anzeigen soll, obwohl dieser >0 ist?

VolkerK62, 2024-02-02:
Vielen Dank @GrimmiMeloni  für die ausführlichen Erklärungen

GrimmiMeloni, 2024-02-09:
Nur der Übersicht halber. Aktuell diskutieren wir die mögliche Lösung in https://github.com/evcc-io/evcc/pull/12084

naltatis, 2024-02-11:
Ist mit https://github.com/evcc-io/evcc/pull/12084 erledigt und kann zu richtig? Oder übersehe ich was?

closed 2024-02-11 by andig