evcc-io/evcc #11958: Laden startet nicht beim angeschlossenen Fahrzeug

julred, 2024-01-30:
### Describe the bug

Hallo zusammen,

ich nutze die cfos Brain Power Solar iVm evcc.
Ich habe nun des häufigeren beobachtet, dass ein Ladevorgang der aufgrund eines mangelnden PV Überschusses gestoppt werden musste, das Fahrzeug über Nacht angeschlossen bleibt, am Folgetag den Ladevorgang nicht mehr startet. Es ist egal ob der Modus auf PV Überschuss bleibt oder ob Schnellladen gewählt wurde.
Um den Ladevorgang händisch zu starten, muss ich das Ladekabel entfernen und wieder einstecken.

Ich war bereits mit dem cfos Support im Kontakt, da ich glaubte dass das Auto nicht korrekt "geweckt" wird.

Hier die Rückmeldung:
```

 Guten Tag,

betreiben Sie denn die Wallbox im "Beobachten"-Modus?
Das müssen Sie wenn Sie die Wallbox per evcc steuern möchten, Dann dürfte "aufwachen", etc. aber nicht funktionieren. 
evcc müsste dies selbst implementieren. Dazu gibt es Modbus Register 8086.
vgl. https://www.cfos-emobility.de/de/cfos-power-brain/modbus-registers.htm

Die Frage die sich unsere Techniker stellen  ist, ob sein Auto überhaupt einschläft. 
Da wird eher der evcc support helfen müssen.

```

Schaut so aus, dass man ein zusätzliches Modbus Register in die Kommunikationsroutine aufnehmen müsste.
Hat jemand noch diese Phänomene?
P.S: im Webinterface der cfos Wallbox gibt es Parameter wie "Auto aufwecken" (aktiviert) und "Auto schlafen lassen" (deaktiviert). Diese sollen wohl laut cfos Support auch nur funktionieren, wenn die Wallbox mit dem eigenen Lastmanagement läuft - also nicht in Verbindung mit evcc.

### Steps to reproduce

Nur sporadisch
1. Auto mittels PV Überschuss laden
2. Ladung stoppt, aufgrund fehlendem PV Überschuss
3. Auto bleibt über Nacht angeschlossen
4. Am nächsten Tag startet - trotz genügend Überschuss oder bei Schnell Laden der Ladevorgang nicht
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: ****

interval: 10s # control cycle interval

sponsortoken: *****
# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

mqtt:
  broker: 192.168.178.15:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: mqtt
  password: *****

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.178.129
  name: grid1
- type: template
  template: sma-hybrid 
  id: 3  
  host: 192.168.178.77  
  port: 502  
  usage: pv  
  modbus: tcpip  
  name: pv2
- type: template
  template: sma-inverter 
  usage: pv  
  host: 192.168.178.128  
  password: ***  
  name: pv3
- type: template
  template: sma-hybrid 
  id: 3  
  host: 192.168.178.77  
  port: 502  
  usage: battery  
  modbus: tcpip  
  name: battery4
- type: template
  template: cfos 
  id: 1
  host: 192.168.178.73  
  port: 4701
  usage: charge  
  modbus: tcpip  
  name: charge_brain_solar
- type: template
  template: cfos
  id: 1
  host: 192.168.178.74
  port: 4701
  usage: charge
  modbus: tcpip
  name: charge_brain

chargers:
- type: template
  template: cfos 
  host: 192.168.178.73  
  name: cfos_brain_solar
- type: template
  templatE: cfos
  host: 192.168.178.74
  name: cfos_brain

vehicles:
- type: template
  template: offline 
  title: eGolf  
  capacity: 32  
  name: eGolf
  icon: car
  phases: 2
  mode: pv
  minCurrent: 6
  maxCurrent: 16

loadpoints:
- title: Garage eGolf
  charger: cfos_brain_solar
  meter: charge_brain_solar
  vehicle: eGolf
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
- title: Garage Gast
  charger: cfos_brain
  meter: charge_brain
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16

site:
  title: Home
  meters:
    grid: grid1
    pv:
    - pv2
    - pv3
    battery:
    - battery4

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.301 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.082 # [currency]/kWh

hems:
  type: sma
  AllowControl: false
```

### Log details

```text
Der Log liegt hier:
https://pub.microbin.eu/upload/snake-raven-snake
```

### What type of operating system are you running?

Linux

### Version

0.123.9

andig, 2024-01-30:
> betreiben Sie denn die Wallbox im "Beobachten"-Modus? Das müssen Sie wenn Sie die Wallbox per evcc steuern möchten,

Dieser Satz keinen Sinn. Ein solcher Modus wird auf https://www.cfos-emobility.de/de/cfos-power-brain/modbus-registers.htm mit keinem Wort erwähnt. Was meint der Support damit?

@premultiply 8086 ist CP-Unterbrechung. Wäre das als Wakeup sinnvoll?

julred, 2024-01-31:
Finde die Doku auch etwas unglücklich auf cfos. Der Beobachten Modus ist jener Modus wenn das cfos eigene Lastmanagement deaktiviert ist und auf externe Vorgaben wie bspw evcc lauscht. Sehe das gewünschte aufweckverhalten eher im Leistungsumfang der cfos box. 
@andig 
Wie wird das bei anderen Herstellern umgesetzt

premultiply, 2024-01-31:
> 8086 ist CP-Unterbrechung. Wäre das als Wakeup sinnvoll?

Offensichtlich.
Warum die Box das nicht selber macht bleibt wohl das Geheimnis des Herstellers.

julred, 2024-01-31:
> > 8086 ist CP-Unterbrechung. Wäre das als Wakeup sinnvoll?
> 
> Offensichtlich. Warum die Box das nicht selber macht bleibt wohl das Geheimnis des Herstellers.

Habe dem Support geschrieben, dass ich es als Enduser in der Verantwortung des Wallbox Herstellers sehe. Da er ohnehin diese Funktion bei sich implementiert hat - aber nur wenn der hauseigene Lastmanager benutzt wird.
Bei der Elli Wallbox habe ich das Problem nie festgestellt - scheint daher für mich ein cfos Thema - statt eines evcc Themas zu sein.
Können andere cfos Nutzer das gleiche sporadische Verhalten feststellen?

andig, 2024-01-31:
@premultiply wieviele Sekunden brauchen wir? Ist ja schnell eingebaut. 5?

closed 2024-02-03 by andig

julred, 2024-02-09:
@andig ich habe eine Rückmeldung vom Cfos Support erhalten bzgl. der Timings vom cp Pin:
```
Solange die Wallbox sich im disconnected-Zustand befindet, ist der
EVSE-status (Register 8092) eingefroren. Nach Ablauf der disconnect-Zeit
(verbleibende Zeit kann auch aus demselben Register ausgelesen werden),
läuft alles normal weiter. Man sollte also im wesentlichen nur darauf
achten, nicht innerhalb der Wartezeit weitere wake-up-Versuche zu starten.
```

Grade der letzte Satz lässt mich aufhorchen. Wie ist es momentan in EVCC realisiert? Gibt es eine State-Machine bzgl. des Wake-Ups? Wird der Wake-Up regelmäßig getriggert, wenn auf das Fahrzeug gewartet wird?

premultiply, 2024-02-10:
Aktuell wird nur 1x nach 30 Sekunden geweckt.

julred, 2024-02-10:
> Aktuell wird nur 1x nach 30 Sekunden geweckt.

Ich kann auf jeden Fall bestätigen, dass der wakeup im aktuellen Fall Grade geklappt hat. Das Auto war seit gestern 23h dran und hat pauschal 2kwh zum testen geladen. Heute morgen habe ich nochmal 2kwh geladen. Gefühlt dauerte es zwischen Aktivierung der Ladung und Starten des Ladevorgangs ca 3-4minuten. Sind die 30 Sekunden wirklich so einprogrammiert @premultiply ?

premultiply, 2024-02-10:
Ja, falls der charger **extern** von evcc zum aufwecken angetriggert wird. Ansonsten ist das von der Wallbox-Firmware abhängig.

sebster6, 2024-02-28:
@julred Tritt das ursprüngliche Problem bei dir nun nicht mehr auf? Ich hatte gestern und heute  das Verhalten, dass nach einer Ladepause nicht wieder begonnen wurde (nur noch „Warte auf Fahrzeug…“ bis zum Neuanstecken des Fahrzeugs).
Hier hängt ein Cupra Born an der cfos Power Brain Solar. Ich wollte morgen mal tiefer in die Logfiles einsteigen, nachdem ich bis jetzt das Auto im Verdacht sah, das Problem zu sein.

julred, 2024-02-29:
> @julred Tritt das ursprüngliche Problem bei dir nun nicht mehr auf? Ich hatte gestern und heute das Verhalten, dass nach einer Ladepause nicht wieder begonnen wurde (nur noch „Warte auf Fahrzeug…“ [...]

Hin und wieder habe ich nochmal den Effekt. Ich schiebe es momentan noch auf ein nicht geupdatetes Ladesteuergerät von meinem eGolf. Sehe aber auch ab und zu dass er versucht zu laden (CP Kontakt wird 0V gesetzt) und zieht dann kurzzeitig 4W...

ThiloGa, 2024-03-24:
> Können andere cfos Nutzer das gleiche sporadische Verhalten feststellen?

Jap.

cmock, 2024-06-05:
ich hab das verhalten auch, mit evcc 0.126.5; wenn ich an evcc vorbei den wert 10 in register 8086 schreibe, wacht das auto (Smart #1) auf und fängt an zu laden.

cmock, 2024-06-06:
neuer sonnenaufgang, neue investigation: im code sind ja offenbar 5 sekunden CP-abschaltung hinterlegt, das hab ich heute der wallbox geschickt, und sie weckt das auto auch mit diesem wert auf.

offenbar wird das also im code nicht getriggert, aber ich habs noch nicht geschafft zu verstehen, unter welchen umständen das Resurrector-API aufgerufen wird.

kann es daran liegen, daß der Smart #1 im schlafzustand bei anliegender spannung nicht gar nix, sondern 1.3A bzw 5W zieht?

julred, 2024-07-10:
Mit der neusten Beta Firmware der cfos Wallbox (2.1.8) ist die Unterstützung EV40 EV-Ready mit reingekommen. Das Feature soll das Aufwecken der Fahrzeuge realisieren.

Zitat cfos:

> EV-Ready ist eine Spezifikation, die hauptsächlichem von französichen
> Auto-Herstellern verwendet wird.
> 
> Nach der technischen Spezifikation
> https://www.asefa-cert.com/medias/ev-ready_certification_technical_reference_document_v2.0a.pdf
> 
> soll zum Aufwecken die CP-Leitung für 3 Sekunden auf -12 V geschaltet
> werden (EV40).
> 
> Mit der angegeben Option wird dies statt dem simulierten Ausstecken
> durchgeführt. Dies wirkt sich auch auf das von EVCC verwendete Kommando
> zum Aufwecken aus.

premultiply, 2024-07-10:
Wenn es richtig implementiert wäre bräuchte es dazu keinerlei externes Kommando sondern die EVSE würde dies wie in der Spec dargestellt selbstständig tun.

Viele Hersteller tun dies bereits ganz selbstverständlich und ohne großes Getöse.

julred, 2024-07-10:
> Wenn es richtig implementiert wäre bräuchte es dazu keinerlei externes Kommando sondern die EVSE würde dies wie in der Spec dargestellt selbstständig tun.
> 
> Viele Hersteller tun dies bereits ganz selbstverständlich und ohne großes Getöse.

Ich würde gerne dieses Feature gerne testen - das kollidiert jedoch mit dem evcc retry command. Kann ich das testweise in evcc deaktivieren?

premultiply, 2024-07-10:
Man könnte https://github.com/evcc-io/evcc/pull/12033 zurückdrehen.

julred, 2024-07-10:
> Man könnte #12033 zurückdrehen.

Könnte man das testweise durchführen? Ich habe leider keine buildchain für go hier am start - oder könnte man den pv retry über einen parameter in der chargerbeschreibung einstellen?