evcc-io/evcc #11972: Tariffs / co2 / electricitymaps not working. Potential API change

jangli78, 2024-01-31:
### Describe the bug

Unable to make co2 tariffs working using electricitymaps. Potential API and/or subscription changes from electricitymaps

### Steps to reproduce

see https://github.com/evcc-io/evcc/discussions/11949

### Configuration details

```yaml
tariffs:
co2:
type: electricitymaps
uri: https://api-access.electricitymaps.com/free-tier/carbon-intensity/latest?zone=LU
token: O.....6
zone: LU
```

### Log details

```text
[em ] ERROR 2024/01/29 22:21:10 unexpected status: 400 (Bad Request)
[main ] ERROR 2024/01/29 22:21:10 failed configuring co2 tariff: cannot create tariff type 'electricitymaps': unexpected status: 400 (Bad Request)
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.123.9

StefanSchoof, 2024-01-31:
I think a important fact is that there is now a free personal tier: https://api-portal.electricitymaps.com/

andig, 2024-01-31:
The free tier does not have forecasts and is not supported.

closed 2024-01-31 by andig