evcc-io/evcc #11984: Tesla New API can't login

AnBo83, 2024-02-01:
### Describe the bug

With the 'new' Tesla API, a login is not possible in Version 0.124.0

### Steps to reproduce

1. Update Config to "template: tesla-command"
2. Get new Access/Refreshtokens
3. Update Config
4. Restart EVCC
5. Cars marked as offline, Logfile says: "cannot create vehicle type 'template': cannot create vehicle type 'tesla-command': oauth2: "login_required" "Login required""
...

### Configuration details

```yaml
vehicles:
- type: template
  template: tesla-command
  title: Tesla1
  accessToken: ***
  refreshToken: ***
  vin: **
  capacity: 77
  name: ev1
  mode: pv
  priority: 10
```

### Log details

```text
cannot create vehicle type 'template': cannot create vehicle type 'tesla-command': oauth2: "login_required" "Login required"
```

### What type of operating system are you running?

Linux

### Version

0.124.0

litithio, 2024-02-01:
Did you create the new tokens using https://tesla.evcc.io/ or how did you create the tokens?

AnBo83, 2024-02-01:
yes, i create new tokens using https://tesla.evcc.io. The tokens are working for 1 time after restart or disconnect the vehicle the tokens doesn't work and the errormessage appears.

andig, 2024-02-01:
Make sure the pasted Tesla tokens are correct. If yes share your full config with info@evcc.io for analysis.

closed 2024-02-01 by andig