evcc-io/evcc #11985: Change to new Tesla API doesn't work

MasterXDave, 2024-02-01:
### Describe the bug

I just changed to to new API. I used the manual from the Blog from cvcc. Changed the lvcc.yaml and generated new Token. 
When I start lvcc, I get the response:
[main  ] FATAL 2024/02/01 12:00:03 cannot create vehicle 'my_car': cannot create vehicle type 'template': invalid key: minsoc

### Steps to reproduce

1.With every single start
2.
3.
...

### Configuration details

```yaml
vehicles:
  - name: my_car
    type: template
    template: tesla-command #tesla-command # change `tesla` to `tesla-command`
    title: Irina 3 # Wird in der Benutzeroberfläche angezeigt # Optional
    accessToken: XXXXX
    refreshToken: XXXXXXXXXXXXXX
    vin: XXXXXXXXXX # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 78 # Akkukapazität in kWh # Optional 
    minCurrent: 4
    maxCurrent: 16
    #onIdentify: # set defaults when vehicle is identified
    # mode: pv # enable PV-charging when vehicle is identified
    minSoc: 20 # immediately charge to 20% regardless of mode unless "off" (disabled)
    targetSoc: 80 # limit charge to 80%
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/02/01 12:06:07 evcc 0.124.0
[main  ] INFO 2024/02/01 12:06:07 using config file: /config/evcc.yaml
[main  ] INFO 2024/02/01 12:06:07 starting ui and api at :7070
[main  ] FATAL 2024/02/01 12:06:07 cannot create vehicle 'my_car': cannot create vehicle type 'template': invalid key: minsoc
[main  ] FATAL 2024/02/01 12:06:07 will attempt restart in: 5m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

litithio, 2024-02-01:
minSoc and targetSoc have been removed from evcc.yaml just like the error message says. They are now configured via the ui. Remove them and try again.

MasterXDave, 2024-02-01:
This one word thank you. 
But I guess it is still not working properly. I have the message : "Hof: max charge current 16A: vehicle not capable of current control" and if I switch to "Schnell" it says "connected" but nothing happens

litithio, 2024-02-01:
Controlling the vehicle via the new API is not yet working and still under development. See #11986 and #11893.

andig, 2024-02-01:
> invalid key: minsoc

See https://github.com/evcc-io/evcc/discussions/10978

andig, 2024-02-01:
> vehicle not capable of current control

See https://github.com/evcc-io/evcc/issues/11846

closed 2024-02-01 by andig