evcc-io/evcc #11987: Tesla Command: oauth2: "login_required"

andig, 2024-02-01:
Moin zusammen,

ich habe auf 0.124.0 aktiualisiert und wollte danach unseren Tesla über die neue API anbinden. Token habe ich generiert und access aus alles, bis auf die Energieprodukte erlaubt:

[image]

Wenn ich jedoch den Zugang in evcc erstellen möchte (egal ob via gui oder config.yaml) bekomme ich folgende Fehlermeldung:

`main  ] ERROR 2024/02/01 09:25:20 creating vehicle molly failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla-command': oauth2: "login_required" "Login required"
`

Irgendjemand, der etwas Licht in mein Dunkel bringen könnte?

Die config sieht so aus:
````
Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- name: grid
  type: template
  template: discovergy
  usage: grid
  user: *****
  password: *****
  meter: 1ESY1161231764

- name: pvgarten
  type: template
  template: tasmota
  usage: pv
  host: 192.168.6.65

- name: pvstrasse
  type: template
  template: tasmota
  usage: pv
  host: 192.168.6.16

chargers:
- type: template
  template: alfen
  id: 1
  host: 192.168.1.130
  port: 502
  modbus: tcpip
  name: alfen

vehicles:
  - name: molly
    type: template
    template: tesla-command
    title: Molly # Wird in der Benutzeroberfläche angezeigt (Optional)
    accessToken: *****
    refreshToken: *****
    vin: *****
    capacity: 76.8 # Akkukapazität in kWh (Optional) 
    #minSoC: 40
    mode: now
    mincurrent: 6
 #  icon: car
  #- name: robert
  #  type: template
  #  template: kia
  #  title: Robert
  #  user: *****
  #  password: *****
  #  #capacity: 
  #  mode: now
  #  mincurrent: 6
  #  icon: car

loadpoints:
  - title: Wallbox # display name for UI
    charger: alfen # charger
    #mode: now
    # vehicle: car1 # set default vehicle (disables vehicle detection)
    #resetOnDisconnect: true # set defaults when vehicle disconnects
    phases: 0 # electrical connection (normal charger: default 3 for 3 phase, 1p3p charger: 0 for "auto" or 1/3 for fixed phases)
    minCurrent: 6 # minimum charge current (default 6A)
    maxCurrent: 16 # maximum charge current (default 16A)

site:
  title: Mein Zuhause
  meters:
    grid: grid
    pv:
      - pvgarten
      - pvstrasse

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.31
  feedin:
    type: fixed
    price: 0.00
  #planner:
  #  type: grünstromindex
  #  zip: 61206

mqtt:
  broker: 192.168.1.219:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: *****
  password: *****

# Message config using evcc go-text-template rendering, evcc variables and sprig-functions
messaging:
  events:
    start: # charge start event
      title: "laden gestartet: {{.vehicleTitle}}"
      msg: |
        {{.title}} laden von {{.vehicleTitle}} gestartet - {{ toString .mode | upper }} Modus.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    stop: # charge stop event
      title: "laden beendet: {{.vehicleTitle}}"
      msg: |
        {{.title}} laden von {{.vehicleTitle}} beendet
        mit  {{round (divf .chargedEnergy 1000) 2 }} kWh in {{.chargeDuration}}.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    connect: # vehicle connect event
      title: "{{.vehicleTitle}} verbunden mit {{.title}}"
      msg: |
        {{.vehicleTitle}} an {{.title}} angeschlossen - {{round (divf .pvPower 1000) 2 }} kW PV.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}
    disconnect: # vehicle connected event
      title: "{{.vehicleTitle}} getrennt von {{.title}}"
      msg: |
        {{.vehicleTitle}} getrennt von {{.title}} nach {{.connectedDuration}}.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        Solar-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Eigenverbrauch: {{round (divf .homePower 1000) 3 }} kW
        {{if .batteryConfigured}}Batteriespeicher-Status: {{round (divf .batteryPower 1000) 3 }} kW ({{.batterySoc }} %){{end}}

  services:
    - type: pushover
      app: *****
      recipients: *****
        - uojiaynfxusu46ktm1h3fv82jahke2
#    - type: telegram
#      token: *****
#      chats: *****
#        - 2023412556

modbusproxy:
  - port: 1502
    uri: 192.168.1.130:502

Version: `0.124.0`
```

_Originally posted by @boehmgui in https://github.com/evcc-io/evcc/discussions/11983_

AnBo83, 2024-02-01:
Kann es sein das euer Dev-Account bei Tesla nur "Einmal-Tokens" generieren kann? Beim erstellen der Tokens funktioniert es ja, wird der Token ein zweites Mal benutzt fehlt die Authentifizierung

andig, 2024-02-01:
Nein

Servicetechnik, 2024-02-01:
> Kann es sein das euer Dev-Account bei Tesla nur "Einmal-Tokens" generieren kann? Beim erstellen der Tokens funktioniert es ja, wird der Token ein zweites Mal benutzt fehlt die Authentifizierung

Ja, das Problem habe ich. Die Fahrzeugdaten werden beim neu erstellten Token einmal abgerufen, dann war es das.
EVCC hat dann die ganzen Werte (SOC etc.), aber es lässt sich weiter nichts steuern und auch nichts weiter abrufen bis man ein neues Token von tesla.evcc.io erstellt und hinterlegt in der .yaml

Gibt es hierfür eine Lösung?

josua832000, 2024-02-01:
Hallo, ich habe genau das gleiche Problem wie AnBo83,
Beim ersten Start ist es gut, bei weiteren Starts funktioniert es nicht mehr,
Ich muss einen Token neu generieren
(Ich verwende Docker Build V0.124.0)

Die Lastkontrolle erfolgt nicht in allen Fällen

andig, 2024-02-01:
Looking at https://github.com/teslamotors/vehicle-command/issues/132, a password change would invalidate the token and trigger this error. Unfortunately, Tesla's error message does not make this clear.

Could that be the case here?

AnBo83, 2024-02-01:
Passwörter wurden nicht geändert. Der Token schein nach der ersten Benutzung die Gültigkeit zu verlieren.

andig, 2024-02-01:
Note: also noted that the structure of the refresh tokens has changed. They are no longer JWT tokens but start with `EU_` or `NA_`. Maybe this introduced an error.

andig, 2024-02-01:
> Passwörter wurden nicht geändert. Der Token schein nach der ersten Benutzung die Gültigkeit zu verlieren.

Das ist irrelevant da die Tokens dann refreshed werden. Genau das scheint aber nicht zu funktionieren.

Thias66, 2024-02-01:
Same here, the refreshToken starts with EU_*****
First use of evcc vehicle shows vehicle data, 

second use of evcc vehicle produce: [main  ] ERROR 2024/02/01 14:24:10 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla-command': oauth2: "login_required" "Login required"
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'tesla-command': oauth2: "login_required" "Login required"

andig, 2024-02-01:
Oh wait a second! I suspect that the Tesla api has changed once more.

I suspect that the refresh token is invalidated after being used once. To validate that I'd need NEW refresh token that is never used for testing. Feel free to reach out to info@evcc.io.

robberburn, 2024-02-01:
Das gleiche hier. Authentifizierung-Token nur einmal gültig

andig, 2024-02-01:
@all please stop posting "me too".

andig, 2024-02-01:
Confirmed thanks to @AnBo83. We'll need to store the refresh tokens. Looking into this now.

RTTTC, 2024-02-02:
Will we need to re-add the vehicle after fix is found?

AnBo83, 2024-02-02:
@andig hier nun offiziell von Tesla:
https://developer.tesla.com/docs/fleet-api#2024-02-01-auth-access-token-policy-change

> As part of a security improvement, the acquisition of a new Refresh token will now render the user’s previous Refresh token and its associated Access tokens invalid.

Das ganze wurde gestern seitens Tesla aktiviert weshalb wir wohl auch seit gestern die Probleme haben.

jori0001, 2024-02-02:
Meine neuen Token beginnen beide mit "eyJh..." also nicht "EU_" oder "NA_". Erstellt vor ein paar Tagen funktionieren sie mit der v0.124.0 weiterhin. Nur der Vollständigkeit halber, falls das von Belangen sein sollt...

andig, 2024-02-02:
Seit gestern erstellt? Das würde mich wundern.

jori0001, 2024-02-02:
Nein nein, vor 3-4 Tagen war das meine. Da hatte ich das nightly mal installiert und da drüber dann die v0.124.0.
Muss das Ganze noch bei nem 2. Tesla machen. Vermutlich laufe ich da dann auch in die hier beschriebenen Probleme.
Die Frage ist nur, wenn ihr das fixt, müssen diejenigen, bei denen es funktioniert (so wie bei mir) nochmal neue Token erstellen oder ist dann nichts zu tun?

andig, 2024-02-03:
Im nightly sollte es ab morgen möglich sein, Tokens zu speichern und wiederzuverwenden (ungetestet). Um zu schauen, was gespeichert ist genügt ein 

    evcc settings get tesla-command

Mit dem Befehl lassen sich auch andere Konfigurationssettings auf der Kommandozeile auslesen 

/cc @VolkerK62

closed 2024-02-03 by andig

Presl1983, 2024-02-04:
Hallo,

kann es sein das es noch immer nicht geht?
Ich bekomme aktuell beim ersten mal starten von evcc diese Meldung :/

[main  ] ERROR 2024/02/04 15:01:08 creating vehicle my_car failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla-command': cannot find vehicle, got: [5YJ3xxxxxxxx8738]

danach wenn ich nochmal evcc in HA starte kommt diese Meldung:

[main  ] ERROR 2024/02/04 15:02:44 creating vehicle my_car failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla-command': oauth2: "login_required" "Login required"

mucki12, 2024-02-04:
@Presl1983 
Hast du es denn mit der aktuellsten Nightly getestet?

Presl1983, 2024-02-04:
darf ich fragen was ein Nightly ist?
Hätte es gestern schon versucht rauszufinden :/

mucki12, 2024-02-04:
Das sind "Zwischenversionen". In das offizielle Release ist der Fix ja noch gar nicht drinnen - kann also gar nicht gehen. Müsstest du mal testhalber die letzte Nightly installieren:
https://github.com/evcc-io/evcc/actions/workflows/nightly.yml?query=workflow%3ABuild
https://docs.evcc.io/docs/installation/linux/#erstinstallation

Presl1983, 2024-02-04:
danke für die Aufklärung :)