evcc-io/evcc #12009: Skalierung Ladestromvorgabe

BrickTop87, 2024-02-02:
**Is your feature request related to a problem? Please describe.**
Ich habe mittlerweile herausgefunden, dass mein Auto nicht genau der Ladestromvorgabe folgt, sondern auf die Vorgabe einen Faktor von 15/16 anwendet. Das heißt z.B. konkret: evcc gibt über die Wallbox 16 A vor, das Auto lädt mit 15 A. Bei Vorgabe 6 A dann konsequenterweise ~5,6 A. Das lässt sich mit verschiedenen Stromvorgaben gut reproduzieren. 

Die Regelung des PV-Überschussladens stört es kaum, weil die immer die Differenz Soll/Ist am Grid-Meter auf die Ladestrom-Sollvorgabe aufschlägt und irgendwann schwingt sich das schon ein.

Da ich für den Loadpoint aber keinen eigenen Meter habe (und auch nicht zwingend brauche) gibt es folgende (kosmetische) Probleme:
1. Die Energiebilanz während des Ladens stimmt nicht. Der Bedarf des Autos wird überschätzt und folglich der des Haushalts unterschätzt. Bei 3ph 16 A liegt die Bilanz dadurch immerhin um fast 700 W daneben.
2. Die aufintegrierte Ladeenergie ist zu hoch.

**Describe the solution you'd like**
Vorweg: Mir ist klar, dass man das ohne Meter nicht zu 100% korrigieren kann. Neben dem Strom ist ja auch die Spannung variabel und die kann evcc ohne Meter nur als 230 V annehmen.

Ich schlage als Lösung eine (Experten-)Setting im YAML vor, mit der man für das Fahrzeug einen Skalierungsfaktor für den Strom vorgeben kann mit Default 1. Ich müsste dann 16/15 ~ 1,067 vorgeben.
evcc arbeitet dann intern mit dem "geringeren" (in meinem Fall) Strom und erst beim Rausschreiben des Sollwerts an die Wallbox wird der Faktor angewendet.
Energiebilanz und Ladeenergie sollten dann besser zur Realität passen. Evtl. profitiert auch die Regelschleife etwas, aber das muss man vermutlich testen.

**Describe alternatives you've considered**
Alternativ kann ich auch mit der Ungenauigkeit leben, es ist auf jeden Fall eher ein "nice to have"-Thema und nicht dringlich.

**Additional context**
Vielleicht haben andere User auch schon ein ähnliches Verhalten bemerkt und man könnte eine kleine Datenbank machen, welchen Fahrzeugtyp man wie "vertrimmen" muss?

andig, 2024-02-02:
Ich glaube nicht, dass wir das machen wollen. Das Problem falscher Abschätzung ist aufgrund des fehlenden Zählers prinzipbedingt und rein optisch. Mehr (Experten-)Settings sind nicht die Lösung. Weitere Alternativen hier wären:
- Shelly als Zwischenzähler installieren
- Custom charger mit entsprechender Umsetzung
- Nachkorrektur im Ladelog

closed 2024-02-02 by andig

BrickTop87, 2024-02-05:
Verstehe ich. Das geht schon sehr in Richtung Feinoptimierung und ihr wollt die Optionen nicht noch weiter aufblasen.

Falls jemand etwas Ähnliches machen möchte hier kurz meine Lösung:
Ich habe mir die Toolchain eingerichtet, um evcc selbst zu kompilieren. Im charger habe ich die Berechnung des Soll-Stroms direkt vor Kommunikation an die Wallbox entsprechend modifiziert (in meinem Fall [hier](https://github.com/evcc-io/evcc/blob/113ba95363b93bd8f37584ee492b86f107c2b8e5/charger/wallbe.go#L161) und [hier](https://github.com/evcc-io/evcc/blob/113ba95363b93bd8f37584ee492b86f107c2b8e5/charger/wallbe.go#L149)).
Bei neuen Releases muss ich zwar jetzt selbst neu bauen, das habe ich aber inklusive Patch für das eine File nun automatisiert.