evcc-io/evcc #12013: Phasenumschaltung wird für Heidelberg Energy Control angeboten, funktioniert aber nicht

andreashm, 2024-02-02:
### Describe the bug

Bis Version 0.123.9 lief es korrekt:
Bei Druck auf Schieberegler konnte man sehen, ob 11kW oder 3,7kW angeschlossen waren (sprich: Ob die Box 1p oder 3p mit Strom versorgt wurde). Das wurde auch für die PV-Steuerung entsprechend berücksichtigt.

Seit Version 0.124.0 gibt es hier einen Bug:
Die Box wird nicht mehr nach den angeschlossenen Phasen gefragt, stattdessen muss man über Optionsschaltflächen auswählen, ob sie 1p oder 3p angeschlossen ist. Auswirkungen hat das aber keine, es wird das Fahrzeug 3p geladen, wenn die Wallbox 3p angeschlossen ist. Ob eine falsche Auswahl Auswirkungen auf die PV-Steuerung hat, kann ich wetterbedingt leider derzeit noch nicht ermitteln. Die Anzeige ist aber fehlerhaft bis irreführend.

Im Bild sieht man die falsche Anzeige bei 1p-Anschluss der Wallbox, weil es nicht mehr erkannt wird.
[image]

### Steps to reproduce

Schließe eine Energie-Control von Heideberg an und nutze Version 0.124.0

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.124.0

andig, 2024-02-02:
Logfile fehlt.

andig, 2024-02-02:
/cc @naltatis

naltatis, 2024-02-02:
@andreashm Was hattest du denn vorher bei dem Ladepunkt als `phases` eingetragen? Mit der neuen Version ist es in der Tat so, dass wir eine feste Phaseneinstellung als Konfiguration am Ladepunkt erwarten.

naltatis, 2024-02-02:
>  Das beißt sich doch aber dann mit der "automatischen" Erkennung über die Phasenspannungen.

@andig @premultiply seid ihr da im Thema? Diese automatische Erkennung hatte ich bislang nicht auf dem Zettel. Wie bilden wir die im Code ab?

andig, 2024-02-02:
Das ist https://github.com/evcc-io/evcc/blob/4c84322c5cacbf4ad5bce2a6ff1c371763563d44/core/loadpoint.go#L1319

VolkerK62, 2024-02-02:
Ich hab im UI `phases: 3` definiert. Wie früher in der yaml.
Über die Spannungen wird 1p erkannt.
In /api/state:
- phasesActive: 1
- phasesConfigured: 3
- phasesEnabled: 1

naltatis, 2024-02-02:
Zur UI: Hier kommunizieren wir nicht klar genug, welche Auswirkungen diese Angabe hat. Bei 1p3p Geräten kontrolliert man damit direkt das Verhalten der Wallbox (auto, 3p, 1p). Bei "normalen" Chargern ist das lediglich die Information für evcc wie die Wallbox angeschlossen ist (3p o. 1p). Falschangabe führt zu keiner Verhaltensänderung, aber ggf. zu anderen internen Berechnungen. Das war in der yaml Einstellung (`phases: 0/3/1`) bislang genau so "überladen". Hier in der UI fällt es aber deutlicher auf.

Kurzfristige Maßnahme: Unterstützt die Box keine Phasenumschaltung zeigen wir einen Hinweistext, an der klarmacht, dass das hier lediglich eine Info für evcc ist.

Mittelfristige Maßnahme: Für Boxen ohne Phasenumschaltung wird diese Angabe ins Config UI (für Ersteinrichtung) verschoben. Voraussetzung ist, dass wir Ladepunkte/Charger dort anlegen können. Boxen mit Phasenumschaltung behalten die Phasenänderungsoptionen im Main UI (heutiger Ort).

VolkerK62, 2024-02-02:
ABER: Vor den Update wurde im UI die Ladeleistung bei mincurrent und maxcurrent richtig angezeigt, nämlich auf Basis der erkannten Phasen. Jetzt wird dafür wohl der eingestellte Phasenwert genommen.

naltatis, 2024-02-02:
@VolkerK62 Ja, du hast recht:
https://github.com/evcc-io/evcc/pull/11571/files#diff-3b9fd87655a9ea0010ac8a5b5128979d0fc96df5a48831983faddc6899c166bbL253

Ich schau mir das mal an.

andreashm, 2024-02-02:
Hi, ich habe glaube keine phases in der yaml angegeben, weil ich im Schaltschrank manuell von 3p auf 1p umschalten kann.
Ich kann gerade leider nicht nachschauen.

andig, 2024-02-02:
Keine angegeben bedeutet 3p per default und dann läuft die Erkennung.

naltatis, 2024-02-02:
Standardwert war bislang `3` ([siehe Doku](https://docs.evcc.io/docs/reference/configuration/loadpoints/#phases)). Das hat sich mit der Umstellung ins UI auch nicht verändert. Daher sollte es "nur" das Darstellungsproblem sein was hier gefixt ist. 👆

Tombra1889, 2024-02-02:
Ich fände es in diesem Kontext optimal, wenn ich in dem Wertebereich nur die Werte auswählen könnte, die in der yaml hinterlegt sind:
[image]
Meine Wallbox unterstützt nur eine Leistung bis 16 A. Dieses habe ich auch so hinterlegt, auswählen könnte ich aber trotzdem bis 32 A.

VolkerK62, 2024-02-02:
Der Loadpoint-Eintrag in der yaml wird ignoriert. Die Einstellung findet jetzt im UI statt.

@andig @naltatis  könnte/sollte man hier den Maxwert aus der Wallbox nehmen? Keine Ahnung, ob die von allen auslesbar sind.

naltatis, 2024-02-02:
Ja, das kommt mit dem nächsten Schritt in der Config UI wieder. Da würden dann harte min/max Werte definiert werden und hier (Settings am Loadpoint) nur noch innerhalb der Grenzen für den Alltag umstellbar.

Bis dahin ist dieser Dialog das globale Min/Max für den Ladepunkt (so wie voher die yaml war.)

naltatis, 2024-02-02:
Ich hab die Doku entsprechend aktualisiert: https://github.com/evcc-io/docs/pull/522

PR #12017 (2024-02-02): Clarified phase setting for non 1p3p chargers
fixes #12013

- 🏷️ added expaination for non 1p3p chargers
- 🚥 calculate min/maxpower based on active phases if phase switching isnt supported

[image]
closed 2024-02-02 by andig