evcc-io/evcc #12015: Experimantal Device Configuration - KIA Car settings not loaded on edit

pandamouse84, 2024-02-02:
### Describe the bug

The edit dialog for cars in the experimental device configuration does not load the current settings to the UI. (KIA)

### Steps to reproduce

1. Open Experimantal Device Configuration
2. Edit a saved car (in my case KIA)
3. Saved settings are not loaded/shown in the edit dialog

[image]

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: ****

interval: 30s # control cycle interval

meters:
  - type: template
    template: varta
    usage: grid
    host: ****
    port: 502
    name: gridVarta
  - type: template
    template: varta
    usage: battery
    host: ****
    port: 502
    name: batteryVarta
  - type: template
    template: sma-data-manager
    usage: pv
    host: ****
    port: 502
    name: pvSMA

chargers:
  - type: template
    template: ellipro
    ski: ****
    ip: ****
    name: wallboxElli

loadpoints:
  - title: Carport
    charger: wallboxElli
    mode: pv
    phases: 3
    mincurrent: 6
    maxcurrent: 16

site:
  title: Mein Zuhause
  meters:
    grid: gridVarta
    pv:
      - pvSMA
    battery:
      - batteryVarta

eebus:
  certificate:
    private: |
      -----BEGIN EC PRIVATE KEY-----
      ****
      -----END EC PRIVATE KEY-----
    public: |
      -----BEGIN CERTIFICATE-----
      ****
      -----END CERTIFICATE-----
```

### Log details

```text
Don't know how to do that as HomeAssistant Add-on.
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.0

andig, 2024-02-02:
@pandamouse84 could you please check browsers developer console, network tab: what requests are being exchanged when loading the vehicle?

pandamouse84, 2024-02-02:
Sure:

[image]

**1** Returns the following XML:
```xml
{
    "result": {
        "config": {
            "cache": "15m",
            "capacity": "74",
            "icon": "car",
            "language": "de",
            "password": "****",
            "template": "kia",
            "title": "KIA EV6",
            "user": "****"
        },
        "id": 1,
        "name": "db:1",
        "type": "template"
    }
}
```

pandamouse84, 2024-02-02:
<details>
<summary>
vehicles
</summary>
``` xml
 {
  "result": [
        {
            "name": "Aiways",
            "template": "aiways"
        },
        {
            "name": "Audi",
            "template": "audi"
        },
        {
            "name": "BMW",
            "template": "bmw"
        },
        {
            "name": "Citroën",
            "template": "citroen"
        },
        {
            "name": "Dacia",
            "template": "dacia"
        },
        {
            "name": "DS",
            "template": "ds"
        },
        {
            "name": "evNotify",
            "template": "evnotify",
            "group": "generic"
        },
        {
            "name": "Fiat",
            "template": "fiat"
        },
        {
            "name": "Ford",
            "template": "ford"
        },
        {
            "name": "Hyundai Bluelink",
            "template": "hyundai"
        },
        {
            "name": "ISO15118",
            "template": "iso15118",
            "group": "generic"
        },
        {
            "name": "Jaguar",
            "template": "jaguar-landrover"
        },
        {
            "name": "Jeep",
            "template": "fiat"
        },
        {
            "name": "Kia Bluelink",
            "template": "kia"
        },
        {
            "name": "Land Rover",
            "template": "jaguar-landrover"
        },
        {
            "name": "mazda2mqtt",
            "template": "mazda2mqtt",
            "group": "generic"
        },
        {
            "name": "mg2mqtt",
            "template": "mg2mqtt",
            "group": "generic"
        },
        {
            "name": "Mini",
            "template": "mini"
        },
        {
            "name": "Nissan Leaf",
            "template": "nissan"
        },
        {
            "name": "Nissan Leaf (pre 2019)",
            "template": "carwings"
        },
        {
            "name": "NIU E-Scooter",
            "template": "niu-e-scooter",
            "group": "scooter"
        },
        {
            "name": "Offline",
            "template": "offline",
            "group": "generic"
        },
        {
            "name": "Opel",
            "template": "opel"
        },
        {
            "name": "Open Vehicle Monitoring System",
            "template": "ovms",
            "group": "generic"
        },
        {
            "name": "Peugeot",
            "template": "peugeot"
        },
        {
            "name": "Polestar",
            "template": "polestar"
        },
        {
            "name": "Porsche",
            "template": "porsche"
        },
        {
            "name": "PSA Car Controller",
            "template": "flobz",
            "group": "generic"
        },
        {
            "name": "Renault",
            "template": "renault"
        },
        {
            "name": "Seat CupraConnect Gen3 (Ateca, Leon, Formentor)",
            "template": "seat"
        },
        {
            "name": "Seat CupraConnect Gen4 (Born)",
            "template": "cupra"
        },
        {
            "name": "Skoda Andere",
            "template": "skoda"
        },
        {
            "name": "Skoda Enyaq",
            "template": "enyaq"
        },
        {
            "name": "Smart #1",
            "template": "smart-hello"
        },
        {
            "name": "Smart EQ",
            "template": "smart"
        },
        {
            "name": "Tesla aktuell",
            "template": "tesla-command"
        },
        {
            "name": "Tesla veraltet",
            "template": "tesla"
        },
        {
            "name": "TeslaMate",
            "template": "teslamate",
            "group": "generic"
        },
        {
            "name": "Tronity",
            "template": "tronity",
            "group": "generic"
        },
        {
            "name": "Volkswagen We Connect ID",
            "template": "vw"
        },
        {
            "name": "Volvo aktuell",
            "template": "volvo-connected"
        },
        {
            "name": "Volvo veraltet",
            "template": "volvo"
        }
    ]
}
``` 
</details>

naltatis, 2024-02-02:
Thanks! I'll look into this.

RTTTC, 2024-02-15:
https://github.com/evcc-io/evcc/discussions/12265

closed 2024-03-09 by andig