evcc-io/evcc #12019: EVCC not starting after upgrade to 0.124.0

Irulusug, 2024-02-02:
### Describe the bug

Post upgrading to version 0.124.0, I receive below error:

cannot create charger 'my_charger'
cannot create charger type 'template'
cannot create charger type 'vestel'
failsafe timeout
dial tcp 192.168.1.56:502
i/o timeout

- Sponsortoken has been checked, trial token tested
- Tesla Tokens generated on 2nd Feb. Access token begins with eyJh, Refresh Token with EU_

Tested stable and nighty builds. 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: debug  
levels:  
  cache: error  
  
# unique installation id  
plant: *****  
  
interval: 10s # control cycle interval  
  
sponsortoken: *****  
  
  
# sponsors can set telemetry: true to enable anonymous data aggregation  
# see https://github.com/evcc-io/evcc/discussions/4554  
telemetry: false  
  
meters:  
- type: template  
  template: fronius-solarapi-v1   
  usage: grid    
  host: 192.168.1.67    
  name: grid1  
- type: template  
  template: fronius-solarapi-v1   
  usage: pv    
  host: 192.168.1.67    
  name: pv2  
  
chargers:  
  - name: my_charger  
    type: template  
    template: vestel  
      
    # Modbus TCP  
    modbus: tcpip  
    id: 255  
    host: 192.168.1.56 # Hostname  
    port: 502 # Port   
  
vehicles:  
- type: template  
  template: tesla-command   
  title: Tessi    
  accessToken: *****  
  refreshToken: *****  
  capacity: 75    
  name: ev3  
  
loadpoints:  
- title: Garage  
  charger: my_charger  
  mode: off  
  phases: 3  
  mincurrent: 6  
  maxcurrent: 16  
  
site:  
  title: Home  
  meters:  
    grid: grid1  
    pv:  
    - pv2
```

### Log details

```text
Feb 02 18:09:28 raspberrypi systemd[1]: Started evcc.service - evcc.
Feb 02 18:09:28 raspberrypi evcc[18419]: [main  ] INFO 2024/02/02 18:09:28 evcc 0.124.0 (12836127)
Feb 02 18:09:28 raspberrypi evcc[18419]: [main  ] INFO 2024/02/02 18:09:28 using config file: /etc/evcc.yaml
Feb 02 18:09:28 raspberrypi evcc[18419]: [main  ] INFO 2024/02/02 18:09:28 starting ui and api at :7070
Feb 02 18:09:29 raspberrypi evcc[18419]: [db    ] INFO 2024/02/02 18:09:29 using sqlite database: /var/lib/evcc/evcc.db
Feb 02 18:09:39 raspberrypi evcc[18419]: [main  ] FATAL 2024/02/02 18:09:39 cannot create charger 'my_charger': cannot create charger type 'template': cannot create charger type 'vestel': failsafe timeout: dial tcp 192.168.1.56:502: i/o timeout
Feb 02 18:09:39 raspberrypi evcc[18419]: [main  ] FATAL 2024/02/02 18:09:39 will attempt restart in: 5m0s
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.124.0 (12836127)

andig, 2024-02-02:
> dial tcp 192.168.1.56:502: i/o timeout

Keine Netzwerkverbindung. Da kann evcc nix machen.

closed 2024-02-02 by andig

Irulusug, 2024-02-02:
Netzwerk Verbindung ist vorhanden und funktioniert, sowohl vom betroffenen Raspberry wie auch einer zweiten v0.123.7 Installation auf einem zweiten Raspberry

andig, 2024-02-02:
Das mag sein, aber die Fehlermeldung ist eindeutig. Da antwortet nichts.