evcc-io/evcc #12023: EVCC wants old pv meter depsite removed from config

buliwyf42, 2024-02-02:
### Describe the bug

I'm trying the new 0.124 and wantet to configure my pv meter via the gui.
Therefore i tried to remove them from evcc.yaml
But this config won't start The nsite needs my pv meter which i removed from the config. 

### Steps to reproduce

See in description

### Configuration details

```yaml
meters:
  - name: my_fronius_smartmeter
    type: template
    template: fronius-gen24
    usage: grid
    host: 192.168.xxx.xxx
#  - name: my_fronius_pv1
#    type: template
#    template: fronius-gen24
#    usage: pv
#    host: 192.168.xxx.xxx
#  - name: my_fronius_pv2
#    type: template
#    template: sunspec-inverter
#    host: 192.168.xxx.xxx
#    modbus: tcpip  
#  - name: my_byd_battery
#    type: template
#    template: fronius-gen24
#    usage: battery
#    host: 192.168.xxx.xxx
#    capacity: 12.8
  - name: Warmwasser
    type: template
    usage: charge
    template: shelly-1pm
    host: 192.168.xxx.xxx

site:
  title: Ebelsbach
  meters:
    grid: my_fronius_smartmeter
#    pv: 
#      - my_fronius_pv1
#      - my_fronius_pv2
#    battery: 
#      - my_byd_battery    
#  bufferSoc: 95 
#  prioritySoc: 20  
#  residualPower: 100
#  bufferStartSoc: 99
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/config/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/02/02 20:55:37 evcc 0.124.0
[main  ] INFO 2024/02/02 20:55:37 using config file: /config/evcc.yaml
[main  ] INFO 2024/02/02 20:55:37 starting ui and api at :7070
[db    ] INFO 2024/02/02 20:55:38 using sqlite database: /config/evcc.db
[mqtt  ] INFO 2024/02/02 20:55:38 connecting evcc-1062728316 at tcp://mqtt.home.lan:1883
[lp-2  ] WARN 2024/02/02 20:55:44 poll mode '{always 30s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1  ] ERROR 2024/02/02 20:55:47 charge total import: Get "http://192.168.xxx.xx/status": dial tcp 192.168.211.33:80: connect: no route to host
[main  ] FATAL 2024/02/02 20:55:47 failed configuring site: not found: my_fronius_pv1
[main  ] FATAL 2024/02/02 20:55:47 will attempt restart in: 5m0s
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.0

andig, 2024-02-02:
Wo kommt das her:

 > not found: my_fronius_pv1

Vermutung: du startest eine andere Config als du hier zeigst. Anders ist es nicht möglich.

buliwyf42, 2024-02-02:
> Wo kommt das her:
> 
> > not found: my_fronius_pv1
> 
> Ausdenken kann sich evcc das nicht. Vermutung: du startest eine andere Config als du hier zeigst. Anders ist es nicht möglich.

Nein, denn wenn ich die Datei bearbeite und die # herausnehme, startet der Dienst wie gewohnt.

andig, 2024-02-02:
Es gibts trotzdem keinen Mechanismus wie sich evcc den Bezeichner ausdenken sollte. Wo kommt der her?