evcc-io/evcc #12027: Adding Tesla via Gui fails

buliwyf42, 2024-02-02:
### Describe the bug

I removed my Tesla from the config evcc.yaml and tried to add it via the gui.
While the validation worked, adding failed anyway.

### Steps to reproduce

See Description

[image]

### Configuration details

```yaml
vehicles:
#  - name: NEO
#    type: template
#    template: tesla-command
#    capacity: 57.5 
#    phases: 3
#    mincurrent: 6
#    maxcurrent: 16
#    icon: car
```

### Log details

```text
Unfortunately, nothing in the log files.
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

evcc version 0.124.0

AnBo83, 2024-02-03:
I think its the same: https://github.com/evcc-io/evcc/issues/11987

buliwyf42, 2024-02-03:
> I think its the same: #11987

It seems so.

closed 2024-02-03 by buliwyf42

andig, 2024-02-04:
@naltatis auch hier müsste bei synchronen Requests die Fehlermeldung ins Popup?

andig, 2024-02-04:
@buliwyf42 this should work in the nightly version