evcc-io/evcc #12035: fix `phases: 1` bei steckdosen

VolkerK62, 2024-02-03:
Steckdosen müssen mit `phases: 1` konfiguriert werden. 
In Verbindung mit Config UI würde es Sinn machen, das per default zu setzen und im UI keine Auswahl zuzulassen.

andig, 2024-02-04:
Leider haben wir `phases` nicht als Parameter an den Chargern und wir erkennen bei einem Charger auch nicht, von welchem Typ er ist.

@naltatis ?

naltatis, 2024-02-04:
Ah schade. Ich hatte gehofft, dass wir erkennen können, ob ein Charger in SwitchSocket ist.

andig, 2024-02-04:
Lass uns das mal zu Ende denken. Alle Steckdosen (wenn auch nicht alle Charger) nutzen heute schon `embed`. Dem könnten wir analog `Vehicle` ein optionales `Phases() int` API beibringen.

Was würden wir dann damit machen? Nur einmalig beim Start auslesen/publizieren? Auch im Loadpoint verwursten?!

naltatis, 2024-02-04:
Ich würd hier zwei Dinge sehen:

- Phasen-Default beim Start auf `1` setzen
- Die "harte Einphasigkeit" irgendwie publizieren bzw. später im Config-UI ersichtlich machen sodass wir in diesem Fall keine Phasen-Konfiguration anbieten.

andig, 2024-02-04:
Siehe PR ;)

closed 2024-02-08 by andig