evcc-io/evcc #12036: phases: 2 beim vehicle wird in Version 0.124.0 ignoriert

RobBboR, 2024-02-03:
### Describe the bug

Hallo zusammen,

nach einem Update von 0.123.5 auf 0.124.0 wird `phases: 2` beim vehicle ignoriert. In der GUI gibt es eine neue Auswahl, wo ich mit verbundenem Fahrzeug neuerdings zwischen 1 und 3 Phasen wählen kann (Screenshot unten). Ich habe allerdings `phases: 2` für das Fahrzeug konfiguriert, damit die minimale Ladeleistung 2,8 kW ist. Leider unterstützt EVCC nicht die automatische Phasenumschaltung meiner Wallbox, so dass im Loadpoint `phases: 3` eingetragen ist, da 2 hier nicht möglich ist. Im Log steht jetzt, dass diese Einstellung ignoriert wird ([lp-1  ] WARN 2024/02/03 15:44:00 deprecated: phases setting is ignored, please remove)

Mit dieser Konfiguration wurden in diesem Fenster in der Vorversion eine minimale Ladeleistung von 2,8 kW angezeigt, was auch wunderbar funktioniert hat. Jetzt steht hier der minimale Ladestrom von 4,1 kW (was zu viel ist), bzw. 1,4 kW (was nicht funktionieren wird, da nicht auf eine Phase umgeschaltet werden kann).

Leider kann ich mit dem momentanen Wetter nicht überprüfen, ab welchem Überschuss der Ladevorgang jetzt wirklich beginnt. Mindestens ein Bug in der GUI dürfte es aber sein.

Cheers
Rob

[image]

### Steps to reproduce

1. Konfiguriere `phases: 2` für das Fahrzeug
2. Verbinde es mit der Wallbox
3. `phases: 2` wird zumindest von der GUI ignoriert

### Configuration details

```yaml
telemetry: false

meters:
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: SE10K-****  
  port: 1502  
  usage: grid  
  modbus: tcpip  
  timeout: 20s  
  name: grid1
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: SE10K-****  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  timeout: 20s  
  name: pv2
- type: template
  template: solaredge-hybrid 
  id: 1  
  host: SE10K-****  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  timeout: 20s  
  name: battery3

chargers:
- type: template
  template: smaevcharger 
  host: SMA30****  
  user: ******
  password: *********  
  name: wallbox4

loadpoints:
- title: Garage
  charger: wallbox4
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 32
  vehicle: VWeup

vehicles:
- name: VWeup
  type: template
  template: offline
  title: VW e-up!
  mode: pv
  capacity: 32.3
  phases: 2

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.3713
  feedin:
    type: fixed
    price: 0.08

site:
  title: ********
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
[main  ] INFO 2024/02/03 15:43:57 evcc 0.124.0
[main  ] INFO 2024/02/03 15:43:57 using config file: /etc/evcc.yaml
[main  ] INFO 2024/02/03 15:43:57 starting ui and api at :7070
[db    ] INFO 2024/02/03 15:43:57 using sqlite database: /home/rob/.evcc/evcc.db
[lp-1  ] WARN 2024/02/03 15:44:00 deprecated: minCurrent setting is ignored, please remove
[lp-1  ] WARN 2024/02/03 15:44:00 deprecated: maxcurrent setting is ignored, please remove
[lp-1  ] WARN 2024/02/03 15:44:00 deprecated: phases setting is ignored, please remove
[site  ] WARN 2024/02/03 15:44:00 battery configured but residualPower is missing or <= 0 (add residualPower: 100 to site), see https://docs.evcc.io/en/docs/reference/configuration/site#residualpower
[site  ] INFO 2024/02/03 15:44:00 site config:
[site  ] INFO 2024/02/03 15:44:00   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/02/03 15:44:00     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/02/03 15:44:00     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/02/03 15:44:00     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/02/03 15:44:00   vehicles:
[site  ] INFO 2024/02/03 15:44:00     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/02/03 15:44:00 loadpoint 1:
[lp-1  ] INFO 2024/02/03 15:44:00   mode:        pv
[lp-1  ] INFO 2024/02/03 15:44:00   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2024/02/03 15:44:00   meters:      charge ✓
[lp-1  ] INFO 2024/02/03 15:44:00     charge:    power ✓ energy ✗ currents ✓
[site  ] DEBUG 2024/02/03 15:44:00 set priority soc: 100
[lp-1  ] DEBUG 2024/02/03 15:44:00 phase timer inactive
[lp-1  ] DEBUG 2024/02/03 15:44:00 pv timer inactive
[lp-1  ] DEBUG 2024/02/03 15:44:00 guard timer inactive
[lp-1  ] INFO 2024/02/03 15:44:00 vehicle updated: unknown -> VW e-up!
[lp-1  ] DEBUG 2024/02/03 15:44:00 set charge mode: pv
[site  ] WARN 2024/02/03 15:44:00 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/02/03 15:44:00 ----
[lp-1  ] DEBUG 2024/02/03 15:44:00 charge power: 0W
[main  ] FATAL 2024/02/03 15:44:00 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
```

### What type of operating system are you running?

Linux

### Version

0.124.0

VolkerK62, 2024-02-03:
wenn du kannst, nimm mal das aktuelle Nightly.
Wobei das Problem nur "optisch" sein sollte.
An der Funktion hat sich nichts verändert. Der Phases Wert im UI gibt an, wie die Wallbox angeschlossen ist. Ersetzt den Phases Eintrag in der yaml.
Lediglich die angezeigte Ladeleistung im Loadpoint-UI ist optisch falsch (im Nightly behoben), im internen Ablauf ist alles richtig.
Das kannst du prüfen, wenn du das auto ansteckst und dann auf "Plan" klickst. Da wird auch dann nämlich die richtige (tatsächliche) Ladeleistung angezeigt.

RobBboR, 2024-02-04:
Ja, im Nightly passt es wieder, danke! Laden getestet. Fängt auch bei 2,8 kW an.
Ja, die Warnung hatte ich auch so verstanden, dass sie sich auf den Loadpoint gezieht. Steht ja "lp-1" vorne. Wollte es trotzdem erwähnen.

[image]

RobBboR, 2024-02-04:
@VolkerK62 Eventuell habe ich das Issue zu früh geschlossen. Ich habe jetzt auch mal `mincurrent` (6A) und `maxcurrent` (16A) für das Fahrzeug definiert. Allerdings ist bei Max. Ladestrom immer noch 32A ausgewählt und dahinter steht "~ 14,7 kW" (wie im letzten Screenshot). Ich hätte hier jetzt mit angeschlossenem Fahrzeug 16A und 7,3 kW erwartet. Wenn sich die phases vom vehicle auf den letzteren Wert auswirken, sollte das bei den currents auch so sein, oder?

andig, 2024-02-04:
@RobBboR das ist beim aktuellen Zwischenstand so- rein kosmetisches Problem.

/cc @naltatis

closed 2024-02-04 by andig

VolkerK62, 2024-02-04:
@andig   in dem Zusammenhang, bei https://github.com/evcc-io/evcc/issues/11550  hat der `bot` zugeschlagen

VolkerK62, 2024-02-04:
@RobBboR  
>Wenn sich die phases vom vehicle auf den letzteren Wert auswirken, sollte das bei den currents auch so sein, oder?

das ist auch so, wenn du `mincurrent, maxcurrent` bei Vehicle konfigurierst  ;)