evcc-io/evcc #12040: `goodwe-wifi` -> "Keine Messwerte"

MarkusMit, 2024-02-03:
### Describe the bug

In my Home Assistant setup, I tried to change my evcc configuration from MQTT to the new **GoodWe over WiFi** (#11750) but unfortunately I only get "Keine Messwerte" on the evcc UI.

I tried both, configuration with and without `uri`, but the result is the same.
Unfortunately, the [documentation](https://docs.evcc.io/docs/devices/meters#goodwe-over-wifi) is not very ... explanatory ... in that regard of when to configure a `uri` and when not.

### Steps to reproduce

Home Assistant, on Raspberry Pi 4
- Core 2024.1.6
- Supervisor 2023.12.1
- Operating System 11.4
- Frontend 20240104.0

**evcc** addon 0.124.0

FritzBox WiFi

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: homeassistant.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 13s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
sponsortoken: *****
plant: *****

# log settings
log: debug
levels:
  site: error #debug
  lp-1: error #trace
  lp-2: error
  cache: error
  db: error

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
[... 289 lines cut ...]
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/02/03 17:50:15 evcc 0.124.0
[main  ] INFO 2024/02/03 17:50:15 using config file: /config/evcc.yaml
[main  ] INFO 2024/02/03 17:50:15 starting ui and api at :7070
[mqtt  ] INFO 2024/02/03 17:50:15 connecting evcc-1075262415 at tcp://localhost:1883
[mqtt  ] DEBUG 2024/02/03 17:50:16 tcp://localhost:1883 connected
[main  ] DEBUG 2024/02/03 17:50:37 vehicle status: B (IONIQ 5)
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.0

MarkusMit, 2024-02-03:
Update regarding log: Found out that I have to set `site: debug` to get a little bit more info:
```
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2024/02/03 19:41:46 evcc 0.124.0
[main  ] INFO 2024/02/03 19:41:46 using config file: /config/evcc.yaml
[main  ] INFO 2024/02/03 19:41:46 starting ui and api at :7070
[mqtt  ] INFO 2024/02/03 19:41:47 connecting evcc-***** at tcp://localhost:1883
[mqtt  ] DEBUG 2024/02/03 19:41:47 tcp://localhost:1883 connected
[site  ] INFO 2024/02/03 19:41:49 site config:
[site  ] INFO 2024/02/03 19:41:49   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/02/03 19:41:49     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/02/03 19:41:49     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/02/03 19:41:49     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/02/03 19:41:49   vehicles:
[site  ] INFO 2024/02/03 19:41:49     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[site  ] WARN 2024/02/03 19:41:49 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/02/03 19:41:49 ----
[site  ] DEBUG 2024/02/03 19:41:49 pv power: 0W
[site  ] DEBUG 2024/02/03 19:41:49 battery soc: 0%
[site  ] DEBUG 2024/02/03 19:41:49 battery power: 0W
[site  ] DEBUG 2024/02/03 19:41:49 grid power: 0W
[site  ] DEBUG 2024/02/03 19:41:49 site power: 100W
[main  ] DEBUG 2024/02/03 19:41:49 vehicle status: B (MeinAuto)
[site  ] DEBUG 2024/02/03 19:42:02 ----
[site  ] DEBUG 2024/02/03 19:42:02 pv power: 0W
[site  ] DEBUG 2024/02/03 19:42:02 battery soc: 0%
[site  ] DEBUG 2024/02/03 19:42:02 battery power: 0W
[site  ] DEBUG 2024/02/03 19:42:02 grid power: 0W
[site  ] DEBUG 2024/02/03 19:42:02 site power: 100W
[site  ] DEBUG 2024/02/03 19:42:15 ----
[site  ] DEBUG 2024/02/03 19:42:15 pv power: 0W
[site  ] DEBUG 2024/02/03 19:42:15 battery soc: 0%
[site  ] DEBUG 2024/02/03 19:42:15 battery power: 0W
[site  ] DEBUG 2024/02/03 19:42:15 grid power: 0W
...
```

andig, 2024-02-04:
@MarkusMit Goodwe-Wifi requires UDP. I guess that doesn't work in your setup for some reason.

@motze92 the implementation needs to add timeout errors for the case that data is not received. The the readings above should error instead of returning zero.

andig, 2024-02-04:
@motze92 should also fix the infinite recursion from

```go
func (m *Server) readData() {
	for _, inverter := range m.inverters {
		addr, err := net.ResolveUDPAddr("udp", inverter.IP+":8899")
		if err != nil {
			return
		}

		if _, err := m.conn.WriteToUDP([]byte{0xF7, 0x03, 0x89, 0x1C, 0x00, 0x7D, 0x7A, 0xE7}, addr); err != nil {
			return
		}
		time.Sleep(5 * time.Second)
		if _, err := m.conn.WriteToUDP([]byte{0xF7, 0x03, 0x90, 0x88, 0x00, 0x0D, 0x3D, 0xB3}, addr); err != nil {
			return
		}
	}
	m.readData()
}
```

closed 2024-02-08 by andig

MarkusMit, 2024-02-10:
> @MarkusMit Goodwe-Wifi requires UDP. I guess that doesn't work in your setup for some reason.

@andig - As far as I understand it, Home Assistant's [GoodWe integration](https://www.home-assistant.io/integrations/goodwe/) also works via UDP (see e.g. [goodwe/coordinator.py](https://github.com/home-assistant/core/blob/a5cc0ae890f8953e0e339d1c37584929c2a69623/homeassistant/components/goodwe/coordinator.py#L43-L47)), and that works quite fine on my setup. Could there be another reason for `goodwe-wifi` not working?

motze92, 2024-02-10:
Do you use Home Assistant at the same time? The Inverter can't handle multiple clients paralell. Also while you are using the app, sometimes evcc doesn't get values.

MarkusMit, 2024-02-10:
I have both in Home Assistant:  The GoodWe integration, and I tried to use `goodwe-wifi` with the *evcc* addon in HA as well.

MarkusMit, 2024-03-17:
> I have both in Home Assistant: The GoodWe integration, and I tried to use `goodwe-wifi` with the _evcc_ addon in HA as well. So that will not work "by design"?

Shouldn't this be [documented](https://docs.evcc.io/docs/devices/meters/#goodwe-over-wifi)?

andig, 2024-03-17:
If anyone can confirm that this is the actual problem, happy to take a PR.

roboco6, 2024-04-01:
My first comment, so first of all: huge thanks to the people behind this project. Absolutely impressive work - thank you so much! 

I tried all easter weekend to try to get this to work. Some things worked better than expected, but I have problems with the meter data vom the GoodWe inverter. 

Setup: 
- GoodWe ET 15 Inverter, with Battery and WiFi Module
- ABL EMH1 1W220101079 (mit Protoss PE11-h im LAN)
- ID.4
- Raspberry Pi Zero W Rev 1.1

GUI always shows incorrect data - except of the SoC of the battery, which is always accurate.

Screenshots taken at the same time: 

[image]
[image]

```
Apr 01 13:50:09 raspberrypi evcc[358]: [site  ] DEBUG 2024/04/01 13:50:09 ----
Apr 01 13:50:09 raspberrypi evcc[358]: [lp-1  ] DEBUG 2024/04/01 13:50:09 charge power: 0W
Apr 01 13:50:09 raspberrypi evcc[358]: [site  ] DEBUG 2024/04/01 13:50:09 pv power: 3576W
Apr 01 13:50:09 raspberrypi evcc[358]: [site  ] DEBUG 2024/04/01 13:50:09 battery soc: 84%
Apr 01 13:50:09 raspberrypi evcc[358]: [site  ] DEBUG 2024/04/01 13:50:09 battery power: -7220W
Apr 01 13:50:09 raspberrypi evcc[358]: [site  ] DEBUG 2024/04/01 13:50:09 grid meter: 39W
Apr 01 13:50:09 raspberrypi evcc[358]: [site  ] DEBUG 2024/04/01 13:50:09 site power: -7181W
Apr 01 13:50:09 raspberrypi evcc[358]: [lp-1  ] DEBUG 2024/04/01 13:50:09 charge currents: [0 0 0]A
Apr 01 13:50:09 raspberrypi evcc[358]: [lp-1  ] DEBUG 2024/04/01 13:50:09 charger status: A
Apr 01 13:50:37 raspberrypi evcc[358]: [site  ] DEBUG 2024/04/01 13:50:37 ----
sennewald@raspberrypi:~ $ evcc meter
[main  ] INFO 2024/04/01 14:30:31 evcc 0.124.10
[main  ] INFO 2024/04/01 14:30:31 using config file: /etc/evcc.yaml
[db    ] INFO 2024/04/01 14:30:33 using sqlite database: /home/sennewald/.evcc/evcc.db
[main  ] FATAL 2024/04/01 14:30:33 cannot create meter 'grid1': cannot create meter type 'template': cannot create meter type 'goodwe-wifi': listen udp 0.0.0.0:8899: bind: address already in use
```

```
open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

unique installation id
plant: xxx

interval: 30s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: goodwe-wifi 
  usage: grid  
  uri: 192.168.10.197  
  name: grid1
- type: template
  template: goodwe-wifi 
  usage: pv  
  uri: 192.168.10.197  
  name: pv2
- type: template
  template: goodwe-wifi 
  usage: battery  
  uri: 192.168.10.197  
  name: battery3

chargers:
- type: template
  template: abl 
  id: 1  
  host: 192.168.10.195  
  port: 502  
  modbus: rs485tcpip  
  name: wallbox5

vehicles:
- type: template
  template: vw 
  title: ID.4  
  user: xxx 
  password: xxx
  capacity: 52  
  timeout: 10s  
  name: ev4

loadpoints:
- title: Carport
  charger: wallbox5
  vehicle: ev4
  mode: pv

site:
  title: My home
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
 ```

My guess: I configured something wrong due to lack of know how. 

If someone finds some mistake in my setup - I am more than thankful! 

Again - thanks ever so much to the people who made this possible. 

Rasmus

andig, 2024-04-01:
@roboco6 thanks for the feedback. Make sure evcc is the only client of the inverter. If not working please open new issue with trace log attached.

roboco6, 2024-04-01:
@andig thank you for your quick response. The only other client I could think of would be the "SEMS Portal" which is GoodWe's GUI App - not working as soon as evcc is started (because of only one modbus connection at a time - I have learned here). 

I wouldn't know how to cut the connection / do you think that is needed?

The other connection is the administration app "SolarGo" which works over wifi or bluetooth only. Even if that is shut down, the data shown in EVCC is acting weird.