evcc-io/evcc #12041: Nightly startet nicht mehr

giss69, 2024-02-03:
### Describe the bug

Hallo,

ich bin gestern wegen tesla.commands auf nightly umgestiegen und es lief alles.
Gerade vor ca, 30 Min. hat evcc wohl ein Update gemacht und evcc startet nicht mehr.
Im log sind nur 3 Einträge:
`[main  ] INFO 2024/02/03 20:35:50 evcc 0.124.0 (5627f57e)
[main  ] INFO 2024/02/03 20:35:50 using config file: /etc/evcc.yaml
[main  ] INFO 2024/02/03 20:35:50 starting ui and api at :7070`

woran könnte es liegen?
Evcc läuft im Docker  auf Synology.

Danke.

### Steps to reproduce

1.neustart
2.
3.
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 15s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
# modbusproxy:
#  - port: 5200
#    uri: 192.168.30.53:502
#    id: 1
  #    # rtu: true
  #    # readonly: true
meters:
  - name: pv
    type: template
    template: huawei-dongle-powersensor
    usage: pv      
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.10.5 # Hostname
    port: 503 # Port 
    timeout: 15s
    
  - name: grid
    type: template
    template: huawei-dongle-powersensor
    usage: grid     
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.10.5 # Hostname
    port: 503 # Port 
    timeout: 15s
    
  - name: battery
    type: template
    template: huawei-dongle-powersensor
    usage: battery
    
    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.10.5 # Hostname
    port: 503 # Port 
    timeout: 15s

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
# meters:
  # - name: pv
    # type: custom
   # power: # Leistung (W)
  #    source: mqtt
   #   topic: solaranzeige/box1/pv_leistung

 # - name: grid
  #  type: custom
 #   power: # Leistung (W)
   #   source: mqtt
   #   topic: solaranzeige/box1/einspeisung_bezug
   #   scale: -1

chargers:
  - name: twc3
    type: template
    template: twc3
    host: 192.168.30.77
    
loadpoints:
  - title: Parkplatz
    charger: twc3
    # vehicle: tesla_white
    mode: pv # Standardmodus zum Laden
    phases: 0 # Automatische Umschaltung zwischen 1 und 3 Phasen
    enable:
      threshold: 0 # 0 = Export die minimale Ladeleistung erreicht haben
      delay: 1m
    disable:
      threshold: 500 # Maximaler Netzbezug von 200W ist erlaubt
      delay: 3m
    mincurrent: 2 # Mindestens 6A
    maxcurrent: 16 # Maximal 16A
   # resetOnDisconnect: true  
  

# vehicle definitions
# name can be freely chosen and is used as reference when assigning vehicle to loadpoint
# for documentation see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: tesla_white
    type: template
    template: tesla-command
    title: white # Wird in der Benutzeroberfläche angezeigt # Optional
    accessToken: 
    refreshToken:
    vin: XP7YGCEK7NB024880 # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 79 # Akku-Kapazität in kWh # Optiona
  - name: tesla_grey
    type: template
    template: tesla-command
    title: Maverick # Wird in der Benutzeroberfläche angezeigt # Optional
    accessToken: 
    refreshToken:
    vin: LRWYGCEK6NC455190 # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
    capacity: 79 # Akku-Kapazität in kWh # Optional     
# site describes the EVU connection, PV and home battery
site:
  title: Home # display name for UI
  meters:
    grid: grid # grid meter
    pv:
      - pv # list of pv inverters/ meters
    battery:
      - battery # list of battery meters
 #   aux: aux # auxiliary meters for adjusting grid operating point
 # prioritySoc: # gprioritySoc: 50 # Hausbatterie bekommt bis zum Soc 50% Priorität beim laden
 # bufferSoc: # bufferSoc: 80 # Hausbatterie wird oberhalb Soc 80% als Puffer genutzt

# loadpoint describes the charger, charge meter and connected vehicle

# tariffs are the fixed or variable tariffs
tariffs:
  currency: EUR # three letter ISO-4217 currency code (default EUR)
  grid:
    # either static grid price (or price zones)
# type: fixed
#   price: 0.44 # EUR/kWh

    # # or variable via tibber
    type: tibber
    token: "" # access token
    # homeid:  # optional if multiple homes associated to account

    # # or variable via awattar
    # type: awattar
    # region: de # optional, choose at for Austria
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # EUR/kWh
  planner:
    # planner tariff can be used for target charging if not grid tariff is specified
    # GrünStromIndex (Germany only) or ElectricityMaps provide CO2 intensity forecast

    # type: grünstromindex
    # zip: <zip>

    # type: electricitymaps
    # uri: <uri>
    # token: <token>
    # zone: DE

# mqtt message broker
# mqtt:
 # broker: 192.168.10.5:1884
  # topic: evcc # root topic for publishing, set empty to disable
  # user:
  # password:

# influx database
influx:
  # url: http://localhost:8086
  # database: evcc
  # user:
  # password:

# eebus credentials
eebus:
  # uri: # :4712
  # interfaces: # limit eebus to specific network interfaces
  # - en0
  # certificate: # local signed certificate, required, can be generated via `evcc eebus-cert`
  #   public: # public key
  #   private: # private key

# push messages
messaging:
  events:
    start: # charge start event
      title: Charge started
      msg: Started charging in "${mode}" mode
    stop: # charge stop event
      title: Charge finished
      msg: Finished charging ${chargedEnergy:%.1fk}kWh in ${chargeDuration}.
    connect: # vehicle connect event
      title: Car connected
      msg: "Car connected at ${pvPower:%.1fk}kW PV"
    disconnect: # vehicle connected event
      title: Car disconnected
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: Soc updated
      msg: Battery charged to ${vehicleSoc:%.0f}%
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
  services:
  # - type: pushover
  #   app: # app id
  #   recipients:
  #   - # list of recipient ids
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
`[main  ] INFO 2024/02/03 20:35:50 evcc 0.124.0 (5627f57e)
[main  ] INFO 2024/02/03 20:35:50 using config file: /etc/evcc.yaml
[main  ] INFO 2024/02/03 20:35:50 starting ui and api at :7070`
```

### What type of operating system are you running?

Docker container

### Version

0.124.0 (5627f57e)

haroldboom, 2024-02-03:
I am getting the same issue on my stand alone Ubuntu 22 installation, debug is enabled in the config but I only get the following log output when running "evcc -c /etc/evcc.yaml":

`[main  ] INFO 2024/02/03 23:10:43 evcc 0.124.0 (5627f57e)
[main  ] INFO 2024/02/03 23:10:43 using config file: /etc/evcc.yaml
[main  ] INFO 2024/02/03 23:10:43 starting ui and api at :7070
[db    ] INFO 2024/02/03 23:10:43 using sqlite database: /root/.evcc/evcc.db
`
Also my config is located in /etc/evcc.yaml as per the wiki but using 'systemctl start evcc' I get the following log output which seems like it is working however I cant open the GUI (connection refused) I have the firewall disabled on the VM and can SSH into it.

`root@testrig:~# sudo journalctl -u evcc --since "yesterday"
Feb 03 22:51:31 testrig systemd[1]: Started evcc.
Feb 03 22:51:31 testrig evcc[2382]: [main  ] INFO 2024/02/03 22:51:31 evcc 0.124.0 (5627f57e)
Feb 03 22:51:31 testrig evcc[2382]: [main  ] INFO 2024/02/03 22:51:31 missing config file - switching into demo mode
Feb 03 22:51:31 testrig evcc[2382]: [main  ] INFO 2024/02/03 22:51:31 starting ui and api at :7070
Feb 03 22:51:32 testrig evcc[2382]: [db    ] INFO 2024/02/03 22:51:32 using sqlite database: /var/lib/evcc/evcc.db
Feb 03 22:51:33 testrig evcc[2382]: [site  ] WARN 2024/02/03 22:51:33 battery configured but residualPower is missing or <= 0`

giss69, 2024-02-04:
Besteht eine Möglichkeit auf Nightly Build #978 zurück zu kehren?

andig, 2024-02-04:
> Besteht eine Möglichkeit auf Nightly Build #978 zurück zu kehren?

Du kannst per apt jederzeit zurück gehen.

Aktualisiertes nightly in 20min verfügbar

closed 2024-02-04 by andig