evcc-io/evcc #12044: Issues with update to 0.124.0

wavelov3r, 2024-02-04:
### Describe the bug

Evcc ignores mincurrent, and does not set the right amount of energy to deliver to the car.
After the update it tells me that the "model" string into the evcc yaml file was bad. i've commented it.
It gives strange numbers in the logs also (see in the logs)

### Steps to reproduce

1. Update to the last version

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: error
#levels:
#  cache: error

interval: 30s # control cycle interval

sponsortoken: AZZAZA

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554

meters:
- type: template
  template: huawei-dongle-powersensor 
  id: 0  
  host: 127.0.0.1 
  port: 502  
  usage: grid  
  modbus: tcpip
  #rtu: true  
  name: grid
- type: template
  template: huawei-dongle-powersensor 
  id: 0  
  host: 127.0.0.1 
  port: 502  
  usage: pv  
  modbus: tcpip
  #rtu: true  
  name: pv

- name: battery
  type: custom
  power:
    source: modbus
    #rtu: true
    timeout: 10s
    connectdelay: 5s
    #model: huawei
    uri: 127.0.0.1:502
    id: 0
    register:
      address: 37765 #storage_charge_discharge_power
      type: holding
      decode: int32
    scale: -1 #modded skizzo
  soc:
    source: modbus
    #rtu: true
    timeout: 10s
    connectdelay: 5s
    #model: huawei
    uri: 127.0.0.1:502
    id: 0
    register:
      address: 37760 #storage_state_of_capacity
      type: holding
      decode: uint16
    scale: 0.1   

chargers:
  - name: ABB_Terra
    type: template
    template: abb   
    # RS485 via TCP/IP (Modbus RTU)
    modbus: rs485tcpip
    id: 1
    host: 192.168.0.13 # Hostname
    port: 8899 # Port
    
    
    # Modbus TCP
    #modbus: tcpip
    #id: 1
    #host: 192.168.0.13 # Hostname
    #port: 8899 # Port 

vehicles:
- type: template
  template: id 
  title: VW_ID3  
  user: user@gmail.com
  password: passwd
  vin: vinnumber  
  capacity: 70  
  name: ev4

loadpoints:
- title: Serrande
  charger: ABB_Terra
  vehicle: ev4
  mode: pv # charge mode (off, now, minpv, pv)
  phases: 1
  mincurrent: 6
  maxcurrent: 23
  #resetOnDisconnect: true

site:
  title: MySite
  meters:
    grid: grid
    pv:
    - pv
    battery:
    - battery
  residualPower: 100
#  prioritySoc: 60
#  bufferSoc: 100
  #maxGridSupplyWhileBatteryCharging: 2000
  
# mqtt message broker
mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  clientid: homeassistant
  user: mqttbroker
  password: brokerpasswd
```

### Log details

```text
ite  ] DEBUG 2024/02/04 09:33:38 ----
[lp-1  ] DEBUG 2024/02/04 09:33:39 charge power: 1197W
[site  ] DEBUG 2024/02/04 09:33:39 pv power: 2581W
[site  ] DEBUG 2024/02/04 09:33:40 battery soc: 12%
[site  ] DEBUG 2024/02/04 09:33:40 battery power: 0W
[site  ] DEBUG 2024/02/04 09:33:40 grid power: -833W
[site  ] DEBUG 2024/02/04 09:33:40 grid currents: [-3.47 -0 -0]A
[site  ] DEBUG 2024/02/04 09:33:40 site power: -633W
[lp-1  ] DEBUG 2024/02/04 09:33:41 charge voltages: [243 0 0]V
[lp-1  ] DEBUG 2024/02/04 09:33:41 detected connected phases: 1p
[lp-1  ] DEBUG 2024/02/04 09:33:42 charge currents: [5.03 0 0]A
[lp-1  ] DEBUG 2024/02/04 09:33:42 detected active phases: 1p
[lp-1  ] DEBUG 2024/02/04 09:33:44 charger status: C
[lp-1  ] DEBUG 2024/02/04 09:33:44 soc estimated: 80.33% (vehicle: 80.00%)
[lp-1  ] DEBUG 2024/02/04 09:33:44 vehicle soc: 80%
[lp-1  ] DEBUG 2024/02/04 09:33:44 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/02/04 09:33:44 vehicle range: 320km
[lp-1  ] DEBUG 2024/02/04 09:33:45 pv charge current: 9.78A = 7.03A + 2.75A (-633W @ 1p)
[site  ] DEBUG 2024/02/04 09:34:08 ----
[lp-1  ] DEBUG 2024/02/04 09:34:11 charge power: 1200W
[site  ] DEBUG 2024/02/04 09:34:11 pv power: 2596W
[site  ] DEBUG 2024/02/04 09:34:11 battery soc: 12%
[site  ] DEBUG 2024/02/04 09:34:11 battery power: 0W
[site  ] DEBUG 2024/02/04 09:34:12 grid power: -859W
[site  ] DEBUG 2024/02/04 09:34:12 grid currents: [-3.56 -0 -0]A
[site  ] DEBUG 2024/02/04 09:34:12 site power: -659W
[lp-1  ] DEBUG 2024/02/04 09:34:16 charge voltages: [242 0 0]V
[lp-1  ] DEBUG 2024/02/04 09:34:16 detected connected phases: 1p
[lp-1  ] DEBUG 2024/02/04 09:34:17 charge currents: [5.04 0 0]A
[lp-1  ] DEBUG 2024/02/04 09:34:17 detected active phases: 1p
[lp-1  ] DEBUG 2024/02/04 09:34:20 charger status: C
[lp-1  ] DEBUG 2024/02/04 09:34:20 soc estimated: 80.35% (vehicle: 80.00%)
[lp-1  ] DEBUG 2024/02/04 09:34:20 vehicle soc: 80%
[lp-1  ] DEBUG 2024/02/04 09:34:20 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/02/04 09:34:20 vehicle range: 320km
[lp-1  ] DEBUG 2024/02/04 09:34:22 pv charge current: 9.91A = 7.04A + 2.87A (-659W @ 1p)
[site  ] DEBUG 2024/02/04 09:34:38 ----
[lp-1  ] DEBUG 2024/02/04 09:34:43 charge power: 1204W
[site  ] DEBUG 2024/02/04 09:34:43 pv power: 2627W
[site  ] DEBUG 2024/02/04 09:34:43 battery soc: 12%
[site  ] DEBUG 2024/02/04 09:34:43 battery power: 0W
[site  ] DEBUG 2024/02/04 09:34:43 grid power: -892W
[site  ] DEBUG 2024/02/04 09:34:43 grid currents: [-3.64 -0 -0]A
[site  ] DEBUG 2024/02/04 09:34:43 site power: -692W
[lp-1  ] DEBUG 2024/02/04 09:34:44 charge voltages: [243 0 0]V
[lp-1  ] DEBUG 2024/02/04 09:34:44 detected connected phases: 1p
[lp-1  ] DEBUG 2024/02/04 09:34:46 charge currents: [5.05 0 0]A
[lp-1  ] DEBUG 2024/02/04 09:34:46 detected active phases: 1p
[lp-1  ] DEBUG 2024/02/04 09:34:48 charger status: C
[lp-1  ] DEBUG 2024/02/04 09:34:48 soc estimated: 80.36% (vehicle: 80.00%)
[lp-1  ] DEBUG 2024/02/04 09:34:48 vehicle soc: 80%
[lp-1  ] DEBUG 2024/02/04 09:34:48 vehicle soc limit: 100%
[lp-1  ] DEBUG 2024/02/04 09:34:48 vehicle range: 320km
[lp-1  ] DEBUG 2024/02/04 09:34:49 pv charge current: 10.1A = 7.05A + 3.01A (-692W @ 1p)
[lp-1  ] DEBUG 2024/02/04 09:34:50 max charge current: 10.1A
```

### What type of operating system are you running?

Linux

wavelov3r, 2024-02-04:
[image]

VolkerK62, 2024-02-04:
Maybe the car doesn´t want more.
What happens, if you change to "fast" mode?

wavelov3r, 2024-02-04:
Nothing happen! i've charged the car at 7kwh with success in past days.

VolkerK62, 2024-02-04:
>..i've noticed also if i set residualpower to -4000, in the log it increments the site power???

that´s ok, that´s the way it works.
```
[lp-1  ] DEBUG 2024/02/04 10:45:21 set charge mode: now
[lp-1  ] DEBUG 2024/02/04 10:45:37 max charge current: 25A
```
that´s also ok. Any problem with the meter have no influence here.

The wallbox doesn´t react or the car doesn´t want more.
Did you set any limitation within the car?