evcc-io/evcc #12068: Tesla Login don't work after move to template "tesla-command" (evcc 0.124.0)

dev-da77, 2024-02-05:
### Describe the bug

I updated my evcc.yaml according the changes on evcc 0.124.0 for my tesla. Now Tesla Login does not work. See below.

### Steps to reproduce

1. update to 0.124.0
2. update vehicle configuration (with new generated tokens) - https://docs.evcc.io/en/docs/devices/vehicles/#latest
3. sudo systemctl restart evcc

### Configuration details

```yaml
vehicles:
- type: template
  template: tesla-command
  title: Tesla
  accessToken: <...>
  refreshToken: EU_<...>
  vin: <...>
  name: ev1
```

### Log details

```text
sudo evcc -l debug vehicle

[main  ] INFO 2024/02/04 22:39:24 evcc 0.124.0
[main  ] INFO 2024/02/04 22:39:24 using config file: /etc/evcc.yaml
[db    ] INFO 2024/02/04 22:39:24 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2024/02/04 22:39:24 connecting evcc-xxxxxxxxx at tcp://ha:1883
[mqtt  ] DEBUG 2024/02/04 22:39:24 tcp://ha:1883 connected
[main  ] ERROR 2024/02/04 22:39:26 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla-command': oauth2: "login_required" "Login required"
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'tesla-command': oauth2: "login_required" "Login required"
Capacity: 0.0kWh
Features: [Offline]
```
```

### What type of operating system are you running?

Linux

### Version

0.124.0

AnBo83, 2024-02-05:
Use the nightly, it's a Bug in 0.124.0 https://github.com/evcc-io/evcc/issues/11987

ifeulner, 2024-02-05:
Also the old "tesla" template seems not to work anymore? (0.124)

AnBo83, 2024-02-05:
No, the old Customer-API has been disabled by Tesla.

closed 2024-02-05 by andig

stepir, 2024-02-05:
I'm using the latest nightly and can't set it up - via cli

```
Teste die Konfiguration von Tesla aktuell ...
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x0 pc=0x1e6b9c3]
```

and via UI

[image]