evcc-io/evcc #12070: SolarEdge weird numbers one of the batteries

stepir, 2024-02-05:
### Describe the bug

Hi

I have a system composed of one modbus master and 3 different slaves. The master and the first 2 slaves have batteries.

Only the battery of the master is giving weird power numbers:

[image]

### Steps to reproduce

the three batteries are configured in the same way only the modbus id changes

[image]

[image]

[image]

[image]

### Configuration details

```yaml
batteries are configured via UI
```

### Log details

```text
[site  ] DEBUG 2024/02/05 14:00:45 ----
[lp-1  ] DEBUG 2024/02/05 14:00:45 charge power: 0W
[site  ] DEBUG 2024/02/05 14:00:46 pv power: 4026W
[site  ] DEBUG 2024/02/05 14:00:46 battery soc: 6%
[site  ] DEBUG 2024/02/05 14:00:46 battery power: 340282346638528859811704183484516925440W
[site  ] DEBUG 2024/02/05 14:00:46 grid power: 560W
[site  ] DEBUG 2024/02/05 14:00:46 site power: 340282346638528859811704183484516925440W
[lp-1  ] DEBUG 2024/02/05 14:00:46 charge voltages: [222 221 217]V
[lp-1  ] DEBUG 2024/02/05 14:00:46 detected connected phases: 3p
[lp-1  ] DEBUG 2024/02/05 14:00:46 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/02/05 14:00:46 charge total import: 7635.000kWh
[lp-1  ] DEBUG 2024/02/05 14:00:46 charger status: A
[site  ] DEBUG 2024/02/05 14:00:50 ----
```
```

### What type of operating system are you running?

Docker container

### Version

latest nightly

andig, 2024-02-05:
Having documented that we will not handle docker issues here this issue should be closed. Please provide output of

    evcc meter --trace log

for further analysis.

closed 2024-02-05 by andig

premultiply, 2024-02-05:
Well known bug in SE firmware on sleeping batteries with only NaN-Value in non-swapped word order.

Please contact Solaredge to fix this bug in their firmware.

stepir, 2024-02-05:
this is id 1

```
/app # evcc sunspec
[main  ] INFO 2024/02/05 14:21:33 evcc 0.124.0 (507a0eb4)
Error: accepts 1 arg(s), received 0
Usage:
  evcc sunspec <connection> [flags]

Flags:
  -i, --id int   Slave id (default 1)

Global Flags:
  -c, --config string   Config file (default "~/evcc.yaml" or "/etc/evcc.yaml")
  -h, --help            Help
  -l, --log string      Log level (fatal, error, warn, info, debug, trace) (default "info")
      --log-headers     Log headers

accepts 1 arg(s), received 0
/app # evcc sunspec 192.168.2.225:9000
[main  ] INFO 2024/02/05 14:21:46 evcc 0.124.0 (507a0eb4)
--------- Model 1 common ---------
Mn         SolarEdge     string
Md   SE5000H-RW000BNN    string
Opt                      string
Vr           0004.001    string
SN           7309592F    string
DA               1.00    uint16
Pad            0x0065    pad
--------- Model 101 inverter ---------
A                    1.65    uint16
AphA                 1.65    uint16
AphB                  n/a    uint16
AphC                  n/a    uint16
A_SF                   -2    sunssf
PPVphAB            231.10    uint16
PPVphBC               n/a    uint16
PPVphCA               n/a    uint16
PhVphA             231.10    uint16
PhVphB                n/a    uint16
PhVphC                n/a    uint16
V_SF                   -1    sunssf
W                    0.00    int16
W_SF                    0    sunssf
Hz                  49.98    uint16
Hz_SF                  -3    sunssf
VA                 381.10    int16
VA_SF                  -1    sunssf
VAr               -381.10    int16
VAr_SF                 -1    sunssf
PF                   9.41    int16
PF_SF                  -3    sunssf
WH             4125095.00    acc32
WH_SF                   0    sunssf
DCA                  0.00    uint16
DCA_SF                  0    sunssf
DCV                406.40    uint16
DCV_SF                 -1    sunssf
DCW                  0.00    int16
DCW_SF                  0    sunssf
TmpCab                n/a    int16
TmpSnk              34.99    int16
TmpTrns               n/a    int16
TmpOt                 n/a    int16
Tmp_SF                 -2    sunssf
St                      4    enum16
StVnd                   0    enum16
Evt1                  n/a    bitfield32
Evt2                  n/a    bitfield32
EvtVnd1        0x00000000    bitfield32
EvtVnd2               n/a    bitfield32
EvtVnd3               n/a    bitfield32
EvtVnd4        0x00000000    bitfield32
--------- Model 1 common ---------
Mn           WattNode    string
Md      WND-3Y-400-MB    string
Opt          Export+I    string
Vr                 25    string
SN            4124348    string
DA               2.00    uint16
Pad            0x00cb    pad
--------- Model 203 ac_meter ---------
A                           -0.80    int16
AphA                        -4.50    int16
AphB                        -4.50    int16
AphC                         8.20    int16
A_SF                           -1    sunssf
PhV                        230.31    int16
PhVphA                     230.31    int16
PhVphB                     231.23    int16
PhVphC                     226.64    int16
PPV                       -258.07    int16
PhVphAB                   -257.88    int16
PhVphBC                   -257.58    int16
PhVphCA                   -258.75    int16
V_SF                           -2    sunssf
Hz                          49.96    int16
Hz_SF                          -2    sunssf
W                          210.00    int16
WphA                       923.00    int16
WphB                       978.00    int16
WphC                     -1690.00    int16
W_SF                            0    sunssf
VA                         705.00    int16
VAphA                      990.00    int16
VAphB                     1008.00    int16
VAphC                     1692.00    int16
VA_SF                           0    sunssf
VAR                        673.00    int16
VARphA                     359.00    int16
VARphB                     242.00    int16
VARphC                      70.00    int16
VAR_SF                          0    sunssf
PF                         -28.09    int16
PFphA                      -89.44    int16
PFphB                      -93.97    int16
PFphC                       99.13    int16
PF_SF                          -2    sunssf
TotWhExp              11916631.00    acc32
TotWhExpPhA           11245820.00    acc32
TotWhExpPhB           16041058.00    acc32
TotWhExpPhC           10072140.00    acc32
TotWhImp              24829780.00    acc32
TotWhImpPhA           15250936.00    acc32
TotWhImpPhB            9606084.00    acc32
TotWhImpPhC           25415146.00    acc32
TotWh_SF                        0    sunssf
TotVAhExp                     NaN    acc32
TotVAhExpPhA                  NaN    acc32
TotVAhExpPhB                  NaN    acc32
TotVAhExpPhC                  NaN    acc32
TotVAhImp                     NaN    acc32
TotVAhImpPhA                  NaN    acc32
TotVAhImpPhB                  NaN    acc32
TotVAhImpPhC                  NaN    acc32
TotVAh_SF                     n/a    sunssf
TotVArhImpQ1                  NaN    acc32
TotVArhImpQ1PhA               NaN    acc32
TotVArhImpQ1PhB               NaN    acc32
TotVArhImpQ1PhC               NaN    acc32
TotVArhImpQ2                  NaN    acc32
TotVArhImpQ2PhA               NaN    acc32
TotVArhImpQ2PhB               NaN    acc32
TotVArhImpQ2PhC               NaN    acc32
TotVArhExpQ3                  NaN    acc32
TotVArhExpQ3PhA               NaN    acc32
TotVArhExpQ3PhB               NaN    acc32
TotVArhExpQ3PhC               NaN    acc32
TotVArhExpQ4                  NaN    acc32
TotVArhExpQ4PhA               NaN    acc32
TotVArhExpQ4PhB               NaN    acc32
TotVArhExpQ4PhC               NaN    acc32
TotVArh_SF                    n/a    sunssf
Evt                    0x00000000    bitfield32
```

stepir, 2024-02-05:
> Well known bug in SE firmware on sleeping batteries with only NaN-Value in non-swapped word order.
> 
> Please contact Solaredge to fix this bug in their firmware.

ah you are right now I remember - I've opened a bug with SE support and the answer it is by design when battery is in sleeping state.

Can EVCC just report 0 in this case? or any other suggestion to work around the issue?

premultiply, 2024-02-05:
As written above bug in SE Firmware in non-SunSpec battery power register.

You need to get SE to fix their firmware.

stepir, 2024-02-05:
as said I tried an they won't change it - will remove batteries from my config

stepir, 2024-02-05:
> > Well known bug in SE firmware on sleeping batteries with only NaN-Value in non-swapped word order.
> > Please contact Solaredge to fix this bug in their firmware.
> 
> In other words: If device is asleep, SE will return a NaN value (or think it does) but get the encoding wrong. That's a firmware bug.
> 
> > ah you are right now I remember - I've opened a bug with SE support and the answer it is by design when battery is in sleeping state.
> 
> Oh yeah. The could have returned any bit pattern than, maybe "off" repeated, say 011011011011011 for the fun of it. Returning `NaN` may be by design, returning something that is a valid number but might look like it could be `NaN` if the encoding were different is clearly a bug. Pretty sure they didn't try to even understand the issue...

for the sake of knowledge this is still the case in the latest firmware (I just updated it today)

premultiply, 2024-02-05:
evcc already returns 0 for (real) NaN values but SE firmware does not return a NaN value but another very large float number as they swapped the words ONLY for this value.
They think it is a NaN value but the words are swapped in relation to all other values.

We (and nobody) can not decode binary numbers which switch their binary encoding in between.

stepir, 2024-02-05:
> evcc already returns 0 for (real) NaN values but SE firmware does not return a NaN value but another very large float number as they swapped the words ONLY for this value. They think it is a NaN value but the words are swapped in relation to all other values.
> 
> We (and nobody) can not decode binary numbers which switch their binary encoding in between.
> 
> Please contact SE to fix their firmware.

well this https://github.com/WillCodeForCats/solaredge-modbus-multi correctly report 'unknown' value in this state

[image]

[image]

[image]

As I said i've contacted SE but with no luck

andig, 2024-02-05:
@premultiply if we really wanted to do this- and I'm not convinced we want- one could potentially use the new `switch` plugin

premultiply, 2024-02-05:
🤔 Mmmmh.

stepir, 2024-02-05:
Not sure what I can do to help - but for sure count on me for testing :-D