evcc-io/evcc #12071: Hoymiles OpenDTU error

MrJoki007, 2024-02-05:
### Describe the bug

When i try to add my Hoymiles inverter over opendtu i get this error: power: unexpected status: 401 (Unauthorized)
Im trying to add it over the experimental gui
[image]

### Steps to reproduce

1. Add hoymiles opendtu solar meter
2. 
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: 192.168.178.84 # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: xxxxxxxxx

interval: 10s # control cycle interval

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

chargers:
- type: template
  template: go-e-v3 
  host: 192.168.178.47  
  name: wallbox1

loadpoints:
- title: carport
  charger: wallbox1
  mode: pv
  phases: 0

site:
  title: home
  meters:
    grid: grid1
  residualPower: 100

meters:  
- type: custom # IR-Lesekopf Tasmota  
  name: grid1  
  power:  
    source: http  
    uri: http://192.168.178.24/cm?cmnd=Status%208  
    jq: .StatusSNS.Gesamt.Leistung  

vehicles:
  - name: kona
    type: template
    template: offline
    title: Kona
    capacity: 39.2 # Akkukapazität in kWh (optional)
```

### Log details

```text
[main  ] INFO 2024/02/05 14:35:31 using config file: /mnt/Data/share/Daten/DockerApps/evcc/evcc_patched/evcc.yaml
[main  ] INFO 2024/02/05 14:35:31 starting ui and api at :7070
[db    ] INFO 2024/02/05 14:35:31 using sqlite database: /root/.evcc/evcc.db
[lp-1  ] DEBUG 2024/02/05 14:35:32 charge total import: 6387.151kWh
[site  ] INFO 2024/02/05 14:35:32 site config:
[site  ] INFO 2024/02/05 14:35:32   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/02/05 14:35:32     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/02/05 14:35:32     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2024/02/05 14:35:32     pv 2:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/02/05 14:35:32     battery 1: power ✓ energy ✓ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2024/02/05 14:35:32   vehicles:
[site  ] INFO 2024/02/05 14:35:32     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2024/02/05 14:35:32 loadpoint 1:
[lp-1  ] INFO 2024/02/05 14:35:32   mode:        pv
[lp-1  ] INFO 2024/02/05 14:35:32   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/02/05 14:35:32   meters:      charge ✓
[lp-1  ] INFO 2024/02/05 14:35:32     charge:    power ✓ energy ✓ currents ✓
[site  ] DEBUG 2024/02/05 14:35:32 set buffer soc: 90
[site  ] DEBUG 2024/02/05 14:35:32 set buffer start soc: 98
[site  ] DEBUG 2024/02/05 14:35:32 set priority soc: 90
[lp-1  ] DEBUG 2024/02/05 14:35:32 phase timer inactive
[lp-1  ] DEBUG 2024/02/05 14:35:32 pv timer inactive
[lp-1  ] DEBUG 2024/02/05 14:35:32 guard timer inactive
[lp-1  ] DEBUG 2024/02/05 14:35:32 max charge current: 6A
[lp-1  ] DEBUG 2024/02/05 14:35:32 guard enable in 5m0s
[site  ] WARN 2024/02/05 14:35:32 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] DEBUG 2024/02/05 14:35:32 ----
[lp-1  ] DEBUG 2024/02/05 14:35:32 charge power: 1700W
[site  ] ERROR 2024/02/05 14:35:33 pv 2 power: unexpected status: 401 (Unauthorized)
[site  ] DEBUG 2024/02/05 14:35:33 pv power: 1746W
[site  ] DEBUG 2024/02/05 14:35:33 battery soc: 91%
[site  ] DEBUG 2024/02/05 14:35:33 battery power: -5W
[site  ] DEBUG 2024/02/05 14:35:33 grid power: 33W
[site  ] DEBUG 2024/02/05 14:35:33 site power: 128W
[lp-1  ] DEBUG 2024/02/05 14:35:33 charge currents: [6.1 0 0]A
[lp-1  ] DEBUG 2024/02/05 14:35:33 charge total import: 6387.152kWh
[lp-1  ] DEBUG 2024/02/05 14:35:33 charger status: C
[lp-1  ] INFO 2024/02/05 14:35:33 car connected
[lp-1  ] DEBUG 2024/02/05 14:35:33 vehicle api refresh
[lp-1  ] DEBUG 2024/02/05 14:35:33 pv timer elapse
[lp-1  ] DEBUG 2024/02/05 14:35:33 pv timer inactive
[lp-1  ] DEBUG 2024/02/05 14:35:33 charger: guard elapse
[lp-1  ] DEBUG 2024/02/05 14:35:33 guard timer inactive
[lp-1  ] DEBUG 2024/02/05 14:35:33 charge total import: 6387.152kWh
[lp-1  ] INFO 2024/02/05 14:35:33 start charging ->
[lp-1  ] DEBUG 2024/02/05 14:35:33 wake-up timer: stop
[lp-1  ] DEBUG 2024/02/05 14:35:33 available power 1572W < 4140W min 3p threshold
[lp-1  ] DEBUG 2024/02/05 14:35:33 start phase scale1p timer
[lp-1  ] DEBUG 2024/02/05 14:35:33 phase scale1p in 3m0s
[lp-1  ] DEBUG 2024/02/05 14:35:33 pv charge current: 5.81A = 6A + -0.186A (128W @ 3p)
^C[lp-1  ] DEBUG 2024/02/05 14:35:35 charge total import: 6387.153kWh
```

### What type of operating system are you running?

Linux

### Version

0.124.0

andig, 2024-02-05:
What happens if you open 

    http://{{ .host }}/api/livedata/status

in your browser?

andig, 2024-02-06:
HTTP 401 means "unauthorised". Why does your device require authorisation from evcc but not from browser? Did you enable authorisation in OpenDTU? Which authorisation? Can you turn it off? Do we need to add it to evcc?

MrJoki007, 2024-02-06:
> HTTP 401 means "unauthorised". Why does your device require authorisation from evcc but not from browser? Did you enable authorisation in OpenDTU? Which authorisation? Can you turn it off? Do we need to add it to evcc?

I found the problem.
When you was logged in once with the browser, you can log out and still see everything.
But when i did it with another browser, that was never logged in it does not work.
You have to enable this setting in order to get it working.
[image]

andig, 2024-02-06:
Perfect!

closed 2024-02-06 by andig