evcc-io/evcc #12073: Tesla M3 Laderegelung funktioniert nicht mehr

pit65535, 2024-02-05:
### Describe the bug

Ich habe grad auf die neueste Version upgedated (HA-Addon) und auch neue API-tokens über evcc.io erzeugt, soweit sieht alles ganz gut aus, aber irgendwie ändert evcc die Ladestromstärke nicht und zeigt in der Oberfläche den folgenden Fehler:
Garage: max charge current 1A: vehicle not capable of current control
Das LOG dazu sieht so aus:
s.u.

kann mir jemand einen Tip geben, wo es noch hängt?

### Steps to reproduce

1. Update auf 0.124.0 durchgeführt
2. neue AP-Tokens erzeugt
3.
...

### Configuration details

```yaml
evcc läuft als addon in Homeassistant
```

### Log details

```text
[lp-1  ] DEBUG 2024/02/05 15:00:38 charge power: 5762W
[lp-2  ] DEBUG 2024/02/05 15:00:38 charge power: 0W
[site  ] DEBUG 2024/02/05 15:00:38 pv power: 6816W
[site  ] DEBUG 2024/02/05 15:00:38 battery soc: 100%
[site  ] DEBUG 2024/02/05 15:00:38 battery power: 5W
[site  ] DEBUG 2024/02/05 15:00:38 grid power: 21W
[site  ] DEBUG 2024/02/05 15:00:38 grid currents: [2.3 1.42 2.6]A
[site  ] DEBUG 2024/02/05 15:00:38 site power: 27W
[lp-1  ] DEBUG 2024/02/05 15:00:38 charge voltages: [239 235 238]V
[lp-1  ] DEBUG 2024/02/05 15:00:38 detected connected phases: 3p
[lp-1  ] DEBUG 2024/02/05 15:00:38 charge currents: [8.1 8.1 8.1]A
[lp-1  ] DEBUG 2024/02/05 15:00:38 detected active phases: 3p
[lp-1  ] DEBUG 2024/02/05 15:00:38 charger status: C
[lp-1  ] DEBUG 2024/02/05 15:00:38 soc estimated: 78.87% (vehicle: 78.00%)
[lp-1  ] DEBUG 2024/02/05 15:00:38 vehicle soc: 79%
[lp-1  ] DEBUG 2024/02/05 15:00:38 vehicle soc limit: 90%
[lp-1  ] DEBUG 2024/02/05 15:00:38 vehicle range: 426km
[lp-1  ] DEBUG 2024/02/05 15:00:38 pv charge current: 0A = 0A + -0.0387A (27W @ 3p)
[lp-1  ] DEBUG 2024/02/05 15:00:38 site power 27W >= 0W disable threshold
[lp-1  ] DEBUG 2024/02/05 15:00:38 pv disable timer start: 3m0s
[lp-1  ] DEBUG 2024/02/05 15:00:38 pv disable in 3m0s
[lp-1  ] ERROR 2024/02/05 15:00:38 max charge current 1A: vehicle not capable of current control
[site  ] DEBUG 2024/02/05 15:00:48 ----
[lp-1  ] DEBUG 2024/02/05 15:00:48 charge power: 5745W
[lp-2  ] DEBUG 2024/02/05 15:00:48 charge power: 0W
[site  ] DEBUG 2024/02/05 15:00:48 pv power: 6849W
[site  ] DEBUG 2024/02/05 15:00:48 battery soc: 100%
[site  ] DEBUG 2024/02/05 15:00:48 battery power: -43W
[site  ] DEBUG 2024/02/05 15:00:48 grid power: -32W
[site  ] DEBUG 2024/02/05 15:00:48 grid currents: [2.14 1.41 2.56]A
[site  ] DEBUG 2024/02/05 15:00:48 site power: -75W
[lp-2  ] DEBUG 2024/02/05 15:00:48 charge voltages: [1.5 0 0]V
[lp-2  ] DEBUG 2024/02/05 15:00:48 charge currents: [0 0 0.3]A
[lp-2  ] DEBUG 2024/02/05 15:00:48 charger status: A
[site  ] DEBUG 2024/02/05 15:00:58 ----
[lp-1  ] DEBUG 2024/02/05 15:00:58 charge power: 5719W
[lp-2  ] DEBUG 2024/02/05 15:00:58 charge power: 0W
[site  ] DEBUG 2024/02/05 15:00:58 pv power: 6916W
[site  ] DEBUG 2024/02/05 15:00:58 battery soc: 100%
[site  ] DEBUG 2024/02/05 15:00:58 battery power: -54W
[site  ] DEBUG 2024/02/05 15:00:58 grid power: -51W
[site  ] DEBUG 2024/02/05 15:00:58 grid currents: [2.07 1.43 2.63]A
[site  ] DEBUG 2024/02/05 15:00:58 site power: -105W
[lp-1  ] DEBUG 2024/02/05 15:00:58 charge voltages: [239 235 238]V
[lp-1  ] DEBUG 2024/02/05 15:00:58 detected connected phases: 3p
[lp-1  ] DEBUG 2024/02/05 15:00:58 charge currents: [8 8.1 8]A
[lp-1  ] DEBUG 2024/02/05 15:00:58 detected active phases: 3p
[lp-1  ] DEBUG 2024/02/05 15:00:58 charger status: C
[lp-1  ] DEBUG 2024/02/05 15:00:58 soc estimated: 78.91% (vehicle: 78.00%)
[lp-1  ] DEBUG 2024/02/05 15:00:58 vehicle soc: 79%
[lp-1  ] DEBUG 2024/02/05 15:00:58 vehicle soc limit: 90%
[lp-1  ] DEBUG 2024/02/05 15:00:58 vehicle range: 426km
[lp-1  ] DEBUG 2024/02/05 15:00:58 pv charge current: 0.152A = 0A + 0.152A (-105W @ 3p)
[lp-1  ] DEBUG 2024/02/05 15:00:58 pv disable timer reset
[lp-1  ] DEBUG 2024/02/05 15:00:58 pv timer inactive
[lp-1  ] ERROR 2024/02/05 15:00:58 max charge current 1A: vehicle not capable of current control
[site  ] DEBUG 2024/02/05 15:01:08 ----
[lp-1  ] DEBUG 2024/02/05 15:01:08 charge power: 5727W
[lp-2  ] DEBUG 2024/02/05 15:01:08 charge power: 0W
[site  ] DEBUG 2024/02/05 15:01:08 pv power: 6858W
[site  ] DEBUG 2024/02/05 15:01:08 battery soc: 100%
[site  ] DEBUG 2024/02/05 15:01:08 battery power: 43W
[site  ] DEBUG 2024/02/05 15:01:08 grid power: -45W
[site  ] DEBUG 2024/02/05 15:01:08 grid currents: [2.16 1.6 2.59]A
[site  ] DEBUG 2024/02/05 15:01:08 site power: -2W
[lp-2  ] DEBUG 2024/02/05 15:01:08 charge voltages: [1.5 0 0]V
[lp-2  ] DEBUG 2024/02/05 15:01:08 charge currents: [0 0 0.3]A
[lp-2  ] DEBUG 2024/02/05 15:01:08 charger status: A
[site  ] DEBUG 2024/02/05 15:01:18 ----
[lp-1  ] DEBUG 2024/02/05 15:01:18 charge power: 5774W
[lp-2  ] DEBUG 2024/02/05 15:01:18 charge power: 0W
[site  ] DEBUG 2024/02/05 15:01:18 pv power: 6882W
[site  ] DEBUG 2024/02/05 15:01:18 battery soc: 100%
[site  ] DEBUG 2024/02/05 15:01:18 battery power: -38W
[site  ] DEBUG 2024/02/05 15:01:18 grid power: -65W
[site  ] DEBUG 2024/02/05 15:01:18 grid currents: [2.15 1.6 2.72]A
[site  ] DEBUG 2024/02/05 15:01:18 site power: -103W
[lp-1  ] DEBUG 2024/02/05 15:01:18 charge voltages: [239 235 239]V
[lp-1  ] DEBUG 2024/02/05 15:01:18 detected connected phases: 3p
[lp-1  ] DEBUG 2024/02/05 15:01:18 charge currents: [8.1 8.1 8.1]A
[lp-1  ] DEBUG 2024/02/05 15:01:18 detected active phases: 3p
[lp-1  ] DEBUG 2024/02/05 15:01:18 charger status: C
[lp-1  ] DEBUG 2024/02/05 15:01:18 soc estimated: 78.94% (vehicle: 78.00%)
[lp-1  ] DEBUG 2024/02/05 15:01:18 vehicle soc: 79%
[lp-1  ] DEBUG 2024/02/05 15:01:18 vehicle soc limit: 90%
[lp-1  ] DEBUG 2024/02/05 15:01:18 vehicle range: 426km
[lp-1  ] DEBUG 2024/02/05 15:01:18 pv charge current: 0.149A = 0A + 0.149A (-103W @ 3p)
[lp-1  ] ERROR 2024/02/05 15:01:18 max charge current 1A: vehicle not capable of current control
[site  ] DEBUG 2024/02/05 15:01:28 ----
[lp-1  ] DEBUG 2024/02/05 15:01:28 charge power: 5777W
[lp-2  ] DEBUG 2024/02/05 15:01:28 charge power: 0W
[site  ] DEBUG 2024/02/05 15:01:28 pv power: 6886W
[site  ] DEBUG 2024/02/05 15:01:28 battery soc: 100%
[site  ] DEBUG 2024/02/05 15:01:28 battery power: -60W
[site  ] DEBUG 2024/02/05 15:01:28 grid power: -58W
[site  ] DEBUG 2024/02/05 15:01:28 grid currents: [2.15 1.44 2.65]A
[site  ] DEBUG 2024/02/05 15:01:28 site power: -117W
[lp-2  ] DEBUG 2024/02/05 15:01:28 charge voltages: [1.5 0 0]V
[lp-2  ] DEBUG 2024/02/05 15:01:28 charge currents: [0 0 0.3]A
[lp-2  ] DEBUG 2024/02/05 15:01:28 charger status: A
```

### What type of operating system are you running?

HomeAssistant Add-on

### Version

0.124.0

AnBo83, 2024-02-05:
Das ist ein Bug in der 0.124.0
Der Fix ist in der Nightly bereits enthalten, über HA als AddOn aber noch nicht verfügbar. (nur über Umwege eines Forks und neuer Repo in HA)

andig, 2024-02-05:
Duplikat: https://github.com/evcc-io/evcc/issues/11846

closed 2024-02-05 by andig