evcc-io/evcc #12077: "main.loadpointSettings.disclaimerHint" missing

TobiasNeuschulz, 2024-02-05:
### Describe the bug

No Text configured for variable "main.loadpointSettings.disclaimerHint".
[image]

### Steps to reproduce

1. Open GUI
2. Open Planner -> Arrival
3. main.loadpointSettings.disclaimerHint: <Text>
...

### Configuration details

```yaml
Wasn't able to see a text for this variable in for example: "evcc/i18n/de.toml"
```

### Log details

```text
---
```

### What type of operating system are you running?

Linux

### Version

0.124.0

andig, 2024-02-05:
@naltatis looks like its in `batterySettings` instead https://github.com/evcc-io/evcc/commit/20fb3edfc90c07fe00c49519c75c54de6fb85b59

naltatis, 2024-02-05:
Der Text ist hier versehentlich entfernt worden: https://github.com/evcc-io/evcc/pull/11571
Ich stell den wieder her und benenn den gleich korrekt für den Anwendungsfall.

closed 2024-02-05 by andig