evcc-io/evcc #12078: API: vehicle object nicht vollständig vs. "evcc vehicle"

cvg-de, 2024-02-05:
### Describe the bug

Zumindest bei nicht verbundenem Fahrzeug fehlen im API / vehicle z.B. SOC, Range und Odometer. Auf der Kommandozeile werden sie bei "evcc vehicle" angezeigt.
Es wäre gut, wenn in der API die gleichen Infos stünden wie bei Aufruf auf der Kommandozeile.

```
API:
GET http://<address>:7070/api/state
{{<...> "vehicles":{"ev4":{"title":"Q4 etron","icon":"car","capacity":77}},"version":"0.124.0"}}

$ evcc vehicle
[main  ] INFO 2024/02/05 18:35:25 evcc 0.124.0
[main  ] INFO 2024/02/05 18:35:25 using config file: /etc/evcc.yaml
[db    ] INFO 2024/02/05 18:35:25 using sqlite database: /home/admin/.evcc/evcc.db
Soc:            77%
Capacity:       77.0kWh
Charge status:  A
Range:          308km
Odometer:       9025km
Finish time:    2024-02-05 17:24:00 +0100 CET
Climate active: false
Target Soc:     80%
Features:       []
```

### Steps to reproduce

1. http://<address>:7070/api/state aufrufen
2. evcc vehicle aufrufen
3. vergleichen

### Configuration details

```yaml
vehicles:
- type: template
  template: audi
  title: Q4 etron
  user: a@bc.de
  password: geheim
  vin: WAUZZZxxx
  capacity: 77
  name: ev4
```

### Log details

```text
Weder API noch das evcc-Kommando erzeugen Log-Einträge
```

### What type of operating system are you running?

Linux

### Version

0.124.0

andig, 2024-02-05:
Es werden nur aktive Fahrzeuge abgefragt.

closed 2024-02-05 by andig

cvg-de, 2024-02-06:
Habe das jetzt mit verbundenem Fahrzeug getestet: Die API zeigt auch bei aktivem Fahrzeug im Vehicle-Objekt nur die oben genannten Standard-Angaben. Aus meiner Sicht wäre es folgerichtig, in der API unter Vehicle alle bekannten Infos zum Fahrzeug auszugeben, wie auf der Kommandozeile.

andig, 2024-02-06:
Ja, das kommt nach und nach