evcc-io/evcc #12080: Sunny Island 4.4 Ladezustand der Batterie wird nicht angezeigt

MiRi1975, 2024-02-05:
### Describe the bug

Ich sehe, dass die Batterie sich entlädt, aber der Ladezustand steht bei 0% in der Übersicht.
Hatte ich mit dem template-hybrid in der Version 123.9, bei sma-inverter-modus lief alles normal, jedoch funktionierte das Schnellladen ohne Batterienutzung nicht wie gewollt. Dies lief mit dem sma-hybrid template, jedoch ohne Batterieladestand., Jetzt, mit 124.0 und verwendetem template SMA-inverter-modbus, funktioniert auch dann die Anzeige nicht mehr.

### Steps to reproduce

1. 124.0 installieren mit template sma-inverter-modbus 
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

meters:
- type: template
  template: sma-home-manager 
  usage: grid  
  host: 192.168.2.116  
  name: grid1
- type: template
  template: sma-inverter 
  usage: pv  
  host: 192.168.2.129
  password: ****  
  name: pv2
- type: template
  template: sma-inverter 
  usage: pv  
  host: 192.168.2.120  
  password: ****  
  name: pv3
- type: template
  template: sma-inverter-modbus
  usage: battery  
  host: 192.168.2.125  
  name: battery4
  modbus: tcpip
  id: 3
  port: 502

chargers:
- type: template
  template: easee 
  user: ****  
  password: *****  
  charger: **  
  name: wallbox6

vehicles:
- type: template
  template: tesla
  title: Model Y
  accessToken: *-**
  refreshToken: ****-*
  vin: ******
  capacity: 76.8
[... 167 lines cut ...]
[site  ] DEBUG 2024/02/05 18:56:35 site power: 342W
[lp-1  ] DEBUG 2024/02/05 18:56:35 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/02/05 18:56:35 charge total import: 1751.850kWh
[lp-1  ] DEBUG 2024/02/05 18:56:35 charger status: A
[site  ] DEBUG 2024/02/05 18:56:45 ----
[lp-1  ] DEBUG 2024/02/05 18:56:45 charge power: 0W
[site  ] DEBUG 2024/02/05 18:56:45 pv power: 0W
[site  ] DEBUG 2024/02/05 18:56:45 battery soc: 0%
[site  ] DEBUG 2024/02/05 18:56:45 battery power: 340W
[site  ] DEBUG 2024/02/05 18:56:45 grid power: -1W
[site  ] DEBUG 2024/02/05 18:56:45 grid powers: [-221 196 24]W
[site  ] DEBUG 2024/02/05 18:56:45 grid currents: [-1.86 1.59 0.577]A
[site  ] DEBUG 2024/02/05 18:56:45 site power: 339W
[lp-1  ] DEBUG 2024/02/05 18:56:45 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2024/02/05 18:56:45 charge total import: 1751.850kWh
[lp-1  ] DEBUG 2024/02/05 18:56:45
```

### What type of operating system are you running?

Linux

### Version

124.0

andig, 2024-02-05:
Bitte

    evcc meter —log trace

MiRi1975, 2024-02-05:
[main  ] INFO 2024/02/05 19:12:54 evcc 0.124.0
[main  ] INFO 2024/02/05 19:12:54 using config file: /etc/evcc.yaml
[db    ] INFO 2024/02/05 19:12:54 using sqlite database: /home/pi/.evcc/evcc.db
[db    ] TRACE 2024/02/05 19:12:54 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2024/02/05 19:12:54 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/02/05 19:12:54 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/05 19:12:54 SELECT * FROM `settings` 0 <nil>
[db    ] TRACE 2024/02/05 19:12:54 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="devices" -1 <nil>
[db    ] TRACE 2024/02/05 19:12:54 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="device_details" -1 <nil>
[db    ] TRACE 2024/02/05 19:12:54 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2024/02/05 19:12:54 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2024/02/05 19:12:54 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/05 19:12:54 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="config_details" -1 <nil>
[db    ] TRACE 2024/02/05 19:12:54 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "config_details" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2024/02/05 19:12:54 SELECT * FROM `config_details` LIMIT 1 -1 <nil>
[db    ] TRACE 2024/02/05 19:12:54 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_configs_details"" %" OR sql LIKE "%CONSTRAINT fk_configs_details %" OR sql LIKE "%CONSTRAINT `fk_configs_details`%" OR sql LIKE "%CONSTRAINT [fk_configs_details]%" OR sql LIKE "%CONSTRAINT 	fk_configs_details	%") -1 <nil>
[db    ] TRACE 2024/02/05 19:12:54 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "config_details" AND name = "idx_unique" -1 <nil>
[db    ] TRACE 2024/02/05 19:12:54 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%CONSTRAINT ""fk_devices_details"" %" OR sql LIKE "%CONSTRAINT fk_devices_details %" OR sql LIKE "%CONSTRAINT `fk_devices_details`%" OR sql LIKE "%CONSTRAINT [fk_devices_details]%" OR sql LIKE "%CONSTRAINT 	fk_devices_details	%") -1 <nil>
[db    ] TRACE 2024/02/05 19:12:54 SELECT count(*) FROM sqlite_master WHERE type = "table" AND tbl_name = "config_details" AND (sql LIKE "%""device_id"" %" OR sql LIKE "%device_id %" OR sql LIKE "%`device_id`%" OR sql LIKE "%[device_id]%" OR sql LIKE "%	device_id	%") -1 <nil>
[sma   ] TRACE 2024/02/05 19:12:54 send 192.168.2.116: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/02/05 19:12:55 send discover package
[sma   ] TRACE 2024/02/05 19:12:55 send 192.168.2.116: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/02/05 19:12:55 recv 192.168.2.125: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2024/02/05 19:12:55 send 192.168.2.125: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/02/05 19:12:55 recv 192.168.2.125: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/02/05 19:12:55 new inverter at 192.168.2.125 - Serial=1261045433
[sma   ] TRACE 2024/02/05 19:12:55 found device 1261045433 at 192.168.2.125
[sma   ] TRACE 2024/02/05 19:12:55 recv 192.168.2.129: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2024/02/05 19:12:55 send 192.168.2.129: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/02/05 19:12:55 recv 192.168.2.120: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2024/02/05 19:12:55 recv 192.168.2.116: [proto.GroupPacketEntry, proto.UnknownPacketEntry, proto.SmaNet2PacketEntry, proto.DiscoveryRequestPacketEntry, proto.DiscoveryIPPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry, proto.UnknownPacketEntry]
[sma   ] TRACE 2024/02/05 19:12:55 send 192.168.2.116: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/02/05 19:12:55 recv 192.168.2.129: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/02/05 19:12:55 new inverter at 192.168.2.129 - Serial=1992024657
[sma   ] TRACE 2024/02/05 19:12:55 found device 1992024657 at 192.168.2.129
[sma   ] TRACE 2024/02/05 19:12:55 recv 192.168.2.116: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/02/05 19:12:55 new energy meter at 192.168.2.116 - Serial=1901703995
[sma   ] TRACE 2024/02/05 19:12:55 send 192.168.2.120: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/02/05 19:12:55 recv 192.168.2.120: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]
[sma   ] TRACE 2024/02/05 19:12:55 new inverter at 192.168.2.120 - Serial=3009011453
[... cut ...]
Energy:         19213.8kWh
Current L1..L3: 0A 0A 0A
Voltage L1..L3: 0V 0V 0V
Power L1..L3:   0W 0W 0W

battery4
--------
[modbus] TRACE 2024/02/05 19:12:57 modbus: send 00 01 00 00 00 06 03 04 78 37 00 02
[modbus] TRACE 2024/02/05 19:12:57 modbus: recv 00 01 00 00 00 07 03 04 04 00 00 02 9e
[modbus] TRACE 2024/02/05 19:12:57 modbus: send 00 02 00 00 00 06 03 03 77 31 00 04
[modbus] TRACE 2024/02/05 19:12:57 modbus: recv 00 02 00 00 00 0b 03 03 08 ff ff ff ff ff ff ff ff
Power:  670W
Energy: 0.0kWh

premultiply, 2024-02-05:
template: `sma-si-modbus`

closed 2024-02-05 by andig

MiRi1975, 2024-02-05:
Passt jetzt! Danke!