evcc-io/evcc #12086: Tesla aktuell

Maxim-7777, 2024-02-05:
### Describe the bug

Gruß aus Augsburg, Bayern,
wenn ich ein Fahrzeug hinzufügen wollte (Tesla NEW) funktioniert so weit alles bis ich auf Fahrzeug speichern drücke, dann passiert einfach nichts, das Auto wird nicht gespeichert/Übernomen in konfig.

[image]

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
mqtt:
  broker: 172.16.0.100:1883
  user: deye
  password: 

meters:
- type: template
  template: shelly-3em
  usage: grid
  host: 172.16.0.200
  name: shelly-3EM
- name: Deye-PV
  type: custom
  power:
    source: mqtt
    topic: Deye12/inverter_1/pv_power/state
- name: Deye-SOC
  type: custom
  soc:
    source: mqtt
    topic: Deye12/total/battery_state_of_charge/state
  power:
    source: mqtt
    topic: Deye12/total/battery_power/state
    scale: -1

chargers:
- type: template
  template: go-e-v3
  host: 172.16.0.807
  name: wallbox2

loadpoints:
- title: Carport
  charger: wallbox2
  mode: pv
  phases: 0
  mincurrent: 6
  maxcurrent: 32
  resetOnDisconnect: true

site:
  title: Carport
  meters:
    grid: shelly-3EM
    pv: Deye-PV
    battery: Deye-SOC

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.43
  feedin:
    type: fixed
    price: 0.082
  planner:
    type: grünstromindex
    zip: 86356
```

### Log details

```text
[lp-1  ] ERROR 2024/02/05 21:16:52 charge meter: Get "http://172.16.0.807/status": dial tcp 172.16.0.807:80: connect: no route to host
[site  ] DEBUG 2024/02/05 21:16:52 pv power: 0W
[site  ] DEBUG 2024/02/05 21:16:52 battery soc: 45%
[site  ] DEBUG 2024/02/05 21:16:52 battery power: 941W
[site  ] DEBUG 2024/02/05 21:16:52 grid power: 31W
[site  ] DEBUG 2024/02/05 21:16:52 grid powers: [18 37 -24]W
[site  ] DEBUG 2024/02/05 21:16:52 grid currents: [2.2 1.7 -0.94]A
[site  ] DEBUG 2024/02/05 21:16:52 site power: 972W
[lp-1  ] ERROR 2024/02/05 21:16:55 charge meter: Get "http://172.16.0.807/status": dial tcp 172.16.0.87:80: connect: no route to host
[lp-1  ] ERROR 2024/02/05 21:16:58 charge meter: Get "http://172.16.0.807/status": dial tcp 172.16.0.87:80: connect: no route to host
[lp-1  ] ERROR 2024/02/05 21:17:01 charge rater: Get "http://172.16.0.807/status": dial tcp 172.16.0.87:80: connect: no route to host
[lp-1  ] ERROR 2024/02/05 21:17:04 charge total import: Get "http://172.16.0.807/status": dial tcp 172.16.0.87:80: connect: no route to host
[lp-1  ] ERROR 2024/02/05 21:17:07 charger: Get "http://172.16.0.807/status": dial tcp 172.16.0.807:80: connect: no route to host
[site  ] DEBUG 2024/02/05 21:17:07 ----
[lp-1  ] ERROR 2024/02/05 21:17:16 charge meter: Get "http://172.16.0.807/status": dial tcp 172.16.0.87:80: connect: no route to host
[site  ] DEBUG 2024/02/05 21:17:16 pv power: 0W
[site  ] DEBUG 2024/02/05 21:17:16 battery soc: 45%
[site  ] DEBUG 2024/02/05 21:17:16 battery power: 947W
[site  ] DEBUG 2024/02/05 21:17:16 grid power: 31W
[site  ] DEBUG 2024/02/05 21:17:16 grid powers: [20 37 -27]W
[site  ] DEBUG 2024/02/05 21:17:16 grid currents: [2.22 1.71 -0.96]A
[site  ] DEBUG 2024/02/05 21:17:17 site power: 978W
```

### What type of operating system are you running?

Linux

### Version

 0.124.0 Latest

stepir, 2024-02-05:
you need to use the nightly build it's a known bug

andig, 2024-02-06:
Config ui is an experimental feature. Please reproduce using

    evcc configure

closed 2024-02-06 by andig