evcc-io/evcc #12093: Fahrzeug wird nicht mehr erkannt

cburghardt, 2024-02-06:
### Describe the bug

Ich habe seit dem Upgrade von 0.123.9 auf 0.124.1 leider wieder das gleiche Problem, dass sich mit der 0.123.8 eingeschlichen hatte.
Das Fahrzeug wird nicht erkannt und es kommen keinerlei Werte von mqtt im Log an.
Im Log sieht das so aus:

`Feb 06 11:23:24 evcc evcc[548]: [main  ] INFO 2024/02/06 11:23:24 evcc 0.124.1
Feb 06 11:23:24 evcc evcc[548]: [main  ] INFO 2024/02/06 11:23:24 using config file: /etc/evcc.yaml
Feb 06 11:23:24 evcc evcc[548]: [main  ] INFO 2024/02/06 11:23:24 starting ui and api at :7070
Feb 06 11:23:24 evcc evcc[548]: [db    ] INFO 2024/02/06 11:23:24 using sqlite database: /var/lib/evcc/evcc.db
Feb 06 11:23:24 evcc evcc[548]: [mqtt  ] INFO 2024/02/06 11:23:24 connecting evcc at tcp://192.168.178.109:1883
Feb 06 11:23:24 evcc evcc[548]: [mqtt  ] DEBUG 2024/02/06 11:23:24 tcp://192.168.178.109:1883 connected
Feb 06 11:23:24 evcc evcc[548]: [jlr   ] WARN 2024/02/06 11:23:24 new device id generated, add `deviceid` to config: 0ef962f5-5218-4786-bdb6-c3ea4c25f5d7
Feb 06 11:23:26 evcc evcc[548]: [lp-1  ] DEBUG 2024/02/06 11:23:26 charge total import: 1696.429kWh
Feb 06 11:23:26 evcc evcc[548]: [site  ] INFO 2024/02/06 11:23:26 site config:
Feb 06 11:23:26 evcc evcc[548]: [site  ] INFO 2024/02/06 11:23:26   meters:      grid ✓ pv ✓ battery ✓
Feb 06 11:23:26 evcc evcc[548]: [site  ] INFO 2024/02/06 11:23:26     grid:      power ✓ energy ✗ currents ✓
Feb 06 11:23:26 evcc evcc[548]: [site  ] INFO 2024/02/06 11:23:26     pv 1:      power ✓ energy ✗ currents ✗
Feb 06 11:23:26 evcc evcc[548]: [site  ] INFO 2024/02/06 11:23:26     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
Feb 06 11:23:26 evcc evcc[548]: [site  ] INFO 2024/02/06 11:23:26   vehicles:
Feb 06 11:23:26 evcc evcc[548]: [site  ] INFO 2024/02/06 11:23:26     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
Feb 06 11:23:26 evcc evcc[548]: [site  ] INFO 2024/02/06 11:23:26     vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
Feb 06 11:23:26 evcc evcc[548]: [lp-1  ] INFO 2024/02/06 11:23:26 loadpoint 1:
Feb 06 11:23:26 evcc evcc[548]: [lp-1  ] INFO 2024/02/06 11:23:26   mode:        off
Feb 06 11:23:26 evcc evcc[548]: [lp-1  ] INFO 2024/02/06 11:23:26   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
Feb 06 11:23:26 evcc evcc[548]: [lp-1  ] INFO 2024/02/06 11:23:26   meters:      charge ✓
Feb 06 11:23:26 evcc evcc[548]: [lp-1  ] INFO 2024/02/06 11:23:26     charge:    power ✓ energy ✓ currents ✓
Feb 06 11:23:26 evcc evcc[548]: [site  ] DEBUG 2024/02/06 11:23:26 set buffer soc: 90
Feb 06 11:23:26 evcc evcc[548]: [site  ] DEBUG 2024/02/06 11:23:26 set priority soc: 40
Feb 06 11:23:26 evcc evcc[548]: [lp-1  ] DEBUG 2024/02/06 11:23:26 phase timer inactive
Feb 06 11:23:26 evcc evcc[548]: [lp-1  ] DEBUG 2024/02/06 11:23:26 pv timer inactive
Feb 06 11:23:26 evcc evcc[548]: [lp-1  ] DEBUG 2024/02/06 11:23:26 guard timer inactive
Feb 06 11:23:26 evcc evcc[548]: [lp-1  ] DEBUG 2024/02/06 11:23:26 max charge current: 6A
Feb 06 11:23:26 evcc evcc[548]: [lp-1  ] DEBUG 2024/02/06 11:23:26 guard enable in 5m0s
Feb 06 11:23:26 evcc evcc[548]: [site  ] DEBUG 2024/02/06 11:23:26 ----
Feb 06 11:23:26 evcc evcc[548]: [lp-1  ] DEBUG 2024/02/06 11:23:26 charge power: 1309W
Feb 06 11:23:26 evcc evcc[548]: [site  ] DEBUG 2024/02/06 11:23:26 pv power: 3685W
Feb 06 11:23:26 evcc evcc[548]: [site  ] DEBUG 2024/02/06 11:23:26 battery soc: 43%
Feb 06 11:23:26 evcc evcc[548]: [site  ] DEBUG 2024/02/06 11:23:26 battery power: -1732W
Feb 06 11:23:26 evcc evcc[548]: [site  ] DEBUG 2024/02/06 11:23:26 grid power: 57W`

Und ab hier ist es still. Normal kommen alle paar Sekunden dann per DEBUG Meldung die neuen mqtt Werte rein, hier passiert nichts.
Egal welches Fahrzeug ich anstecke, das UI bleibt bei kein Fahrzeug. Die Verbrauchswerte aktualisieren sich auch nicht.

### Steps to reproduce

Fahrzeug anstecken, Logfile zeigt nichts

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: 2dd502a030abe216cc6cdde4984bc60a198e279b6e5a52e250212eea441cb430

interval: 30s # control cycle interval

meters:
  - name: pv
    type: custom
    power:
      source: mqtt
      topic: homeassistant/sensor/power_solar_generation/state
      timeout: 24h
  - name: grid      
    type: custom
    power:  
      source: mqtt
      topic: homeassistant/sensor/solaredge_modbus_m1_ac_power/state
      scale: -1
      timeout: 10m
    currents:
      - source: mqtt    
        topic: homeassistant/sensor/solaredge_modbus_m1_ac_current_a/state
        timeout: 8h
      - source: mqtt    
        topic: homeassistant/sensor/solaredge_modbus_m1_ac_current_b/state
        timeout: 8h
      - source: mqtt    
        topic: homeassistant/sensor/solaredge_modbus_m1_ac_current_c/state
        timeout: 8h
  - name: battery
    type: custom
    power:
      source: mqtt
      topic: homeassistant/sensor/solaredge_modbus_b1_dc_power/state
      scale: -1
      timeout: 24h 
    soc:
      source: mqtt
      topic: homeassistant/sensor/solaredge_modbus_b1_state_of_energy/state
      timeout: 24h

chargers:
  - name: wallbox
    type: template
    template: go-e-v3
    host: go-echarger-217527

site:
  title: Home
  residualPower: 100
  maxGridSupplyWhileBatteryCharging: 50
  meters:
    pv: pv
    grid: grid
    battery: battery  

loadpoints:
  - title: Garage
    charger: wallbox

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.40  

mqtt:
  broker: 192.168.178.109:1883
  topic: evcc 
  clientid: evcc

vehicles:
  - name: disco
    type: template
    title: Discovery Sport
    template: jaguar-landrover
    phases: 1  
    maxcurrent: 16
  - name: etron
    type: template
    title: ETron
    template: audi
    phases: 3  
    maxcurrent: 20
```

### Log details

```text
Siehe oben
```

### What type of operating system are you running?

Linux

### Version

0.124.1

cburghardt, 2024-02-06:
Gleiches Verhalten wie hier: https://github.com/evcc-io/evcc/issues/11666

andig, 2024-02-06:
Bitte mit `--profile` starten und Ausgabe von /debug/pprof teilen.

brenndorfler, 2024-02-06:
Hmm... ich habe auch gerade geupdated von 1.122.1
Dabei kamen einige Meldungen bzgl. der Konfig, welche ich soweit beheben konnte.

Allerdings wurden bei mir die Werte auch nicht aktualisiert und die Ladesteuerung (fronius-wattpilot) war nicht möglich.
Ich bin dann mit `apt-get install evcc=0.124.0` wieder zurück, da läuft alles wie erwartet.
Nochmal auf die 0.124.1 hat wieder gleiches Verhalten gezeigt.

Bin eigentlich eher Anwender, wenn ich irgendwie bei der Fehlersuche unterstützen kann, bitte kurzen Hinweis was ich tun muss. ;-)

andig, 2024-02-06:
Der Hinweis steht direkt über Deinem Post

cburghardt, 2024-02-06:
Ich bin gerade etwas verwirrt was hier passiert. Ich hatte auf die Version 123.7 downgegraded damit es wieder geht. Um den Fehler zu reproduzieren bin ich nun wieder auf die 124.1 gegangen und kann das Verhalten aber nicht mehr nachstellen. Der Start geht ohne Probleme und die Werte kommen rein. Ich kann mir nur vorstellen dass es hier irgendein lock Problem gibt, aber momentan kriege ich es nicht reproduziert.

brenndorfler, 2024-02-06:
> Der Hinweis steht direkt über Deinem Post

Danke, habe jetzt nochmal die 0.124.1 installiert. Immer noch gleiches Verhalten.
Den Service gestoppt und als root mit `evcc --profile` aufgerufen.

Doofe Frage: Wo finde ich denn die Datei `/debug/pprof`, wohl nicht im root-Verzeichnis.

andig, 2024-02-06:
Das muss im Browser angehängt werden.

brenndorfler, 2024-02-06:
Sorry, ich glaube das überfordert mich... im Browser? 
Also auf dem Client und nicht auf dem Server?
Finde ich dazu irgendwo etwas weiterführende Info?

Hab schon hier gelesen, aber von `evcc --profile` steht da leider nix: https://docs.evcc.io/docs/guides/setup/#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen

Würde ja gerne unterstützen bei der Fehlersuche, aber ich glaube ich brauche dazu etwas mehr Input... Sorry.

mdkeil, 2024-02-06:
> Sorry, ich glaube das überfordert mich... im Browser?

http://EVCC-IP/debug/pprof nachdem evcc mit --profile gestartet wurde.

brenndorfler, 2024-02-06:
> http://EVCC-IP/debug/pprof nachdem evcc mit --profile gestartet wurde.

ah, Danke, wir kommen der Sache näher:

[image]

Und was wird jetzt genau benötigt?

brenndorfler, 2024-02-07:
Hallo,

hier die Debug-Logs.

[debug.txt](https://github.com/evcc-io/evcc/files/14196879/debug.txt)
[goroutine.txt](https://github.com/evcc-io/evcc/files/14196881/goroutine.txt)

Wenn das Fahrzeug nicht verbunden ist, läuft es übrigens. Hatte ich heute Nachmittag kurz probiert. 

Den evcc-Prozess habe ich übrigens gestartet mit ANGESCHLOSSENEM Fahrzeug.
Unten im Debug Log kann man nochmal sehen, wie ich ohne Erfolg probiert habe von PV zu NOW und zurück zu wechseln. Da passiert auch nix...

andig, 2024-02-07:
> Unten im Debug Log kann man nochmal sehen, wie ich ohne Erfolg probiert habe von PV zu NOW und zurück zu wechseln. Da passiert auch nix...

Nochmal langsam. Soweit ich sehe ist das im Logfile erkennbar:

> set charge mode: now

evcc "steht" also nicht. Kommen danach weitere Logfileeinträge? Oder "steht" die Anwendung?

Griepen, 2024-02-07:
Bei uns war gestern auch plötzlich das Fahrzeug weg. Ich hatte nichts an Version oder Konfiguration von EVCC geändert. Habe dann von 121.5 auf 124.1  geupdated. Dann habe ich minSoc und TargetSoc mal gelöscht, weil irgendwas in der Yaml EVCC nicht mehr im Browser anzeigen ließ. Ich krieg keine Daten mehr von der ZOE. Ich kann noch laden, aber über das Auto ist nichts mehr bekannt außer, dass der Name noch dort steht. Seit dem tauchen Vehicle Fehler auf.
`[main  ] INFO 2024/02/07 18:17:07 evcc 0.124.1
[main  ] INFO 2024/02/07 18:17:07 using config file: /etc/evcc.yaml
[db    ] INFO 2024/02/07 18:17:08 using sqlite database: /home/openhabian/.evcc/evcc.db
.Soc:           unexpected status: 403 (Forbidden)
Capacity:       51.0kWh
Charge status:  unexpected status: 403 (Forbidden)
Range:          unexpected status: 403 (Forbidden)
Odometer:       unexpected status: 403 (Forbidden)
Finish time:    unexpected status: 403 (Forbidden)
Climate active: false
Position:       53.30740507,8.4803079
Features:       []
`

Ist es für dieses THema hilfreich hier noch mehr Details von unserem Fall zu posten? Ich denke vielleicht hat sich ein genereller Fehler eingeschlichen. Oder sollte ich da ein neues Thema öffnen?

brenndorfler, 2024-02-07:
> evcc "steht" also nicht. Kommen danach weitere Logfileeinträge? Oder "steht" die Anwendung?

Also, in der GUI werden keine Werte mehr aktualisiert. Ich kann aber noch den Lademodus hin- und herschalten. Das sehe ich auch auf der Console `(evcc -l debug --profile`). Wenn ich aber nix in der GUI triggere, dann ist im Loglevel DEBUG Ruhe auf der Console.
Ich habe es mal zu TRACE hochgedreht, das sehe ich immer wir zwei meiner PV-Anlage (beide SMAs) abgefragt werden:

`[sma   ] TRACE 2024/02/07 20:50:41 login for 192.168.110.241:9522`
`[sma   ] TRACE 2024/02/07 20:50:41 send 192.168.110.241: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]`
`[sma   ] TRACE 2024/02/07 20:50:41 recv 192.168.110.241: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]`
`----8<---- *SCHNIPP*--------8<---- *SCHNIPP*--------8<---- *SCHNIPP*--------8<---- *SCHNIPP*--------8<---- `
`[sma   ] TRACE 2024/02/07 20:50:42 requestValues for 192.168.110.241:9522: 0x5800 0x821E00 0x8220FF`
`[sma   ] TRACE 2024/02/07 20:50:42 send 192.168.110.241: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]`
`[sma   ] TRACE 2024/02/07 20:50:42 recv 192.168.110.241: [proto.GroupPacketEntry, proto.SmaNet2PacketEntry]`
`[sma   ] TRACE 2024/02/07 20:50:42 logout for 192.168.110.241:9522`

Das wiederholt sich mit einer zweiten SMA-Anlage mit anderer IP. 

Also, irgendwas läuft noch, Updates oder Regelungen finden nicht statt.

Was mir auffällt, bzw. ein paar Gedanken von mir:
Ich habe noch eine dritte PV-Anlage, eine Fronius die über Modbus abgefragt wird. Diese sehe ich hier nicht im TRACE-Log.
Außerdem hat die Fronius-Anlage noch auch einen DC-Speicher. Den steuere ich schon mit fhem über MODBUS.
Dazu habe ich irgendwas im Changelog gelesen, dass MODBUS mit PW sein muss, wenn die Batterie von evcc gesteuert werden soll.
Ich habe aber kein Passwort für MODBUS im Wechselrichter gesetzt, da ich den ganzen kram abgeschottet im eigenen VLAN betreibe. Das läuft seit 2018 stabil.

Kann das evtl. relevant sein? Wie gesagt, die 0.124.0 funktioniert 1a... muss also mit den letzten Änderungen passiert sein.

andig, 2024-02-07:
Bitte mal das KOMPLETTE Logfile

brenndorfler, 2024-02-07:
Das debug.txt von oben ist im Level DEBUG bereits vollständig.
Im Level TRACE sehe ich u.a. Tokens, kann ich das irgendwie privat teilen? 
Hier nochmal die Config [config.txt](https://github.com/evcc-io/evcc/files/14201231/config.txt)

andig, 2024-02-08:
@brenndorfler Du hast den Wattpilot? Dann vmtl. weiter in https://github.com/evcc-io/evcc/issues/12136#issuecomment-1934116896.

@cburghardt bitte Details gem. https://github.com/evcc-io/evcc/issues/12093#issuecomment-1929337104

brenndorfler, 2024-02-08:
> @brenndorfler Du hast den Wattpilot? Dann vmtl. weiter in [#12136 (comment)](https://github.com/evcc-io/evcc/issues/12136#issuecomment-1934116896).

Danke, das trifft es..

cburghardt, 2024-02-08:
[goroutine.txt](https://github.com/evcc-io/evcc/files/14212006/goroutine.txt)
@andig goroutines attached

Log dazu:
```
[main  ] INFO 2024/02/08 17:30:17 evcc 0.124.1
[main  ] INFO 2024/02/08 17:30:17 using config file: /etc/evcc.yaml
[main  ] INFO 2024/02/08 17:30:17 starting ui and api at :7070
[db    ] INFO 2024/02/08 17:30:17 using sqlite database: /home/carsten/.evcc/evcc.db
[mqtt  ] INFO 2024/02/08 17:30:17 connecting evcc at tcp://192.168.178.109:1883
[mqtt  ] DEBUG 2024/02/08 17:30:17 tcp://192.168.178.109:1883 connected
[jlr   ] WARN 2024/02/08 17:30:17 new device id generated, add `deviceid` to config: e113d308-1a62-41e8-84cd-05064069757e
[lp-1  ] DEBUG 2024/02/08 17:30:18 charge total import: 1739.457kWh
[site  ] INFO 2024/02/08 17:30:18 site config:
[site  ] INFO 2024/02/08 17:30:18   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2024/02/08 17:30:18     grid:      power ✓ energy ✗ currents ✓
[site  ] INFO 2024/02/08 17:30:18     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2024/02/08 17:30:18     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2024/02/08 17:30:18   vehicles:
[site  ] INFO 2024/02/08 17:30:18     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[site  ] INFO 2024/02/08 17:30:18     vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2024/02/08 17:30:18 loadpoint 1:
[lp-1  ] INFO 2024/02/08 17:30:18   mode:        off
[lp-1  ] INFO 2024/02/08 17:30:18   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2024/02/08 17:30:18   meters:      charge ✓
[lp-1  ] INFO 2024/02/08 17:30:18     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2024/02/08 17:30:18 phase timer inactive
[lp-1  ] DEBUG 2024/02/08 17:30:18 pv timer inactive
[lp-1  ] DEBUG 2024/02/08 17:30:18 guard timer inactive
[lp-1  ] DEBUG 2024/02/08 17:30:18 max charge current: 6A
[lp-1  ] DEBUG 2024/02/08 17:30:18 guard enable in 5m0s
[site  ] DEBUG 2024/02/08 17:30:18 ----
[lp-1  ] DEBUG 2024/02/08 17:30:18 charge power: 0W
[lp-1  ] DEBUG 2024/02/08 17:31:01 set charge mode: now

```
Dann war es still, keine weiteren Ausgaben

cburghardt, 2024-02-08:
Noch als Hintergrund: evcc ging soweit, nun habe ich es neu gestartet und der Fehler ist wieder aufgetreten, es hängt also. Die logs oben sind vom aktuellen Stand, keinerlei Reaktion des Prozesses.

andig, 2024-02-08:
@cburghardt Dein Fehlerbild ist auch klar. Dein Zähler verwendet mqtt mit endlos langem Timeout. Anscheinend gibts da keine Werte, so dass evcc einfach wartet... Lösung: dafür sorgen, dass ist mqtt werden stehen und/oder Timeout auf sinniges Maß reduzieren!

closed 2024-02-08 by andig

cburghardt, 2024-02-08:
@andig über mqtt kommen laufend Werte rein, aber evcc steht trotzdem. Wenn ich parallel einen subscribe auf mqtt mache, kommen permanent updates rein. Vor allem weil ich evcc sofort wieder zum Leben erwecken kann, indem ich auf die 123.7 downgrade.
Wenn hier auf einen Wert gewartet und geblockt (!) wird, ist das ein fehlerhaftes Verhalten von neueren Versionen.

andig, 2024-02-09:
> mqtt Werte kommen nur rein, wenn sie sich ändern, wenn ich den Prozess also zB in der Nacht starte, blockt evcc bis alle Werte angekommen sind. 

Das kannst du mit retained messages lösen. Natürlich kann evcc nur arbeiten wenn es die Daten kennt. Wenn Du nicht auf die Daten warten willst dann müssen kürzere Timeouts konfiguriert werden.

> allerdings ist das Ergebnis natürlich problematisch

Das klingt als würdest Du eine Lösung in evcc erwarten? Welche?

cburghardt, 2024-02-09:
> Das kannst du mit retained messages lösen. Natürlich kann evcc nur arbeiten wenn es die Daten kennt. Wenn Du nicht auf die Daten warten willst dann müssen kürzere Timeouts konfiguriert werden.

Danke für den Hinweis, mal sehen ob mit den retained messages klappt.

> 
> > allerdings ist das Ergebnis natürlich problematisch
> 
> Das klingt als würdest Du eine Lösung in evcc erwarten? Welche?

Mir ist klar, dass evcc nur arbeiten kann, wenn alle Daten vorhanden sind. Aber es würde helfen, wenn der Prozess nicht blockt. Auf dem UI sieht man ja, dass noch Daten fehlen, aber man kommt schlichtweg nicht soweit.
Außerdem weiß man aktuell ja gar nicht, was evcc so treibt, es hängt eben.

andig, 2024-02-09:
> Aber es würde helfen, wenn der Prozess nicht blockt

Das tut es nicht wenn Du entsprechend kurzen Timeout setzt. Irgendwoher muss evcc ja wissen, wie lange es auf Daten warten soll... Und genau an der Stelle haben wir uns einmal im Kreis gedreht!